raspiblitz/raspiblitz #256: Backup HDD

wuestenigel, 2019-01-09:
Hey,

is there a way to backup my HDD?

As a user I want to backup my personal settings, but also the blockchain data so save some time in case of a HDD crash.

Already tried to plug it into my iMac, but it wasn't readable. Is there any existing script on the raspiblitz image that may help?

fluidvoice, 2019-01-09:
admin@node $ `man rsync`
or google about how to run **rsync**.

fluidvoice, 2019-01-09:
Here is a decent page I found. Has everything except how to run over SSH (need to find this still):
https://www.thegeekstuff.com/2011/01/rsync-exclude-files-and-folders/

closed 2019-01-09 by wuestenigel

fluidvoice, 2019-01-09:
Btw, the -n switch will do nothing (dry run) which you can use with --stat to show at the end the total size of the backup, so you know if you have enough space:

`sudo rsync --stats -avzphn admin@raspi:/ /mnt/ext3-mjro-us/backups/raspiblitz`

gave me at the end...

> Number of files: 220,935 (reg: 180,117, dir: 19,512, link: 21,276, special: 30)
> Number of created files: 220,934 (reg: 180,117, dir: 19,511, link: 21,104, dev: 172, special: 30)
> Number of deleted files: 0
> Number of regular files transferred: 180,117
> Total file size: 285.60G bytes
> Total transferred file size: 285.60G bytes

Given the size this will obviously backup not only the SD-card but the blockchain DB as well.
So apparently that one command is all you need to backup everything. I'm gonna try it now.

rootzoll, 2019-01-09:
@fluidvoice thanks for sharing your backup efforts. And you are correct - if you do any backup i would always recommend to stop the lightning services. But you can ignore the bitcoin data on backup - there is no personal data in there and all can be redownloaded/resynced from public networks.

To generally anwere the topic of backup (because it comes up a lot) - started an FAQ on that:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-backup-my-lightning-node

fluidvoice, 2019-01-09:
> @fluidvoice thanks for sharing your backup efforts. And you are correct - if you do any backup i would always recommend to stop the lightning services. But you can ignore the bitcoin data on backup [...]

I think backing up the blockchain DB is much preferred to resyncing for some people, myself included, due to slow internet connection.  Plus some people have experienced DB corruption on power failure. It's a total PITA to  have to resync when that happens.  Granted backing up takes a while too. Better if connected via ethernet.

wuestenigel, 2019-01-10:
Thanks for sharing the scripts you created.

I agree that bitcoin blockchain is public, but on the other hand my node should also have a high availability. If I need to download the Blockchain data again, my LN node will be offline for a week. (Question: Is there a reputation algorithm in LN for calculation routes and does a high downtime decreases this kind of reputation?)

On the other hand I'm not sure if I understood the LND backup correctly. 

1) If I do not backup the LND data, my funds may be lost in case of HDD crash?

2) Even if I make a backup it may occur that I loose my funds because of the built-in punishment system that prevents replay attacks? So If I install an old LND image from another disk I may get punished if there are newer transactions?

fluidvoice, 2019-01-10:
1. No. See: https://github.com/rootzoll/raspiblitz/blob/783c5de56fd4b2ca263699c6d3ccacd87c4da446/README.md#recover-your-coins-from-a-failing-raspiblitz

fluidvoice, 2019-01-13:
dropping some notes here on how to automate backing up using **rsync** :

> edit your user cron file with "crontab -e" and add this line'
> 
> */30 * * * *   rsync -avu --delete /lnd_data_folder/ /pendrive_folder/
> 
> (Replace each folder with your configuration)
> 
> It will copy every 30 minutes LND folder

rootzoll, 2019-01-13:
@fluidvoice --> I explained a bit more why I choosed "update by sd card" in the FAQ now - please see here: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#why-do-i-need-to-re-burn-my-sd-card-for-an-update-after-version-098

ABOUT BACKUP:

LND just announced some backup solution is "comming soon" - at least thats how I understood that tweet from Alex: https://twitter.com/alexbosworth/status/1083394904900825089

Instead of stopping LND+rsyncing by cron and so still risking to have an old backup and loosing all funds, I would recommend to look into a RAID1-Solution to protect against HDD data loss. When the OS writes the data to two HDDs at the same time - that could be the safest way todo it. But I never setup a RAID1 with two HDD over USB on Raspberry ... but if possible, thats most def something to look into to add the raspiblitz as a feature - even if LND comes up with some other backup solution in the future. Please share if anybody has some experience in that context.