raspiblitz/raspiblitz #257: Hardware case and display not fully compatible

raumi75, 2019-01-09:
I bought the case and the display from the German shopping list. the case is officially for a 3.2" display whereas the display is slightly larger (3.5"), so the top frame does not close. It works ok and the price is great, but I have a feeling there might be a better solution.

fluidvoice, 2019-01-09:
I have a 3.5" display an use this one. https://thecryptocloak.com/product/lightningshell/

raumi75, 2019-01-09:
That case looks very good. I personally won't change anything and just use the case without the frame. "works for me"™️

I bought the one from the German shopping list: https://www.amazon.de/gp/product/B06X191RX7

fluidvoice, 2019-01-09:
> That case looks very good. I personally won't change anything and just use the case without the frame. "works for me"
> 
> I bought the one from the German shopping list: https://www.amazon.de/gp/product/B06X191RX7

this one is cheap and might work for you: https://www.amazon.de/Ils-Generation-Schicht-5-Zoll-Display-Raspberry/dp/B079122RMQ/

goodc0re, 2019-01-09:
Had the same issue with the case from the German shopping list and think that the shopping list should be changed or a note added that the case doesn't fit.

rootzoll, 2019-01-09:
Its correct that the final topping of the case in the german shopping list does not fit. But that was OK from my side because the bigger LCD covers the top perfectly. If someone ca recommend a better fitting case in the same price point on amazon feel free to make a pull request. I added a short note to ignore to topping in the README.

closed 2019-01-09 by rootzoll