raspiblitz/raspiblitz #258: opening channel from LND to c-lightning fails

fluidvoice, 2019-01-09:
> Have just managed to connect to a few clearnet nodes, must have just been the onion ones I was trying to open channels with, strange they could connect but couldn't open channel

Myself and one other person have seen opening a channel to a **c-lightning** (Blockstream) node from an **LND** node (ie Raspiblitz) fails like this. For me it's very repeatable.  Other LND nodes are not a problem for me.  I will try to look into the problem.  I can connect to c-lightning nodes but not open a channel. 

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issue_comments#issuecomment-452756599_

rootzoll, 2019-01-09:
This sounds like a case for the LND devs to look into - please someone forward this issue to LND Dev Slack channel https://lightningcommunity.slack.com or file an issue at LND GitHub: https://github.com/LightningNetwork/lnd

I am closing this issue for RaspiBlitz for now. If anybody finds an ansere please post a link to it on this issue so that others can find it better. Thanks.

closed 2019-01-09 by rootzoll

fluidvoice, 2019-01-09:
I did. Here it is for anyone want's to follow: 
https://github.com/lightningnetwork/lnd/issues/2443#issue-397466897