raspiblitz/raspiblitz #259: After enabling Channel Autopilot and LND auto-unlock ZAP Mobile wallet can't connect

eurtnl, 2019-01-09:
After enabling Channel Autopilot and LND auto-unlock ZAP Mobile wallet can't connect. 

Tried a new qr, but also no connection.
Any idea how to solve?

rootzoll, 2019-01-09:
When switchen on auto-unlock it changes the tlsextraip=0.0.0.0 in the lnd.conf ... hmmm could be that this conflicts with the tls.cert used by ZAP .. not sure yet. Will put this on my list to test before 1.0 release. If anybody has more insight, please share.

Smiggel, 2019-01-09:
Is rpclisten=0.0.0.0:10009 not commented in the lnd.conf?

eurtnl, 2019-01-09:
Well i was able to connect ZAP desktop tls.cert

raumi75, 2019-01-09:
> Well i was able to connect ZAP desktop with the same tls.cert

What did you do to make it work? I'm having the same issues? Should I comment this line out in `~/assets/lnd.bitcoin.conf` ? and restart the `service lnd restart`

rootzoll, 2019-01-13:
I will re-check that the rpclisten is active/open by default on final release tests. Thx.

rootzoll, 2019-01-18:
When enable/disable the Auto-Unlock there is the need to reconnect the mobile/desktop wallets, because the macaroon/TLS changes. I added a Dialog-Box with the info about that, when switching Auto-Unlock on/off.

 Testing OK so far now. Closin this issue for next release so far.

closed 2019-01-18 by rootzoll