raspiblitz/raspiblitz #260: What happend?

eurtnl, 2019-01-09:
[image]
Today i decided to fund my wallet with a small amount for testing purpose. So i send 0.001 BTC to my wallet, made and some dinner, connected ZAP dektop. And i found my wallet as shown above.

As you can see, 10 minutes after receiving my deposit, something/one send 0.00064982 BTC (**it was not me!**). Now when you deduct 0.00064982 from 00.1 that would leave 0,00035018. But my wallet says it has 0,00089888. Why the difference?

What happend?

fluidvoice, 2019-01-09:
from SSH:
`lncli listchaintxns`

fluidvoice, 2019-01-09:
Either you opened a channel with (ln.BitSoapBox.com) or they opened a channel with you of 0.00060000 BTC:
https://1ml.com/node/027a532a96510407b6ca0252fdb045fae98a40febe08b443533b9edc2dd5db127d
The other differences are probably because of bitcoin transaction fees.

eurtnl, 2019-01-09:
Yeah i figured that, now stopping autopilot :-)

closed 2019-01-09 by eurtnl

rootzoll, 2019-01-09:
To prevent that confusing behavior AutoPilot is turned off by default ;)