raspiblitz/raspiblitz #266: New installaton with seeds doesn't show previous status

lucbran, 2019-01-11:
Installed node and got it working. Opened channel and transfered few satoshis.
After a problem in booting I decided for a new fresh setuup and following instructions I reinstalled wallet with previous seed.
Now wallet installed but not showing linked channels and funds show zero (not even funds which were not linked to a channel).
Can I retrieve my funds through BIP39 using seed and get those funds back into my wallet?

rootzoll, 2019-01-13:
I think this was the recover instructions you followed:
https://github.com/rootzoll/raspiblitz#1-recover-from-wallet-seed

Until the channel counter parties close the channel can take (much) time and its not 100% sure they will. 

@lucbran to get more details/support on that topic please ask at the LND Dev Slack channel https://lightningcommunity.slack.com .. if you get information there you think are good to add to the instructions above, please post here or make a pull request. I close this issue for now.

closed 2019-01-13 by rootzoll

lucbran, 2019-01-14:
Ok but I also had few satoshis in wallet not yet n any channel.. They dont appear anywhere in my new installation..
Tried retrieving private addrss from seed (using BIP 39) but returns: invalid mnemonic...

rootzoll, 2019-01-15:
@lucbran if you can ask about your recovery issue in the LND slack and share your findings after that would be great. I think you are not the only one interested in this. thx.