raspiblitz/raspiblitz #272: wallet unlock error: sig is larger than the max allowed size

fluidvoice, 2019-01-14:
My node perhaps rebooted from power failure, dunno. But it was unsynced, then after re-syncing it continually asks to unlock the wallet and fails with this:

2019-01-14 02:30:30.861 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
2019-01-14 02:30:30.863 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-01-14 02:30:30.865 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-01-14 02:30:30.872 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-01-14 02:30:30.872 [INF] RPCS: password RPC server listening on [::]:10009
2019-01-14 02:30:30.873 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-01-14 02:39:25.028 [INF] LNWL: Opened wallet
2019-01-14 02:39:25.516 [INF] LTND: Primary chain is set to: bitcoin
2019-01-14 02:39:25.570 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://*:28332
2019-01-14 02:39:25.570 [INF] LTND: Initializing bitcoind backed fee estimator
2019-01-14 02:39:25.570 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://*:28333
2019-01-14 02:39:25.575 [DBG] LNWL: Using minimum fee rate of 253 sat/kw
2019-01-14 02:39:32.149 [INF] LNWL: The wallet has been unlocked without a time limit
2019-01-14 02:39:32.190 [INF] LTND: LightningWallet opened
2019-01-14 02:39:32.190 [DBG] LNWL: Birthday block has already been verified: height=553299, hash=0000000000000000002fb5012be16342623a11878216e6b9fc2b75cf4f702ece
2019-01-14 02:39:32.297 [INF] HSWC: Restoring in-memory circuit state from disk
2019-01-14 02:39:32.347 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2019-01-14 02:39:32.373 [INF] HSWC: Trimming open circuits for chan_id=557641:1467:0, start_htlc_id=0
2019-01-14 02:39:32.373 [INF] HSWC: Trimming open circuits for chan_id=557944:1598:1, start_htlc_id=0
2019-01-14 02:39:32.522 **[ERR] SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 227, max 80]**

2019-01-14 02:39:32.523 [INF] LTND: Shutdown complete

_Originally posted by @fluidvoice in https://github.com/_render_node/MDEyOklzc3VlQ29tbWVudDQ1Mzk2MjYyMw==/timeline/issue_comment#issuecomment-453962623_

fluidvoice, 2019-01-14:
Same error was seen here: https://github.com/rootzoll/raspiblitz/issues/175#issuecomment-439709822

fluidvoice, 2019-01-14:
rebooted and did SSH `lncli unlock` and tail of bitcoind and lnd looks OK for now.
Waiting for re-sync.

fluidvoice, 2019-01-14:
sync finished a few minutes later and the LCD came up as normal.  So what caused the problem is still a mystery.
I notice again that the LCD is displaying TOR is active now (again) even though it was turned off before.
This is an intermittent issue where TOR sometimes is on and sometimes off after a reboot.  This needs to be investigated.

fluidvoice, 2019-01-15:
This just happened again.  I don't know if the node rebooted due to power failure or other but I think not.
Needs further investigation.

fluidvoice, 2019-01-15:
`sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log |less`
2019-01-15 04:52:52.026 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
2019-01-15 04:52:52.026 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-01-15 04:52:52.029 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-01-15 04:52:52.035 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-01-15 04:52:52.035 [INF] RPCS: password RPC server listening on [::]:10009
2019-01-15 04:52:52.039 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-01-15 07:37:00.537 [INF] LNWL: Opened wallet
2019-01-15 07:37:01.212 [INF] LTND: Primary chain is set to: bitcoin
2019-01-15 07:37:01.463 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://*:28332
2019-01-15 07:37:01.463 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://*:28333
2019-01-15 07:37:01.479 [INF] LTND: Initializing bitcoind backed fee estimator
2019-01-15 07:37:01.526 [DBG] LNWL: Using minimum fee rate of 253 sat/kw
2019-01-15 07:37:08.097 [INF] LNWL: The wallet has been unlocked without a time limit
2019-01-15 07:37:08.106 [INF] LTND: LightningWallet opened
2019-01-15 07:37:08.106 [DBG] LNWL: Birthday block has already been verified: height=553299, hash=0000000000000000002fb5012be16342623a11878216e6b9fc2b75cf4f702ece
2019-01-15 07:37:08.320 [INF] HSWC: Restoring in-memory circuit state from disk
2019-01-15 07:37:08.361 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2019-01-15 07:37:08.593 [INF] HSWC: Trimming open circuits for chan_id=557641:1467:0, start_htlc_id=0
2019-01-15 07:37:08.594 [INF] HSWC: Trimming open circuits for chan_id=557944:1598:1, start_htlc_id=0
2019-01-15 07:37:08.594 [INF] HSWC: Trimming open circuits for chan_id=558507:2703:1, start_htlc_id=0
2019-01-15 07:37:08.727 [INF] LNWL: Started rescan from block 000000000000000000273e0260c982f7148039ce44bc2de90e9d0f5662a6c900 (height 558542) for 137 addresses
2019-01-15 07:37:08.732 [INF] LNWL: Starting rescan from block 000000000000000000273e0260c982f7148039ce44bc2de90e9d0f5662a6c900
2019-01-15 07:37:08.991 [ERR] SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 183, max 80]

2019-01-15 07:37:08.992 [INF] LTND: Shutdown complete
2019-01-15 07:38:09.497 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
2019-01-15 07:38:09.498 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-01-15 07:38:09.502 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-01-15 07:38:09.514 [INF] RPCS: password RPC server listening on [::]:10009
2019-01-15 07:38:09.514 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-01-15 07:38:09.515 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

fluidvoice, 2019-01-15:
just did `sudo -u bitcoin lncli unlock` and it seems to have worked. 
`tail -f` of lnd.log looks like it's progressing normally. Awaiting sync finish.

fluidvoice, 2019-01-15:
Sync appeared to have finished and LCD display came up.  Zero peers connected though.

$ `sudo -u bitcoin lncli getinfo`
{
    "identity_pubkey": "038d3eb5339f78774697448ce2414b429b41edd8f65dcc2edc33e591001476c948",
    "alias": "thundahub",
    "num_pending_channels": 0,
    "num_active_channels": 0,
    "num_peers": 0,
    "block_height": 558660,
    "block_hash": "0000000000000000001b074532566129152ee7b04b0658711a47c6480ee5882d",
    "synced_to_chain": true,
    "testnet": false,
    "chains": [
        "bitcoin"
    ],
    "uris": [
        "038d3eb5339f78774697448ce2414b429b41edd8f65dcc2edc33e591001476c948@:9735"
    ],
    "best_header_timestamp": "1547567804",
    "version": "0.5.1-beta commit=",
    "num_inactive_channels": 3

fluidvoice, 2019-01-15:
Some minutes later though it reverted back to wallet locked...

$ `sudo -u bitcoin lncli getnetworkinfo`
{
    "graph_diameter": 0,
    "avg_out_degree": 7.146921278254092,
    "max_out_degree": 919,
    "num_nodes": 2566,
    "num_channels": 18339,
    "total_network_capacity": "40200977035",
    "avg_channel_size": 2192103.006434375,
    "min_channel_size": "1100",
    "max_channel_size": "16777216"
}
$ `sudo -u bitcoin lncli getinfo`
[lncli] Wallet is encrypted. Please unlock using 'lncli unlock', or set password using 'lncli create' if this is the first time starting lnd.

rootzoll, 2019-01-15:
when the wallet locks thats most likely that the lnd shutdown (due to an error) and the systemd restarted the lnd service - know needed a wallet unlock again. 

The loogs above show the error for shutdown is this "2019-01-15 07:37:08.991 [ERR] SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 183, max 80]"

@fluidvoice did you git any feedback on this error from the LND slack why this is still an error or if the merge seen at the end of the thread is to fix this with the next LND release?
https://github.com/lightningnetwork/lnd/issues/2016

fluidvoice, 2019-01-16:
> @fluidvoice did you git any feedback on this error from the LND slack why this is still an error or if the merge seen at the end of the thread is to fix this with the next LND release?

Didn't try slack yet. Will do.

fluidvoice, 2019-01-16:
I'm not able to reproduce this problem consistently.  Rebooted and stopping and restarting bitcoind and lnd services is not showing the problem now.

fluidvoice, 2019-01-16:
I was told this was fixed here: https://github.com/lightningnetwork/lnd/pull/2030

rootzoll, 2019-01-16:
OK. LND latest 0.5.1-beta release (whats runnig on RaspiBlitz) was tagged on Guthub 28 Nov 2018. This fix was merged on 6 dec 2018 ... so it will be part of next LND release.

@fluidvoice thanks for investigating this. I will close the issue for now.

closed 2019-01-16 by rootzoll

fluidvoice, 2019-01-20:
> when the wallet locks thats most likely that the lnd shutdown (due to an error) and the systemd restarted the lnd service - know needed a wallet unlock again.
> 
> The loogs above show the error for shutdown is this "2019-01-15 07:37:08.991 [ERR] SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 183, max 80]"
> 
> @fluidvoice did you git any feedback on this error from the LND slack why this is still an error or if the merge seen at the end of the thread is to fix this with the next LND release?
> [lightningnetwork/lnd#2016](https://github.com/lightningnetwork/lnd/issues/2016)

Some interesting observations of this problem still is happening might give a clue to the problem.  Reboot, and the system came up with the problem.
Notice this... "**--externalip=${publicIP}**"

> *** LND SYSTEMD STATUS ***
> ● lnd.service - LND Lightning Daemon
>    Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
>    Active: activating (auto-restart) (Result: exit-code) since Sun 2019-01-20 01:59:47 PST; 46s ago
>   Process: 4001 ExecStart=/usr/local/bin/lnd --externalip=${publicIP} (code=exited, status=1/FAILURE)
>  Main PID: 4001 (code=exited, status=1/FAILURE)

Then stopped and restarted lnd with  `sudo systemctl start lnd.service`
 And it shows "**--externalip=**" and the problem persists...

> *** LND SYSTEMD STATUS ***
> ● lnd.service - LND Lightning Daemon
>    Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sun 2019-01-20 02:01:12 PST; 2min 12s ago
>  Main PID: 5057 (lnd)
>    CGroup: /system.slice/lnd.service
>            └─5057 /usr/local/bin/lnd --externalip=

Then I stop and restart lnd again but using the **pi** user
and it comes up with "**--externalip=179.125.20.32**" and now lnd comes up without the error.

> admin@thundahub:~ $ `sudo -u bitcoin systemctl start lnd.service`
> ==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
> Authentication is required to start 'lnd.service'.
> Multiple identities can be used for authentication:
>  1.  ,,, (pi)
>  2.  root
> Choose identity to authenticate as (1-2): 1
> Password: 
> ==== AUTHENTICATION COMPLETE ===
> admin@thundahub:~ $ sudo -u bitcoin systemctl status lnd.service
> ● lnd.service - LND Lightning Daemon
>    Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sun 2019-01-20 02:29:58 PST; 5s ago
>  Main PID: 2618 (lnd)
>    CGroup: /system.slice/lnd.service
>            └─2618 /usr/local/bin/lnd --externalip=179.125.20.32
> admin@thundahub:~ $ `lncli unlock`
> Input wallet password: 
> 
> lnd successfully unlocked!
> 
>