raspiblitz/raspiblitz #273: TOR active intermittently

fluidvoice, 2019-01-14:
sync finished a few minutes later and the LCD came up as normal.  So what caused the problem is still a mystery.
I notice again that the LCD is displaying TOR is active now (again) even though it was turned off before.
This is an intermittent issue where TOR sometimes is on and sometimes off after a reboot.  This needs to be investigated.

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issue_comments#issuecomment-453968958_

fluidvoice, 2019-01-14:
~ $ `sudo cat /mnt/hdd/raspiblitz.conf` 
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='0.99'
hostname=
network=bitcoin
chain=main
publicIP=
runBehindTor=off
autoUnlock=on
publicIP=179.125.XX.XXX

fluidvoice, 2019-01-14:
It looks like the logic is in the python code here:

https://github.com/rootzoll/raspiblitz/blob/d515fa1dbf59ce4e4a58ba9f31c185e6b83eb939/home.admin/infoblitzd.py#L1104

```
                for la in json.loads(_network_info)["localaddresses"]:
                    if ":" in la["address"]:
                        if la["address"] in self.ipv6_addresses:
                            self.bitcoin_local_adresses.append("[{}]:{}".format(la["address"], la["port"]))
                    elif ".onion" in la["address"]:
                        self.bitcoin_local_adresses.append("{}:{}".format(la["address"], la["port"]))

                        if self.bitcoin_onion_reachable:
                            self.tor_active = Metric("+ Tor")
                        else:
                            self.tor_active = Metric("+ Tor?")
                    else:
                        self.bitcoin_local_adresses.append("{}:{}".format(la["address"], la["port"]))
```

fluidvoice, 2019-01-14:
$ `sudo -u bitcoin bitcoin-cli getnetworkinfo`
{
  "version": 170001,
  "subversion": "/Satoshi:0.17.0.1/",
  "protocolversion": 70015,
  "localservices": "000000000000040d",
  "localrelay": true,
  "timeoffset": -1,
  "networkactive": true,
  "connections": 18,
  "networks": [
    {
      "name": "ipv4",
      "limited": false,
      "reachable": true,
      "proxy": "",
      "proxy_randomize_credentials": false
    },
    {
      "name": "ipv6",
      "limited": false,
      "reachable": true,
      "proxy": "",
      "proxy_randomize_credentials": false
    },
    {
      "name": "onion",
      "limited": true,
      "reachable": false,
      "proxy": "",
      "proxy_randomize_credentials": false
    }
  ],
  "relayfee": 0.00001000,
  "incrementalfee": 0.00001000,
  "localaddresses": [
    {
      "address": "2804:cd4:e006:5500:2514:b20f:2cc4:9b1c",
      "port": 8333,
      "score": 67
    }
  ],
  "warnings": ""
}

rootzoll, 2019-01-14:
infoblitzd.py is still in development and not active - so it can be ignored for now.

I added a fix that tries to prevent creating a double entry for publicIP - such in case.
I wil add this ticket for final testing - to check after an

rootzoll, 2019-01-18:
fresh setups show that there is now no double entry for publicIP anymore. Hope this fixes the problems. Final testing with TOR for release will be done in issue: #249

Closing this issue for now.

closed 2019-01-18 by rootzoll