raspiblitz/raspiblitz #274: libpam-systemd upgrade from 90finishSetup.sh prints warning screen

freerko, 2019-01-14:
During the `sudo apt-get upgrade -f -y --allow-change-held-packages` in `90finishSetup.sh` the package libpam-systemd gets upgraded and prints this message:

>PAM configuration
>
>One or more of the files /etc/pam.d/common-{auth,account,password,session} have been locally modified.  Please indicate whether these local changes should be overridden using the system-provided configuration.  If you decline this option, you will need to manage your system's authentication configuration by hand.                                                                                                                                                                                                                                                                                  
>
>Override local changes to /etc/pam.d/common-*?                                                                                                        
> \<Yes>                                                 \<No>

Selecting \<yes> results in the line `session required pam_limits.so` removed in both `/etc/pam.d/common-session` and `/etc/pam.d/common-session-noninteractive`.

This may confuse new users and should be prevented or documented in the Readme

rootzoll, 2019-01-14:
Such interaction is most def confusing. I decided to remove the "upgrade", because upgrade of the OS will be provided by updating the RaspiBlitz later on thru fresh SD card images (which will be based on the latest OS anyway) ... detail on backup see: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-my-raspiblitz-after-version-098

closed 2019-01-14 by rootzoll

xpsx, 2020-08-02:
should i force update the file?