raspiblitz/raspiblitz #275: Add letsencrypt certific

raumi75, 2019-01-16:


rootzoll, 2019-01-16:
As far as I understand LND is using here its created TLS.cert to respond to HTTPS on the REST API port. Lets Encrypt is an known SSL cert to the browser - that would get rid of the "Unknown Cert" warning and having to add it manually - but I am not sure if you can replace the LND TLS.cert with one from Letscrypt and it would still work. Has anyone tried this out yet?

raumi75, 2019-01-16:
I can give it a shot.

First I forwarded port 80 to my RaspiBlitz.

~~~sh
$ sudo -i
$ apt install certbot
$ ufw disable
$ certbot certonly --standalone  --email myemail@gmx.de -d mydomain.com
$ ufw enable
~~~

I ran into a permission problem when I tried to use the `*.pem` files in `lnd.conf` and my quick and dirty solution was to copy the cert files.

~~~sh 
$ cd /etc/letsencrypt/live/mydomain.com/
$ cp *.pem /mnt/hdd/lnd/
$ cd /mnt/hdd/lnd/
$ chown bitcoin:bitcoin *.pem
~~~

then I edited the /mnt/hdd/lnd/lnd.conf file and added:

~~~
tlscertpath=/mnt/hdd/lnd/fullchain.pem
tlskeypath=/mnt/hdd/lnd/privkey.pem
~~~

Now, I restarted lnd and It's *stuck at*

  Lightning Filtering Blockchain
  Progress: ?/558819

So far so good. Maybe soonish I will see it it runs.

raumi75, 2019-01-16:
Doesn't seem to work

I tried using fullchain.pem, chain.pem and cert.pem as tlscertpath. Even tried naming it letsencrypt.cert and letsencrypt.key. Every time got the same result

~~~sh
$ sudo service lnd restart
$ lncli unlock
Input wallet password: 
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: remote error: tls: handshake failure"
~~~

Anyone got any ideas?

raumi75, 2019-01-16:
It looks like the tls.key is a EC key (256 bit) whereas letsencrypt issues an rsa key (2048 bit). Maybe someone more knowledgeable could take a look at this...

frennkie, 2020-08-08:
We have included Let's Encrypt now. Nginx uses it. If LND REST API is needed then we would need to add a new nginx-site config that (e.g. from Tor) forwards to `https://localhost:8080`. Does anybody want/need this?

nyxnor, 2021-05-03:
pls close this

closed 2021-05-03 by rootzoll