raspiblitz/raspiblitz #276: Channel funding "max available" doesn't account for onchain fees

freerko, 2019-01-16:
If I have exactly 4mBTC in my LND wallet the "Funding of Channel" menu suggests `max available : 400000` which isn't true because it doesn't account for onchain transaction/miner fees. Trying to open a channel with the stated maximum of 400000 satoshis results in this error:

`[lncli] rpc error: code = Unknown desc = not enough witness outputs to create funding transaction, need 0.00403193 BTC only have 0.004 BTC  available`

Maybe for the short term it would be possible to somehow calculate the real maximum amount available to open a channel (as exactly as possible to avoid dust).

Later on the BBopenChannel.sh script should be changed to allow a non-numeric value like `all` as input since LND is working on something similar to the c-lightning `lightning-cli fundchannel <node_id> all` option, see issue https://github.com/lightningnetwork/lnd/issues/619 and PR https://github.com/lightningnetwork/lnd/pull/2198.

rootzoll, 2019-01-16:
The BBopenChannel.sh is just intented to be a quick wrapper around the lncli for a bit of easy access and learning experience ... and its true the lncli is not very user friendly. Hope LND is listening to the users.

I could spend energy in making the BBopenChannel.sh more smart ... but other wallet software is much better at that already. Have you tried to activate the RTL WebGUI (under servcies)? ... that WebGUI should replace the scripts like BBopenChannel.sh in the future anyway - the moment its not default yet. If anybody wants still to improve the BBopenChannel.sh script feel free - I will be happy to merge a Pull Request here.

geco91, 2019-02-26:
for version 1.1 think of making python scripts here to replace the basic scripts

cstenglein, 2023-06-07:
Closing since it looks like it's stale @rootzoll

closed 2023-06-07 by cstenglein