raspiblitz/raspiblitz #277: Is Dynamic DNS working?

justrossthings, 2019-01-17:
Hi There!

First off big thanks for working on this. Really fun way to get started with a node :)

I'm not sure if my dynamic DNS set up is working. When I open the main menu in ssh I see the domain name I specified but on the status screen my public address and wallet host name are still my IP address. 

See attached screenshots:
[image]
[image]

Here are my router ddns settings :
[image]

For raspiblitz set up I used my ddns domain name with no update URL.

Should I throw out my router ddns settings and instead set a freedns.afraid.org account with the direct URL through raspiblitz only? I presumed that I needed to set up dynamic dns on a router level but I'm not so sure now.

rootzoll, 2019-01-17:
Hi. the dyndns deature is brand new on the raspiblitz - so thanks for the feedback.

The info on the LCD is not yet reflecting if a dyndns is set - thats a good idea. Also the RaspiBlitz should check if the dyndns is pointing to the correct IP - if not it should display the dyndns in red.

If you dont set the dyndns update URL you need your router to do the update - thats correct.

To test if dyndns is working really you would need to call it from the outside - you could try to call in your browser: https://[YOUR-DYN-DNS]:8080 ... and see if you get an response (if it tells you 'not secure' thats an OK answere - FAIL is there is no answere at all)

I will add this issue to the version 1.0 release to add at leat the dyndns to the LCD screen and make a ping test to see it points to the correct IP.

rootzoll, 2019-01-17:
OK if dyndns is set on RaspiBlitz - it will now be displayed on LCD (with next version). If the publicIP is not the IP the dyndns points to, it will be displayed in red.

closed 2019-01-17 by rootzoll