raspiblitz/raspiblitz #278: No funds after restoring with seed.

jetlagv1, 2019-01-17:
Closed issue #205 was mine.
I didn't have time for a while but got around to restoring the raspiblitz with the use of the seed and password. It is running again but the funds are not showing. Not just the funds locked in channels but also the funds in the wallet. Everything shows 0 balance. Most of the funds where on chain and not in channels so how do I get acces to the wallet balance again? I also have a backup of the bitcoin and lnd folder from the original install if that is of any use.

rootzoll, 2019-01-17:
@jetlagv1 please check out the LND Slack for that question: https://lightningcommunity.slack.com .. on-chain funds should get recovered when you restore by seed. If they dont that is an LND error I cannot give deeper support on.

About restoring old bitcoin/lnd folders on the raspiblitz .. that could be an action of last resort. Its risky, because if you have an "old state" in the lnd backup, the channel counterparty could see this as you trying to cheat and claim the whole channel fundings. But for on-chains funds that should be of no risk to try. Please report your experience if you do.

jetlagv1, 2019-01-17:
I cannot get into that channel. I don't have an account in that workspace

rootzoll, 2019-01-17:
@jetlagv1 follow the SLACK invite link on this page to register: https://dev.lightning.community
please report back on your findings or if you try to restore bitcoin/lnd backups - this is a topic of interest. thx.

jetlagv1, 2019-01-17:
admin@DMN737:~ $ sudo journalctl -u lnd -b --no-pager -n20
-- Logs begin at Thu 2019-01-17 13:01:13 GMT, end at Thu 2019-01-17 14:04:52 GMT. --
Jan 17 14:04:44 DMN737 lnd[3030]: 2019-01-17 14:04:44.415 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605925465006342144, flag=259, update_time=2019-01-17 14:03:38 +0000 GMT) from 23.237.77.11:57343
Jan 17 14:04:44 DMN737 lnd[3030]: 2019-01-17 14:04:44.415 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=613134962741936129, flag=0, update_time=2019-01-17 14:04:28 +0000 GMT) from 23.237.77.11:57343
Jan 17 14:04:44 DMN737 lnd[3030]: 2019-01-17 14:04:44.418 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=603584604680224769, flag=1, update_time=2019-01-17 14:03:52 +0000 GMT) from 23.237.77.11:57343
Jan 17 14:04:49 DMN737 lnd[3030]: 2019-01-17 14:04:49.033 [DBG] PEER: Received Ping from 213.174.156.70:9735
Jan 17 14:04:49 DMN737 lnd[3030]: 2019-01-17 14:04:49.033 [DBG] PEER: Sending Pong to 213.174.156.70:9735
Jan 17 14:04:49 DMN737 lnd[3030]: 2019-01-17 14:04:49.036 [DBG] PEER: Sending Ping to 213.174.156.70:9735
Jan 17 14:04:49 DMN737 lnd[3030]: 2019-01-17 14:04:49.164 [DBG] PEER: Received Pong from 213.174.156.70:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.286 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=606510405197496320, flag=257, update_time=2019-01-17 14:03:48 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.317 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=600664301885915136, flag=257, update_time=2019-01-17 14:03:44 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.321 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=614473068400803840, flag=256, update_time=2019-01-17 14:03:45 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.326 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=606513703695876096, flag=257, update_time=2019-01-17 14:03:48 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.326 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=606510405197496320, flag=256, update_time=2019-01-17 14:03:48 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.438 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=613134962741936129, flag=0, update_time=2019-01-17 14:04:28 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.438 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=611426321775329281, flag=257, update_time=2019-01-17 14:03:47 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.439 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=606513703695876096, flag=256, update_time=2019-01-17 14:03:48 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.439 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=614463172726358016, flag=256, update_time=2019-01-17 14:03:45 +0000 GMT) from 98.215.0.206:9735
Jan 17 14:04:51 DMN737 lnd[3030]: 2019-01-17 14:04:51.440 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=611426321775329281), saving for reprocessing later
Jan 17 14:04:52 DMN737 lnd[3030]: 2019-01-17 14:04:52.197 [ERR] CMGR: Can't accept connection: unable to accept connection from 94.209.71.14:53002: EOF
Jan 17 14:04:52 DMN737 lnd[3030]: 2019-01-17 14:04:52.370 [DBG] RPCS: [walletbalance] Total balance=0 BTC
Jan 17 14:04:52 DMN737 lnd[3030]: 2019-01-17 14:04:52.457 [DBG] RPCS: [walletbalance] Total balance=0 BTC
admin@DMN737:~ $

jetlagv1, 2019-01-17:
Can you see something fishy here?

vhmanu, 2019-01-17:
im having the same issue. After a new setup the onchain funds are gone (LND wallet was restored with the right seed)

OfficialBigAl, 2019-01-22:
I had the same issue - Luckily was able to get backups from HDD of .Bitcoin and .LND folders and once the rebuild was completed (took 2 attempts) - Restored the .LND folder back, waited 24hrs to sync and funds started to appear in both off & on-chain wallets.
 
LND restarting was problematic, I think due to some differences in permissions.  Started manually with Bitcoin account and it works.  I feel like it's a house of cards though and could collapse at anytime!!

I get RPC errors when trying to access channels & balance info or remove funds but I'm patient and will sort that I'm sure.  

At least I can see funds are there... Phew!

rootzoll, 2019-01-22:
@OfficialBigAl if you restore LND folder manually the macaroons/TLS are in mismatch with admin user. In the 1.0 release there will be a option to RESET those macaroons/TLS in the EXPORT menu point. And yes - using lightning at this point in time is still #reckless. One idea of the RaspiBlitz is to have a early home-run lightning node to find all the weakspots to get them fixed and let the LND developers know. So please report that you also were not able to recover your funds in the normal way over at the LND Dev Slack, so LND devs put priority to fix this: https://dev.lightning.community

IvanThinking, 2019-02-13:
Same for me. My old v.96 raspiblitz blew up somehow so I started over. Channels have been closed. No opportunity to release funds with the raspiblitz, as it was all kinds of messed up, but since no open channels I figured no big deal. Started fresh, and used the recovery seed when asked... no funds present now.

rootzoll, 2019-02-13:
Did anybody get a response on that from the LND Slack channel? They are doing a great job, but this is really something that needs to be improved and cannot be the status quo. https://dev.lightning.community

rootzoll, 2019-02-13:
Posted to LND Salck with hope to get some more details on this issue.

githorray, 2019-02-13:
https://github.com/lightningnetwork/lnd/issues/900

molxyz, 2019-02-13:
`lncli unlock --recovery_window=250000` will surely help.

mr-dakota, 2019-02-13:
Just to chip in here... I've had a similar issue and have used recovery window up to 5000 to no avail.
Only issues on LND github seem to point to that as a solution, but I'd be very surprised if any Blitz users have got through more addresses than the default recovery window covers.

Hopefully there is some attention in Slack soon, as although it's not large amount of satoshis for me, recovery is certainly an important issue!!

IvanThinking, 2019-02-13:
> 
> 
> `lncli unlock --recovery_window=250000` will surely help.

It's a little more complicated than just running that command, but the link above should walk the user through the issue.

rootzoll, 2019-02-14:
I updated the FAQ doc to recover coins from seed, to set look-ahead/scan window to 250000
see: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

jetlagv1, 2019-02-15:
Ok. I tried this but still no funds...... The slack didn't get me any further than increasing the look ahead time and replacing the lnd directory with the backup I have. Both to no result. The look ahead time recovers the node without funds, the lnd backup won't let the node start......

rootzoll, 2019-02-15:
@jetlagv1 To make sure we have a "cannot recover" not a "funds stolen" scenario ... can you check if you are seeing any transaction history by lnd after the recover. Can you post the funding transaction, so we can investigate on a blockchainexplorer the flow of on-chain funding.

@jetlagv1 Did you use passwordD to encrypt the seed words? Have you tried to recover the funds in another wallet that accepts BIP39 seed words like electrum? 

If its a "cannot recover" scenario there should be more support/feedback from the lnd slack on this.

jetlagv1, 2019-02-15:
I don't know anymore which transaction was the funding transaction for the on chain wallet on my raspiblitz. The details of my raspiblitz are here: https://1ml.com/node/0217b5c4691c811b14a9221921edf585a5259963523049060eedee2fd81b5c171f

As for the seed, yes I did use a password on the seed and I used this same password when trying to recover. As you mentioned yourself in my other closed issue https://github.com/rootzoll/raspiblitz/issues/205 the seed does not seem to work in a bip39 wallet. It does work when recovering the node because it re-generates the node with the same address atc..

rootzoll, 2019-02-15:
This is really something to look into more deeply - will mark this for after the weekend. Any help on the community in investigating this is much welcome.

jetlagv1, 2019-02-15:
I consider the funds almost lost. I can give you the seed and I can even send you the backup I have of the lnd directory or only the wallet.db and channels.db files if that helps.
Maybe you succeed in recovering the funds and take a donation and send me the leftovers 😎

mr-dakota, 2019-02-15:
I've got this weekend set aside to look into this thoroughly myself. I've followed all of my on-chain funds from closed channels and can see everything is unspent in addresses that i'm certain my Blitz LND should have keys to, which is somewhat reassuring. My funds have been "lost" for a few months now, but I wasn't sure LND had implemented recovery fully, so never looked into it thoroughly.

I've got a fresh BTC blockchain downloaded on a separate device so will hopefully be able to create a new LND instance and run a few tests to see if I can get to the bottom of the issues a few of us have had!

rootzoll, 2019-02-15:
@jetlagv1 maybe share your seed & lnd data with @mr-dakota :)

mr-dakota, 2019-02-18:
I've managed to successfully get my funds back, although I'm not certain what element actually worked. I started the node with a larger recovery-window, but there were still no funds initially, which was a little annoying. 

I then set about generating a series of new addresses to try and hit some old ones to find the funds. I sent fresh btc to a couple of the new addresses too but still nothing, so I sent those new funds away from LND with the intention to spin up another Blitz with an even larger recovery window. I then shut down the original node, and rebooted.

The LND rescan as the Blitz rebooted actually managed to successfully find the old funds, so I can confirm success in restoring from seed, albeit with a few extra random steps!!

rootzoll, 2019-02-18:
@mr-dakota so much thank to going into so much deep research - it would be great if you can discuss this findings over at LND Dev Slack to get more details what might happens in the background and why.

Now that you see this funds back on the RaspiBlitz can you check transaction history to see if there might have been some transactions (closing channels, etc.) waiting a long time for confirmation or something like that.

jetlagv1, 2019-02-18:
@mr-dakota Thanks for the post on your successful recovery. I will try to copy your solution as soon as I have the time and update here on the outcome!

mr-dakota, 2019-02-18:
@rootzoll Not a problem! Recovery is clearly an important element, and I actually had "lost" funds, so was a good excuse to investigate! I plan to discuss a bit further on the LND Slack, and perhaps submit an issue on the LND github, as although funds can be recovered, it's far from seamless and it seems that a fair few people have had troubles with it.

All of the previous channel close transactions were successful, and confirmed way back at the end 2018 in my case, so it seemed to purely a case of LND not scanning addresses thoroughly enough.

Thanks for all your work on the Raspiblitz by the way @rootzoll - I wouldn't be using Lightning without this great project!

jetlagv1, 2019-02-19:
I've tried to copy the workaround as discribed by @mr-dakota but untill now without the desired result.
What I tried:
1. Flashed a sd card with the latest 1.0 version of raspiblitz (nice RTL interface Btw!)
2. Initialized the install using the LN seed and password. 
3. Let the node complete the sync.
4. Send some Satoshis to the on-chain node wallet.
5. Wait for confirmation and reboot. No previous funds recovered. 
6. Send the Satoshis back out of the LN wallet. 
7. Wait for confirmation and reboot again. Still no previous funds recovered...

mr-dakota, 2019-02-19:
@jetlagv1 irritating that this process isn't straightforward, particularly with recovery of on-chain funds which in theory should be simple.

Did you pass the 'recovery_window=5000' argument in with 'lncli create' when setting up LND?

If you run `lncli listchaintxns`, are you only seeing the two most recent on-chain transactions?

Looking at your node on 1ML, there appear to be active channels from September? Is that correct? Are you still waiting on channel partners to close those?

jetlagv1, 2019-02-19:
When installing the new node I entered 500000 when asked for the look ahead number. I didn't run lncli create by hand. The setup is scripted from the raspiblitz image. 
I only see the last recent transactions when running [code]lncli listchaintxns[/code].
I don't know I thr channel partners closed the channels or not but it looks like some didn't.

geco91, 2019-02-26:
I would recommend to try to switch the LND wallet creation to a python script in 1.1
in the beginning the user should be asked if "new wallet" or "restore wallet" 
on restore the 500000 can be set then by the py script - also entering of word seeds can be made more easy

molxyz, 2019-02-27:
> > `lncli unlock --recovery_window=250000` will surely help.
> 
> It's a little more complicated than just running that command, but the link above should walk the user through the issue.

I was not meaning to say this is all you have to do.  And that's right there's more to it. What that line can help is if the first attempt to run recovery results in a crash or does not bring up the funds, one can try again with the same LND instance and run that line when LND asks for "create or unlock", and by increasing the `look-ahead` one can have a better chance to see the funds back. 

I was commenting this post explained by Conner:  https://github.com/lightningnetwork/lnd/issues/900#issuecomment-443957945

molxyz, 2019-02-27:
> When installing the new node I entered 500000 when asked for the look ahead number. I didn't run lncli create by hand. The setup is scripted from the raspiblitz image.
> I only see the last recent [...]

@jetlagv1 Have you tried the recovery by creating a lnd node on a laptop, not on a pi, not using any script?  I'm wondering you'll have a better chance by just creating a clean lnd node like that?  

Also, I've tested by increasing `look-ahead` to a large number (250000) it helps to recover funds. As a matter of fact I helped someone with this on the slack a few days ago and he got his funds back, rewarded me with a channel and nice push_amt... 😄  I can show you our convo if you're on slack. So I'm not sure why your look-ahead number didn't work for you. There must be something else.  

And from my convo with that user, I asked Conner to increase the `look-ahead` address number in the code, which he said he was also thinking of doing it... Here's the PR:  https://github.com/lightningnetwork/lnd/pull/2691

Would love to help you get this figured out.

jetlagv1, 2019-02-27:
I'm on vacation atm. I would really appreciate it if you would try to help me recover the funds. I will get in contact once I'm back home 😎

openoms, 2019-02-27:
Had some interesting experience time today, when. I tried to restore my on-chain funds from seed. Electrum did not show anything from the 24 words +/- 25th word (passphrase). `lnd create` did not bring back the funds either. Had some backups of the LND folder, but have been done on-the-go (not shutting down LND and copied with scp). 
Using both the wallet.db and channel.db got lnd exiting with various errors. 
FinalIy succeeded with a new wallet (deleting the wallet.db and running `lnd create`) and overwriting the wallet.db with one from the backup. Got my on-chain funds and most of the in-channel funds are back as well due to forced closures by the other side.

The takeaway is that it is absolutely essential to have a backup of the LND directory, most importantly the wallet.db and the channel.db (the latter is only useful if done after stopping LND and contains the most recent states).

CommanderPoe, 2019-02-27:
> I had the same issue - Luckily was able to get backups from HDD of .Bitcoin and .LND folders and once the rebuild was completed (took 2 attempts) - Restored the .LND folder back, waited 24hrs to [...]

Im starting to read this thread now since Im in the same position of many of you here, restored my lightning wallet using the correct seed now after few hours i dont see any balance... I was planning to recover as well old channels since i have my old .lnd + wallet.dat that belongs to .bitcoin (dont know if that matters at all to recover my funds + channels ) 

I wanted to publiccly ask if its pertinent to recover my old state after 5 days offline? I'm 95% sure that the state is the same, tho i wanna take the risks of trying to recover those channels. So i was planning you guys can shed some light in here, as far as im concerned i just have to completely replace the .lnd forlder owned by user bitcoin and the wallet.dat of the bitcoin directory? Am i missing something? Is it too crazy trying to recover my channel states after 5 days offline?

molxyz, 2019-02-28:
@CommanderPoe You don't need to restore the wallet using the seed after being offline for a while. All you need is to start your lnd node.  The only time you use the seed is if your node can't restart due to files being corrupted or you have lost the node by some other reason.

CommanderPoe, 2019-02-28:
I did not make myself clear... I bricked my last sd card, had to start fresh! The rest stands, more or less the same issue of you all, no funds after recovering my wallet using the seed so i domt even know if its smart at this point to try recovering those channels.

openoms, 2019-02-28:
@CommanderPoe there are some misunderstandings here. The wallet og the RaspiBlitz is managed by LND, not bitcoind. The channel and wallet information files are on the HDD in /mnt/hdd/lnd. If you only bricked your SD, just burn the raapiblitz inage to a new card and the LND infos, including the channels should be recovered from the HDD during the setup. Make a copy of you  /mnt/hdd/lnd directory first in any case.

CommanderPoe, 2019-02-28:
> @CommanderPoe there are some misunderstandings here. The wallet og the RaspiBlitz is managed by LND, not bitcoind. The channel and wallet information files are on the HDD in /mnt/hdd/lnd. If you only bricked your SD, just burn the raspiblitz image to a new card and start with that. The LND wallet, including the channels should be recovered from the HDD during the setup. Make a copy of you /mnt/hdd/lnd directory first in any case.

I wasnt usung the Raspiblitz before, i was working with the Raspibolt. All in alll, lets say that i started from scratch but still cant recover my on chain  funds or channel state, the copy i have from my old lnd is from the raspi bolt configuration which is more or less the same.

openoms, 2019-02-28:
Ok, so do you have an lnd directory in the root of your HDD? You need to look for the wallet.db and channel.db buried in there to recover. You can run from the terminal: `cd \` and `sudo find - name wallet.db` to look for it.

CommanderPoe, 2019-02-28:
> Ok, so do you have an lnd directory in the root of your HDD? 
>
Yes, i do an old lnd folder from my latest raspibolt node ( copyed it 5 days ago)

>You need to look for the wallet.db and channel.db buried in there to recover. You can run from the terminal: `cd \` and `sudo find -name wallet.db` to look for it.
When you mean wallet.db are you refering to the one on bitcoin folder, lightning directory or both? Cuz as far as im concerned i need to copy and replace both right?

openoms, 2019-02-28:
The wallet of bitcoind is disabled, you cannot recover anything in the bitcoin dir. 
All you need is in the LND directory. I would suggest to make a backup copy of your old lnd dir and put it in the root of the HDD. Same level where the bitcoin dir is with the blockchain data. Make sure it is owned by the bitcoin user running: `sudo chown - R bitcoin:bitcoin /mnt/hdd/lnd`.
You can delete everything else from the hdd (except the lnd and bitcoin directories). When this is done, burn a new raspiblitz SDcard and boot up with it.
Your old lnd state (wallet with onchain funds and the remaining channels) should be recovered during the setup process.

openoms, 2019-02-28:
this is the same thing. but explained by rootzoll: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#2-lnd-channel-state-backup

OfficialBigAl, 2019-03-01:
Thanks for all your contributions on this... I spoke too soon.  On Jan22 all funds were correct but I went away for 3 weeks and on return, the PI was sickly.  No funds showing and refusing to co-operate.  Wouldn't even show menu, just looping LND service and falling over.

Rebuilt it over a couple of days - and got the PI to a stable state.
Tried the re-scan index_window=250000 - Nothing showed up.
Added funds and connect peers and tried opening previously open channels - Still nothing.
At this point RTL failed a channel and I think it screwed the wallet.db  as I entered a failing LND state asking for unlocking PASSWORD C all the time.  Ah Crap, I thought.
Copied that wallet before deleting and restoring the one from prior to the 22nd Jan.
And set it off again, rescanning with 250000 window.
According to the logs funds have been discovered the missing funds !!! Yey!!!  BUT....
LND has now all caught up and is hitting something and looping again :(

Since I royally screwed this up, I placed a issue with the LND team to see if they can help
https://github.com/lightningnetwork/lnd/issues/2722
The Saga continues...

OfficialBigAl, 2019-03-06:
... Drastic times call for Drastic measures.
  
Backed-up and built a new V1.0 on a different SD card.
Run through setup and entered old seed and passwords - All Ok so far...
Found HDD bitcoin chain and LND wallet - All Ok.
Allowed the setup to completely sync before messing around.  
Synced Blockchain and LND chains - Ok.
0 Channels & Sats showing....Waited overnight.

In the morning it was showing 1 channel and a few Sats On and Off Chain but not as much as expected.
The Channel that was pending eventually closed after FORCING it, and Sats returned.

Wasn't able to use the option to "Move fund Off-Chain" - Parameter missing...so had to do it manually.
1ML is still showing all my Active channels but have no idea who or how to resolve this.

Can anyone advise?

OfficialBigAl, 2019-03-11:
... The Sage Continues ... Waited 4 days and no more channels recovered or funds appeared.

Decided to try something radical, I had a copy of both .Bitcoin and .LND folders from original V0.98 from back in Nov/Dec.  So thought "What the hell!"  
Stopped both bitcoind & lnd services and made a backup of entire PI, if I need to return to a stable point, and then copied over the .conf, wallet and macaroon files replacing the new ones.
Gave it a reboot. and watched the log files.
Bitcoind didn't have a problem and synced - Ok.
LND sat and thought for a while and whoosh! started processing all the history transactions and finding channels and funds!  

Took 2 days to finally catch up...but then wallet was 0, Zero - Zitlch.  :(  Bugger...

But then ... Of course, all the Channels are still live!!
Used `lncli pendingchannels` and there they all were, waiting.

Used the Services->Admin/Macaroon function to rectify the permissions issue and proceeded to force close the channels.

2 Days later, I can say 50% of the channels have closed and the fund are safely back in the wallet and was able to move away.

Just waiting for the remainder to close out. 👍

openoms, 2019-03-13:
Interesting detail here: https://twitter.com/ShrimpCapital/status/1105664775822798848
The LND backup seed is having a random number added, which explains why funds are not recovered from the seed. This will be chnaged soon, but for now the only reliable backup is the latest state channel.db and wallet.db (or the lnd directory to keep it simple).

Himbeergeld, 2019-03-15:
I confirm that it helped to copy the lnd directory. 
Background: Initially I started the RaspiBlitz project with a small HDD (old HDD) and successfully funded my LND wallet. Then I decided to move to a bigger SSD (new SSD) instead, so I installed the RaspiBlitz again from scratch on the new SSD, recovered my LND seed but the wallet was showing a "0 sat" balance with the new SSD. Increasing the recovery window (lncli unlock --recovery_window=250000) did not help for me. But connecting my old HDD and copying the lnd directory to the new SSD actually helped. Below is a step-by-step guide of what I did to fix the empty LND wallet issue, hope it helps people who love this project but are Linux beginners (like me):
- started the RaspiBlitz with both, the new SSD (/mnt/hdd) and the old HDD connected
- lsblk  // was showing the old HDD as sdb1 but yet without a mountpoint
- sudo mount /dev/sdb1 /mnt/hdd2  // mounted my old HDD as /mnt/hdd2
- lsblk   // was now showing the old HDD with a mountpoint /mnt/hdd2
- sudo rsync -a --delete /mnt/hdd2/lnd/ /mnt/hdd/lnd/   // copied the lnd folder of the old HDD to the new SSD
- sudo shutdown -r now   // reboot the RaspiBlitz
- disconnected the old HDD
After the reboot (on the new SSD) my LND wallet had its initial balance again.
@rootzoll Could you script this in a separate menu option "LND wallet recovery"? Also, would it make sense that the RaspiBlitz automatically creates a backup of the lnd directory on the microSD card? If yes, the "LND wallet recovery" could have two sub-menus: "1. Recover from external backup" and "2. Recover from last microSD backup". Thank you anyway for this beautiful RaspiBlitz project, it is actually my first contact with e-cash and a great project to learn this technology.

rootzoll, 2019-03-19:
@Himbeergeld version v1.1 has now the script `/home/admin/config.scripts/lnd.rescue.sh` that helps on backup and replay LND data from one Blitz to another. See FAQ: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#1-recover-lnd-data

coinicarus, 2019-05-04:
I am in the situation where i restored my seed, i had  lost the initial HDD i did not have a backup of the /lnd or anything else (yup not just reckless, but foolish) i would expect anything in my lnd channel at the time of the failure to be lost, but shouldnt the BTC from the bitcoin wallet return with the seed restored? how long would it normally take to show up?

escapedcat, 2019-06-08:
I might have ran into the same issue. [I had to do a re-install of the whole raspiblitz setup](https://github.com/rootzoll/raspiblitz/issues/626) afetr some issues. 
I didn't have a any channels open. After the new install I chose [the ONLYSEED option](https://github.com/rootzoll/raspiblitz#only-seed-only-seed-word-list-fallback). 

`lncli walletbalance` shows:
```
{
    "total_balance": "0",
    "confirmed_balance": "0",
    "unconfirmed_balance": "0"
}
```
[Readme says](https://github.com/rootzoll/raspiblitz#only-seed-only-seed-word-list-fallback): 
> Recover of channel funds is not very likely in this scenario.

There are some sats on that wallet. Should I refund with some more sats and everything will re-appear? Or should I just have started with a new wallet? Or what are other options I could try?

Sorry for tagging along on this issue. Let me know if I should ask somewhere else.

openoms, 2019-06-08:
@escapedcat Check out the conversation here: https://github.com/rootzoll/raspiblitz/issues/632

TL;DR: restore with the desktop Lightning App to get the on-chain funds back. Pay attention that the autopilot is on by default in the app, so switch it off in settings asap.

escapedcat, 2019-06-08:
@openoms thx! I tried it but also getting the [`invalid passphrase`-error](https://github.com/rootzoll/raspiblitz/issues/632#issuecomment-496689615).  
I'll watch #632 and see if there's a solution in the future.  
You think it's save to continue using that wallet and add funds to it or better start with a new one?

openoms, 2019-06-11:
> @openoms thx! I tried it but also getting the [`invalid passphrase`-error](https://github.com/rootzoll/raspiblitz/issues/632#issuecomment-496689615).
> I'll watch #632 and see if there's a solution in [...]

Don`t use it further as it is demonstrated that you cannot restore it! Are you sure that your seed and passphrase is recorded correctly?

I would cash out, make a final backup for future research and start a new one.

escapedcat, 2019-06-11:
Problem is that lightning-app isn't asking for a passphrase only for the seed.  
Seed ist correct and address displayed by raspiblitz is showing balance on a blockchain-explorer.  
Alright. I'll try a cashout and see what happens.  
Thx for your feedback!

escapedcat, 2019-06-11:
```
******************************
Sweep all possible Funds
******************************
lncli --chain=bitcoin --network=mainnet sendcoins --sweepall --addr=[MY_ADDRESS] --conf_target=6

FAIL: [lncli] rpc error: code = Unknown desc = transaction has no inputs

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
FAIL --> Was not able to send transaction (see error above)
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Press ENTER to return to main menu.
```

`CASHOUT` didn't work. I guess because LN never saw/found the balance on this wallet after using the `SEEDONLY` recover option.

I assume my sats are save though. I'll try to recover it with another wallet and the seed. Maybe I try [exodus](https://www.exodus.io/).

openoms, 2019-06-11:
@escapedcat you can only cashout if there is a  balance onchain . Also need to force-close all the channels first to get the funds there (if there are any channels open) .
Don't use Exodus it is a closed source wallet. Also LND works with a different seed.
Try this app to restore from seed: https://github.com/lightninglabs/lightning-app/releases
Switch off the autopilot in the settings once opened to keep your funds onchain.

escapedcat, 2019-06-11:
@openoms alright. I didn't know it's a different seed. [Can't use lightning-app for now because it can't handle passphrases (yet)](https://github.com/rootzoll/raspiblitz/issues/632) if I understand correctly.  
I might just create a new LN wallet and wait till the old one can be recovered. It's just a couple of sats.

nyxnor, 2021-04-18:
pls close this. Rootzoll added recover description in the README and FAQ and Openoms already explained solved the issue here.

[2 more replies]

closed 2021-04-18 by openoms