raspiblitz/raspiblitz #279: Running raspiblitz behind vpn

rolandbitcoinspot, 2019-01-17:
Hi there.
I have a raspiblitz and it is working fine. however: i would like to run it behind a vpn so i wont disclose my home ip address. Is it possible to run raspiblitz behind a vpn and does anyone have experience with this ?

Thanks
Keep up the good work!

Roland.

rootzoll, 2019-01-17:
A VPN solution is not part of RaspiBlitz yet, but to hide your home IP you can at the moment activate TOR (services > TOR).

rolandbitcoinspot, 2019-01-18:
But if i activate TOR, my node wont be seen in the "regular" LN explorers wont it ? people will have to specifically connect to it trough tor?

rootzoll, 2019-01-18:
Thats correct, if you want to be a routing node, TOR is not a perfect option. TOR is a good (and cheap) option if you want a personal Lightnig Node to send/receive and not publish your personal IP.

A VPN feature for Routing Nodes is missing. Contributions welcome.

fluidvoice, 2019-01-18:
bumped into this article on how to run your own VPN
[Secure & Ad-free Internet Anywhere With Streisand and Pi Hole](https://ifelse.io/2019/01/12/secure-ad-free-internet-anywhere-with-streisand-and-pi-hole/)

rolandbitcoinspot, 2019-01-21:
thanks for all the replies. i will keep my LN node on my VPS for now.
I also had a bunch of replies on a reddit thread, be sure to check those out.
https://www.reddit.com/r/lightningnetwork/comments/ahoxwr/running_lightning_node_behind_vpn/
i will close this for now.

closed 2019-01-21 by rolandbitcoinspot