raspiblitz/raspiblitz #281: Imposible to download reliably Raspiblitz SD-image from http with a poor Internet connectionn

thelwyn, 2019-01-19:
Downloading the image from China has been a PITA, the http download would always fail at some point, classic for a large download file here. I had to ask a friend back in Europe to upload it for me to Baidu Cloud (the Chinese cloud) and download it from there.

I saw another issue mentioning a torrent but only found torrents for the blockchain (If I understood correctly), without more information and non very informative filename ('update' in many of them for example).

-Is there a torrent file of Raspiblitz image to seed?
-If not, what about (re?)creating a torrent in this github for the poor souls with terrible unstable Internet connections? (I'm willing to seed obviously).

fluidvoice, 2019-01-19:
I wonder if it would work to download and run the script that builds the SD card image:
https://github.com/rootzoll/raspiblitz/blob/master/build.sdcard/raspbianStretchDesktop.sh
It must be run from the booted Pi/Raspbian.

fluidvoice, 2019-01-20:
If the server supports resuming (continuing) failed downloads, which is **should**, you can use the linux command `wget` with the --continue option.  See the man pages for wget for information:

$ `man wget |grep -A4 '\-\-continue'`

>        --continue
>            Continue getting a partially-downloaded file.  This is useful when
>            you want to finish up a download started by a previous instance of
>            Wget, or by another program.  For instance:
> 
> --
>            --start-pos has higher precedence over --continue.  When
>            --start-pos and --continue are both specified, wget will emit a
>            warning then proceed as if --continue was absent.
> 
>            Server support for continued download is required, otherwise
>            --start-pos cannot help.  See -c for details.
>

fluidvoice, 2019-01-20:
This page has some other good info on [How to Download Files on the Shell With Wget](wget https://www.howtoforge.com/linux-basics-how-to-download-files-on-the-shell-with-wget)
Of course you can do `man wget`

rootzoll, 2019-01-20:
@thelwyn version 1.0 release of the RaspiBlitz is comming closer. for such main releases I am much happy to provide a torrent of the image (and publishing in github repo) if this helps.

thelwyn, 2019-01-21:
@fluidvoice I was on Windows when downloading the image. Didn't think of switching to Linux to wget it at that time, I will think of it next time! I also tried your other suggestion and DL Raspbian by torrent + build the image using the script. It worked, after turning on a VPN (I guess something was blocked again by the Great Chinese Firewall...)

@rootzoll that would be great! I'll seed.

rootzoll, 2019-02-18:
@thelwyn raspiblitz version 1.0 torrent file is up: https://github.com/rootzoll/raspiblitz/raw/master/raspiblitz-v1.0-2018-02-18.torrent ... help on seeding is welcome :)

Closing then issue for now.

closed 2019-02-18 by rootzoll

thelwyn, 2019-02-19:
Yay! Downloading now, seeding then :)