raspiblitz/raspiblitz #283: ReadVarBytes problem possible lnd service issue?

fluidvoice, 2019-01-20:
> when the wallet locks thats most likely that the lnd shutdown (due to an error) and the systemd restarted the lnd service - know needed a wallet unlock again.
> 
> The loogs above show the error for shutdown is this "2019-01-15 07:37:08.991 [ERR] SRVR: unable to create server: can't create router: ReadVarBytes: sig is larger than the max allowed size [count 183, max 80]"
> 
> @fluidvoice did you git any feedback on this error from the LND slack why this is still an error or if the merge seen at the end of the thread is to fix this with the next LND release?
> [lightningnetwork/lnd#2016](https://github.com/lightningnetwork/lnd/issues/2016)

Some interesting observations of this problem still is happening might give a clue to the problem.  Reboot, and the system came up with the problem.
Notice this... "**--externalip=${publicIP}**"

> *** LND SYSTEMD STATUS ***
> ● lnd.service - LND Lightning Daemon
>    Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
>    Active: activating (auto-restart) (Result: exit-code) since Sun 2019-01-20 01:59:47 PST; 46s ago
>   Process: 4001 ExecStart=/usr/local/bin/lnd --externalip=${publicIP} (code=exited, status=1/FAILURE)
>  Main PID: 4001 (code=exited, status=1/FAILURE)

Then stopped and restarted lnd with  `sudo systemctl start lnd.service`
 And it shows "**--externalip=**" and the problem persists...

> *** LND SYSTEMD STATUS ***
> ● lnd.service - LND Lightning Daemon
>    Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sun 2019-01-20 02:01:12 PST; 2min 12s ago
>  Main PID: 5057 (lnd)
>    CGroup: /system.slice/lnd.service
>            └─5057 /usr/local/bin/lnd --externalip=

Then I stop and restart lnd again but using the **pi** user
and it comes up with "**--externalip=179.125.20.32**" and now lnd comes up without the error.

> admin@thundahub:~ $ `sudo -u bitcoin systemctl start lnd.service`
> ==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
> Authentication is required to start 'lnd.service'.
> Multiple identities can be used for authentication:
>  1.  ,,, (pi)
>  2.  root
> Choose identity to authenticate as (1-2): 1
> Password: 
> ==== AUTHENTICATION COMPLETE ===
> admin@thundahub:~ $ sudo -u bitcoin systemctl status lnd.service
> ● lnd.service - LND Lightning Daemon
>    Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sun 2019-01-20 02:29:58 PST; 5s ago
>  Main PID: 2618 (lnd)
>    CGroup: /system.slice/lnd.service
>            └─2618 /usr/local/bin/lnd --externalip=179.125.20.32
> admin@thundahub:~ $ `lncli unlock`
> Input wallet password: 
> 
> lnd successfully unlocked!
> 
>

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issue_comments#issuecomment-455855309_

rootzoll, 2019-01-21:
I made at least sure that if the public IP cannot be determined, that it will not be stored - reduce the "empty" problem.

geco91, 2019-02-26:
seems that this problem got fixed. no further reports on that since v1.0 release.
recommend this issue for close

closed 2019-02-26 by rootzoll