raspiblitz/raspiblitz #286: Final Testing of Version 1.0 RC3

rootzoll, 2019-01-22:
Version 1.0 release is getting closer :D Help is appreciate to test-run the (hopefully) last Release Candidate #3 version 0.99-2019-01-22 and see if you find still some critical bugs:

**Download SD-Card image:**

https://wiki.fulmo.org/downloads/raspiblitz-0.99-2019-01-22.img.gz 
sha256: 91ef1e5b4e55a5a90e9faf094756461be841ba02591bb41ccf481755977b191b 

or build your own SD card image: https://github.com/rootzoll/raspiblitz#build-the-sd-card-image

**When UPDATING from old RaspiBlitz version:** 
Version 0.98 and old 0.99-RC1/2: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-my-raspiblitz-after-version-098
Version older then 0.98: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-my-raspiblitz-before-version-098

openoms, 2019-01-22:
This might be relevant here: https://github.com/rootzoll/raspiblitz/issues/244#issuecomment-456573152
I have tested the newest script to build the SDcard on DietPi on an ODroid HC1, ran into this error:
```
Failed to connect to bus: No such file or directory
1548193320
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
```
Hope it is not an issue on the rPi.

fluidvoice, 2019-01-24:
Don't forget you can use the `wget` command in an SSH session to download directly on your RaspiBlitz:
`sudo mkdir /mnt/hdd/download`
`cd /mnt/hdd/download`
`sudo wget --continue --show-progress https://wiki.fulmo.org/downloads/raspiblitz-0.99-2019-01-22.img.gz`
Next up, how to burn/write the SD card using only your Pi/Raspiblitz (if it's possible?) ;-)

vmavromatis, 2019-02-04:
I'm testing this and my blockchain is now syncing the past 3 days for the remaining blocks (last updated in Sep 2018). BUT it's syncing VERY SLOWLY, only 0.1% or 0.2% progress made these 3 days, any idea why? Should I just leave it and wait? I don't know if it's worth plugging the external HDD to my PC and completing the sync (if that's even possible)?

```
admin@vasilightning:~ $ ./XXdebugLogs.sh 

*** RASPIBLITZ LOGS ***
blitzversion: 0.99
chainnetwork: bitcoin / main
 18:59:32 up 19:52,  2 users,  load average: 2.98, 3.05, 3.06

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-02-04 17:38:30 GMT; 1h 21min ago
  Process: 4978 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 4979 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─4979 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Feb 04 17:38:30 vasilightning systemd[1]: Starting Bitcoin daemon...
Feb 04 17:38:30 vasilightning systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-02-04T18:55:33Z UpdateTip: new best=0000000000000000000f0de518a4c57ab59733d746fdbc0d1bb7379853a33dbf height=543465 version=0x20000000 log2_work=89.753846 tx=344876622 date='2018-09-28T15:58:16Z' progress=0.927957 cache=112.9MiB(990046txo)
2019-02-04T18:55:40Z UpdateTip: new best=0000000000000000000bf9b8d5a622b6e118106a230553304edad3b4950675b6 height=543466 version=0x20000000 log2_work=89.753889 tx=344879377 date='2018-09-28T16:05:42Z' progress=0.927960 cache=113.2MiB(993057txo)
2019-02-04T18:55:53Z UpdateTip: new best=00000000000000000011d2558c264dc6ba3242ad9cbba8d8ee4009ad0214f1bb height=543467 version=0x20000000 log2_work=89.753931 tx=344881916 date='2018-09-28T16:07:08Z' progress=0.927961 cache=113.7MiB(997710txo)
2019-02-04T18:56:02Z UpdateTip: new best=00000000000000000020215d88d2a7eac529500f569f3fb017bef2f0d9d52c4d height=543468 version=0x20000000 log2_work=89.753974 tx=344883925 date='2018-09-28T16:16:53Z' progress=0.927965 cache=113.9MiB(999463txo)
2019-02-04T18:56:14Z UpdateTip: new best=000000000000000000037572de6d475462a78bec6eea1ec39a27f5ba4434a849 height=543469 version=0x20000000 log2_work=89.754016 tx=344886498 date='2018-09-28T16:28:02Z' progress=0.927969 cache=114.3MiB(1003318txo)
2019-02-04T18:56:32Z UpdateTip: new best=000000000000000000012d629c578de4a91f077c8e3f48343a7b24b992818879 height=543470 version=0x20000000 log2_work=89.754059 tx=344889356 date='2018-09-28T16:30:01Z' progress=0.927970 cache=114.8MiB(1007426txo)
2019-02-04T18:56:39Z UpdateTip: new best=0000000000000000001c8532757ea43a0c1d50f401dd71df1ab1901da42d15bd height=543471 version=0x20000000 log2_work=89.754101 tx=344891987 date='2018-09-28T16:41:59Z' progress=0.927975 cache=115.0MiB(1009647txo)
2019-02-04T18:56:51Z UpdateTip: new best=0000000000000000001e0ff7c4c4aca44fc91a40210a96fc4f06bd6799757e2d height=543472 version=0x20000000 log2_work=89.754144 tx=344894646 date='2018-09-28T16:48:02Z' progress=0.927978 cache=115.4MiB(1013076txo)
2019-02-04T18:57:08Z UpdateTip: new best=000000000000000000229505c590d27b394e9919c6e46b7d203ddaa397c70a20 height=543473 version=0x20000000 log2_work=89.754186 tx=344897174 date='2018-09-28T16:54:32Z' progress=0.927980 cache=115.9MiB(1017973txo)
2019-02-04T18:57:20Z UpdateTip: new best=00000000000000000006025758e452ba99a8569c9b352f5757a78ff9dcdb7fc9 height=543474 version=0x20000000 log2_work=89.754229 tx=344899571 date='2018-09-28T16:57:01Z' progress=0.927982 cache=116.2MiB(1020613txo)
2019-02-04T18:57:31Z UpdateTip: new best=0000000000000000001948f266561983bc4893a83e637d01345082c90f31b8f5 height=543475 version=0x20000000 log2_work=89.754271 tx=344900659 date='2018-09-28T17:01:02Z' progress=0.927983 cache=115.7MiB(1016349txo)
2019-02-04T18:57:40Z UpdateTip: new best=0000000000000000000e229cde9f957539ce4335518327c3ae970c54d979ab6b height=543476 version=0x20000000 log2_work=89.754313 tx=344903404 date='2018-09-28T17:14:22Z' progress=0.927989 cache=116.0MiB(1019518txo)
2019-02-04T18:57:56Z UpdateTip: new best=0000000000000000000c206339eac3cf775143f132bc69e3114605cb66d114f9 height=543477 version=0x20000000 log2_work=89.754356 tx=344905009 date='2018-09-28T17:20:07Z' progress=0.927991 cache=116.4MiB(1022636txo)
2019-02-04T18:58:10Z UpdateTip: new best=00000000000000000008e862661ee7fd2def47c91437076d01ca3e6d940a4bf0 height=543478 version=0x20000000 log2_work=89.754398 tx=344906581 date='2018-09-28T17:25:47Z' progress=0.927993 cache=116.7MiB(1025426txo)
2019-02-04T18:58:22Z UpdateTip: new best=0000000000000000000ea08a724fcf33b2b6366f0919e63e1ed154c09da62d1d height=543479 version=0x20000000 log2_work=89.754441 tx=344908619 date='2018-09-28T17:37:53Z' progress=0.927998 cache=117.0MiB(1028641txo)
2019-02-04T18:58:37Z UpdateTip: new best=0000000000000000000c3783ccc4122b6bf00df04552d86e06fbfa589f76b4bf height=543480 version=0x20000000 log2_work=89.754483 tx=344911325 date='2018-09-28T17:48:36Z' progress=0.928002 cache=117.5MiB(1032844txo)
2019-02-04T18:58:53Z UpdateTip: new best=000000000000000000104daa4fa5196fc035bcec7ef2576cde39d6ebf9bcff6f height=543481 version=0x20000000 log2_work=89.754526 tx=344914192 date='2018-09-28T17:49:27Z' progress=0.928003 cache=117.7MiB(1035002txo)
[... cut ...]
Feb 04 18:56:51 vasilightning lnd[765]: 2019-02-04 18:56:51.151 [DBG] LNWL: Unable to parse output script in 2e99e1c66b3a6c06188056fb624b9a6d44c198eced7581aeaf1a320716e3b71b:2: opcode OP_DATA_75 requires 76 bytes, but script only has 39 remaining
Feb 04 18:59:10 vasilightning lnd[765]: 2019-02-04 18:59:10.719 [DBG] LNWL: Unable to parse output script in b063871b2ee3e5435fa8a721a952c10a4c81fbef3672206c9ba0f5432776ff50:2: opcode OP_DATA_75 requires 76 bytes, but script only has 39 remaining

- RTL is OFF by config
```

rootzoll, 2019-02-04:
@vmavromatis if your blockchain was from Sep 2018 thats some load to for the raspberrypi to catch up with. If you are not in a hurry and you see progress keep it running. 

There are some ways to transfere a blockchain from another computer (with the latest image release), see: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#i-have-the-full-blockchain-on-another-computer-how-do-i-copy-it-to-the-raspiblitz
But I would only advise if you want to start with a fresh/clean HDD and SD card again.

rootzoll, 2019-02-14:
If anybody likes to help on the final testing if the image I am testing for v1.0 release .. feel free :)
wiki.fulmo.org/downloads/raspiblitz-v1.0-2019-02-14.img.zip

vmavromatis, 2019-02-14:
> @vmavromatis if your blockchain was from Sep 2018 thats some load to for the raspberrypi to catch up with. If you are not in a hurry and you see progress keep it running.
>
> There are some ways to [...]

Thanks, I doing the copyHDD method now from my main PC. Should be 17 hours :P 
```
FILE TRANSFER TIME CALCULATOR
Speed: 5 MB/s
Size: 273 GB
17 Hours, 05 Minutes and 01 Seconds
```
If this won't work I will try the test v1.0 image :) Cheers

Kally95, 2019-02-15:
Hi, I'm trying to put your SD writer on the SDHC itself but having copied over the NOOB stuff from RaspberryPi, I've got no space on a  16GB sd card. Is there anyway I can contact you as I'm trying to set up a LN node. Thanks

openoms, 2019-02-17:
Successfully built with the latest commits. No issues on my rPi 3 B+ with 16GB SD and 500 GB HDD. Also succeeded without issues  on an Odroid HC1 and HC2.

openoms, 2019-02-17:
> Hi, I'm trying to put your SD writer on the SDHC itself but having copied over the NOOB stuff from RaspberryPi, I've got no space on a 16GB sd card. Is there anyway I can contact you as I'm trying to set up a LN node. Thanks
@Kally95 it sounds like you need to start over. Don`t use the NOOb images. You could burn the [prebuilt image](https://wiki.fulmo.org/downloads/raspiblitz-0.99-2019-01-22.img.gz)  or start with the Raspbian Stretch Desktop if you want to build yourself.

Kally95, 2019-02-17:
Hi,

I managed to do it, I had to go with option 2 of downloading from the back up as the torrent wasn’t making any progress but it’s just a waiting game at the moment. Thank you!

lucbran, 2019-02-18:
Hi, I am trying to install update from 0.98 version.
Burnt sd card and started raspiblitz. I can enter password and wallet is unlocked (it was empty anyway so cannot say if previous BTC can be maintained).
Then unfortunately sys freeze : lightning Filtering Blockchain/ Progress 556075/563569.Then system is cotinuously asking to re enter password, unlocks and freezes... No way to go ahead..

openoms, 2019-02-18:
@lucbran in the RaspiBlitz terminal press Ctrl+C and tell us what you get on 'sudo systemctl status lnd'.

lucbran, 2019-02-18:
lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-02-18 08:38:01 GMT; 18s ago
 Main PID: 3866 (lnd)
   CGroup: /system.slice/lnd.service
           └─3866 /usr/local/bin/lnd --externalip=

Feb 18 08:38:01 S.P.Q.R.Nexus systemd[1]: lnd.service: Service hold-off time over, s
Feb 18 08:38:01 S.P.Q.R.Nexus systemd[1]: Stopped LND Lightning Daemon.
Feb 18 08:38:01 S.P.Q.R.Nexus systemd[1]: Started LND Lightning Daemon.
Feb 18 08:38:02 S.P.Q.R.Nexus lnd[3866]: 2019-02-18 08:38:02.228 [INF] LTND: Version
Feb 18 08:38:02 S.P.Q.R.Nexus lnd[3866]: 2019-02-18 08:38:02.229 [INF] LTND: Active 
Feb 18 08:38:02 S.P.Q.R.Nexus lnd[3866]: 2019-02-18 08:38:02.244 [INF] CHDB: Checkin
Feb 18 08:38:02 S.P.Q.R.Nexus lnd[3866]: 2019-02-18 08:38:02.250 [INF] RPCS: passwor
Feb 18 08:38:02 S.P.Q.R.Nexus lnd[3866]: 2019-02-18 08:38:02.250 [INF] RPCS: passwor
Feb 18 08:38:02 S.P.Q.R.Nexus lnd[3866]: 2019-02-18 08:38:02.251 [INF] LTND: Waiting
lines 1-16/16 (END)

openoms, 2019-02-18:
> lnd.service - LND Lightning Daemon
> Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
> Active: active (running) since Mon 2019-02-18 08:38:01 GMT; 18s ago
> Main PID: [...]

your --externalip= is missing. Try running `sudo /home/admin/_bootstrap.sh` which fill fill the publicIP in raspiblitz.conf. 
This issue has been solved before though. Are you running the latest image?

rootzoll, 2019-02-18:
This should be fixed now. I will release v1.0 today ... if you still have problems with that, please file a new issue. I will close this one now in preparation of v1.0 release.

closed 2019-02-18 by rootzoll