raspiblitz/raspiblitz #289: LN Unknown Chain

VinceDot, 2019-01-24:
Hello I am trying to setup my Raspberry PI with Raspiblitz 0.99 and have successfully downloaded the torrents but now that I created my LN wallet I am not being asked to unlock my LN wallet, this is the message I get : 

`Unlock LND Wallet --> lncli --chain= unlock
****************************************************************************
HELP: Enter your PASSWORD C
You may wait some seconds until you get asked for password.
****************************************************************************
[lncli] unknown chain:
`

Thanks for your help

hikedev, 2019-01-25:
I did run into the same issue with "raspiblitz-0.99-2019-01-22". [lncli] unknown chain:
I think its a timing issue when you press to fast or instant ok when you setup lightning the first time.

Even if I did add the raspiblitz.conf with the expected values you will run into more lnd and missing files issues, as a result, the installation will not continue to set up the firewall(ufw) settings etc...

I went with rewriting the img to the sd card from scratch and kept the blockchain like in a normal update and during setup I always did wait the maximum amount of time during the progress screens and now everything works fine.

Thanks for the amount of work you guys puting into this, alone from the view shellscripts i overlooked i can tell this was quite some work. ;)

VinceDot, 2019-01-25:
I will try rewriting the img because effectively I do not have raspiblitz.conf on my HDD. 

Thanks for your help

VinceDot, 2019-01-25:
I'm now getting the same issue as #293 whenever I launch my Raspberry after rewriting the img. Using "raspiblitz-0.99-2019-01-22".

rootzoll, 2019-01-26:
This is a bug we have to look deeper into and analysis already started in ticket #293.

closed 2019-02-03 by VinceDot