raspiblitz/raspiblitz #290: Wallet Seed

DaReaLDeviL, 2019-01-24:
Please help me to understand the 24 word seed of the raspiblitz.

Is this like a normal wallet seed where I put my 24 words in it? And when yes, please tell which wallet I can use.

And how does "password d" effects this seed? It is protected in the raspi, but is there some other protection for a "normal wallet" when I try to recover it?

Are the lightningchannels are bound to this wallet or to the information on the hdd of the raspiblitz? Which is the factor that the channels and states are known and not lost?

rootzoll, 2019-01-28:
> Is this like a normal wallet seed where I put my 24 words in it? And when yes, please tell which wallet I can use.

It should be a BIP39 24-word wallet seed that can be recoverd by others wallets like Electrum (file > new -> standard wallet > already have seed > options > activateb BIP 39) ... but I havent tested it yet.

> And how does "password d" effects this seed? 

On wallet creation you get asked if you want to protect your word seed list with an additional password. If you choose so, RaspiBlitz recommends you to use your PASSWORD D at this point.

To use a an additional password for your seed words is optional. If you choose so, you will need the password to recover your private key from your your seed words later on. Without this password your private key cannot be recovered from your seed words. So the password adds an additional layer of security, if someone finds your written down word list.

> It is protected in the raspi, but is there some other protection for a "normal wallet" when I try to recover it?

Your active LND wallet (with all the state channels etc) is stored on the RaspiBlitz and is protected by PASSWORD C that needs to be entered by you on every restart. Only if you activate the "Auto-Unlock" feature it gets stored on the RaspiBlitz - what will lower the security, but adds comfort.

> Are the lightningchannels are bound to this wallet or to the information on the hdd of the raspiblitz? Which is the factor that the channels and states are known and not lost?

All the channel state is in /mnt/hdd/lnd ... see info on backup:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-backup-my-lightning-node

To make it easier to recover channel funds from seed is in the Lightning feature list for 2019 ... at the moment its a moving target: https://www.coindesk.com/state-of-lightning-whats-the-path-for-network-adoption-in-2019

closed 2019-01-28 by rootzoll

walletofsatoshi, 2019-04-09:
This generated seed is not working on other bip39 wallets like electrum

openoms, 2019-04-14:
I can confirm this, but it is an issue of LND.

walletofsatoshi, 2019-04-14:
So apparently lnd does not use bip39 seed and will not work on other wallets like electrum. Confirm this at slack

Himbeergeld, 2019-04-14:
This is indeed crucial. Looks like LND has implemented an aezeed cipher seed:

> This new scheme aims to address two major features lacking in BIP39: versioning, and a wallet birthday. The lack a version means that wallets may not necessarily know how to re-derive addresses during the recovery process. A lack of a birthday means that wallets don’t know how far back to look in the chain to ensure that they derive all the proper user addresses. Additionally, BIP39 use a very weak KDF. We use scrypt with modern parameters (n=32768, r=8, p=1)

So the LND Desktop App would probably be the main alternative to recover the cipher seed if the RaspiBlitz HW fails.

openoms, 2019-04-14:
Curently the Node Launcher is the closest thing to a desktop GUI for LND: https://github.com/lightning-power-users/node-launcher

I think Zap is planned to have a Lightning only wallet (backed by Neutrino) so that will be interesting.

rootzoll, 2019-04-28:
The Lightning Desktop App was realeased: https://github.com/lightninglabs/lightning-app/releases
I heard that people made good experience to recover on-chain funds with that app from just wallet seed and would recommend it for now.

openoms, 2020-10-19:
just to confirm here LND uses the seed format called `aezeed` which is not BIP39 compatible: it is not compatible with the CC. 
aezeed vs BIP39: https://github.com/lightningnetwork/lnd/tree/master/aezeed

Development is ongoing to make the onchain part of LND usable with other wallets: https://github.com/lightningnetwork/lnd/issues/4678
For mow there is this workaround: https://github.com/openoms/lightning-node-management/blob/master/RestoreLNDonchainFundsInElectrum.md