raspiblitz/raspiblitz #292: Neverending Loop on LND Wallet Unlocking

Jos3xTv, 2019-01-25:
Hi there.

Just finishing my LN node set up, when i come to this step, it starts to loop with no end. This is the screen, clearing afterwards.
```
****************************************************************************
Unlock LND Wallet --> lncli --chain= unlock
****************************************************************************
HELP: Enter your PASSWORD C
You may wait some seconds until you get asked for password.
****************************************************************************
[lncli] unknown chain:
tail: cannot open '/mnt/hdd/lnd/logs//net/lnd.log' for reading: No such file or directory
checking setup script
network(bitcoin)
chain(main)
setupStep(0)
LND is running ...
lnd wallet exists ... checking if locked
cat: .network: No such file or directory
gathering chain info ... please wait
./AAunlockLND.sh: line 8: -cli: command not found
```
How can i resolve this looping?

rootzoll, 2019-01-25:
Its loosing the "network" information from the config file.

Can you go to the terminal (CTRL+c), exceute the following command and post the result:
`sudo cat /mnt/hdd/raspiblitz.conf `

Jos3xTv, 2019-01-25:
The command proposed and a bit more.
```
admin@RaspiBlitz:~ $ sudo cat /mnt/hdd/raspiblitz.conf
cat: /mnt/hdd/raspiblitz.conf: No such file or directory
admin@RaspiBlitz:~ $ cd /mnt/hdd
admin@RaspiBlitz:/mnt/hdd $ ls
bitcoin  lnd  lost+found
admin@RaspiBlitz:/mnt/hdd $ ls -lh
total 24K
drwxr-xr-x 6 bitcoin bitcoin 4.0K Jan 25 23:23 bitcoin
drwxr-xr-x 4 bitcoin bitcoin 4.0K Jan 25 16:26 lnd
drwx------ 2 root    root     16K Jan 23 16:27 lost+found
admin@RaspiBlitz:/mnt/hdd $ ls -a
.  ..  bitcoin  lnd  lost+found
admin@RaspiBlitz:/mnt/hdd $
```

rootzoll, 2019-01-26:
@Jos3xTv a quick fix should be running on terminal:
`echo "bitcoin" > /home/admin/.network`
and then try again with the command
`raspiblitz`

But just one more question - can you describe the moment a bit more when this error showed up? Did you had a reboot before it asked for unlocking and got into looping?

fluidvoice, 2019-01-26:
> Its loosing the "network" information from the config file.
> 
> Can you go to the terminal (CTRL+c), exceute the following command and post the result:
> `sudo cat /mnt/hdd/raspiblitz.conf `

Three people so far are seeing this problem from the config file not present: https://github.com/rootzoll/raspiblitz/issues/293#issue-403350241

rootzoll, 2019-01-26:
@fluidvoice  Yes this needs to be fixed before version 1.0 release.
@Jos3xTv whats in the `lost+found` directory, can you post a directory listing?

Jos3xTv, 2019-01-28:
Hi there.

LND working smoothly. Great work. And in lost+found i found nothing.
```
admin@RaspiBlitz:/mnt/hdd $ cd /lost+found
-bash: cd: /lost+found: Permission denied
admin@RaspiBlitz:/mnt/hdd $ ls -lh
total 1.1G
drwxr-xr-x 6 bitcoin bitcoin 4.0K Jan 28 15:03 bitcoin
drwxr-xr-x 4 bitcoin bitcoin 4.0K Jan 25 16:26 lnd
drwx------ 2 root    root     16K Jan 23 16:27 lost+found
-rwxrwxrwx 1 root    root     110 Jan 26 10:06 raspiblitz.conf
-rw------- 1 root    root    1.0G Jan 26 09:57 swapfile
admin@RaspiBlitz:/mnt/hdd $ sudo su
root@RaspiBlitz:/mnt/hdd# cd /lost+found
root@RaspiBlitz:/lost+found# ls -lh
total 0
root@RaspiBlitz:/lost+found#
```

rootzoll, 2019-01-28:
OK. I will close this issue - but the basic problem will be worked on in #293

closed 2019-01-28 by rootzoll