raspiblitz/raspiblitz #308: Blockchain being written to wrong place (/root/.bitcoin)

IvanThinking, 2019-02-09:
My lightning node (set up 100% using directions here) stopped working in November (v .96). I'm trying to revive it now, ran a diskcheck and found lots of errors in BLOCKCHAIN. After the check it now boots to the main unlock screen. Once I enter password A, I'm basically in a terminal. 

I figured I probably neet to run bitcoind to get it resynched so I tried to do so... I got an out of disk space error after a while. 

Heres what I see:

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G   14G   64M 100% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M  6.3M  458M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   43M   23M   21M  52% /boot
/dev/sda1       916G  243G  627G  28% /mnt/hdd
tmpfs            93M     0   93M   0% /run/user/1000
tmpfs            93M     0   93M   0% /run/user/1001

And then with: sudo du -h /root --max-depth=1

8.9G    /root/.bitcoin
12K     /root/.vnc
36K     /root/.gnupg
65M     /root/.cache
8.9G    /root

And further: sudo du -h /root/.bitcoin --max-depth=1
16K     /root/.bitcoin/chainstate
1.4M    /root/.bitcoin/wallets
8.8G    /root/.bitcoin/blocks
8.9G    /root/.bitcoin

So it looks like /blocks is trying to write to filled up root. Did some config get screwed up? Is this where it is supposed to be?

rootzoll, 2019-02-09:
bitcoind should always be run by user bitcoin and finally be stored on HDD.

Because with 0.96 you have a quite old installation I would recommend going the update route:
https://github.com/rootzoll/raspiblitz/blob/dev/FAQ.md#how-to-update-a-old-raspiblitz-before-version-098

If you had funds in your old RaspiBlitz maybe readup on this:
https://github.com/rootzoll/raspiblitz/blob/dev/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

IvanThinking, 2019-02-12:
Nuked the whole project and started over. Apparently somehow my /bitcoin directory became read-only which forced(?) the blockchain to write to the default location which filled the drive and then caused all kinds of other fun.

closed 2019-02-12 by IvanThinking