raspiblitz/raspiblitz #309: LND Wallet

Otti24xx, 2019-02-09:
Kann mir irgendjemand helfen, anbei mein  Fehlerprotokoll:
Blockchain ist sync, nur bei LND gibt es immer wieder Fehler eine Wallet habe ich schon x mal erstellt, entsperren lässt sie sich nicht, wahrscheinlich macaroon Fehler????

*** RASPIBLITZ LOGS ***
blitzversion: 0.99
chainnetwork: bitcoin / main
 18:15:08 up 35 min,  2 users,  load average: 0,39, 0,36, 0,77

*** CHAINNETWORK SYSTEMD STATUS ***
? bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-02-09 17:39:50 CET; 35min ago
  Process: 684 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 692 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           +-692 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Feb 09 17:39:49 OttisRaspiBlitz systemd[1]: Starting Bitcoin daemon...
Feb 09 17:39:50 OttisRaspiBlitz systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-02-09T16:43:09Z dnsseed thread start
2019-02-09T16:43:09Z addcon thread start
2019-02-09T16:43:09Z msghand thread start
2019-02-09T16:43:09Z opencon thread start
2019-02-09T16:43:09Z init message: Done loading
2019-02-09T16:43:20Z Loading addresses from DNS seeds (could take a while)
2019-02-09T16:43:26Z New outbound peer connected: version: 70015, blocks=562325, peer=0
2019-02-09T16:43:27Z 216 addresses found from DNS seeds
2019-02-09T16:43:27Z dnsseed thread exit
2019-02-09T16:43:33Z New outbound peer connected: version: 70015, blocks=562325, peer=2
2019-02-09T16:43:34Z New outbound peer connected: version: 70015, blocks=562325, peer=3
2019-02-09T16:43:41Z New outbound peer connected: version: 70015, blocks=562325, peer=5
2019-02-09T16:43:42Z New outbound peer connected: version: 70015, blocks=562325, peer=6
2019-02-09T16:43:43Z New outbound peer connected: version: 70015, blocks=562325, peer=7
2019-02-09T16:43:44Z New outbound peer connected: version: 70015, blocks=562325, peer=8
2019-02-09T16:43:51Z New outbound peer connected: version: 70015, blocks=562325, peer=9
2019-02-09T16:47:34Z UpdateTip: new best=000000000000000000125ad7841e8103eecdf38979f6ee786acabac186a1e61e height=562326 version=0x20000000 log2_work=90.330076 tx=381366014 date='2019-02-09T16:46:12Z' progress=0.999999 cache=2.0MiB(18220txo) warning='42 of last 100 blocks have unexpected version'
2019-02-09T16:49:59Z Imported mempool transactions from disk: 6460 succeeded, 242 failed, 0 expired, 62 already there
2019-02-09T16:59:09Z UpdateTip: new best=0000000000000000001121b8825920903f98957bfaf0bec5c19ae5438e1b0414 height=562327 version=0x20400000 log2_work=90.330099 tx=381369427 date='2019-02-09T16:59:01Z' progress=1.000000 cache=3.5MiB(30532txo) warning='42 of last 100 blocks have unexpected version'
2019-02-09T17:03:48Z New outbound peer connected: version: 70015, blocks=562327, peer=10

*** LND SYSTEMD STATUS ***
? lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-02-09 17:39:50 CET; 35min ago
 Main PID: 693 (lnd)
   CGroup: /system.slice/lnd.service
           +-693 /usr/local/bin/lnd --externalip=188.103.254.187

Feb 09 17:39:51 OttisRaspiBlitz lnd[693]: 2019-02-09 17:39:51.854 [INF] RPCS: password RPC server listening on [::]:10009
Feb 09 17:39:51 OttisRaspiBlitz lnd[693]: 2019-02-09 17:39:51.854 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to…and unlock it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Sat 2019-02-09 17:39:37 CET, end at Sat 2019-02-09 18:15:08 CET. --
Feb 09 17:39:50 OttisRaspiBlitz systemd[1]: Started LND Lightning Daemon.
Feb 09 17:39:51 OttisRaspiBlitz lnd[693]: 2019-02-09 17:39:51.795 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
Feb 09 17:39:51 OttisRaspiBlitz lnd[693]: 2019-02-09 17:39:51.797 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Feb 09 17:39:51 OttisRaspiBlitz lnd[693]: 2019-02-09 17:39:51.828 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
Feb 09 17:39:51 OttisRaspiBlitz lnd[693]: 2019-02-09 17:39:51.851 [INF] RPCS: password gRPC proxy started at [::]:8080
Feb 09 17:39:51 OttisRaspiBlitz lnd[693]: 2019-02-09 17:39:51.854 [INF] RPCS: password RPC server listening on [::]:10009
Feb 09 17:39:51 OttisRaspiBlitz lnd[693]: 2019-02-09 17:39:51.854 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config

*** RASPIBLITZ LOGS ENDE ***

weitere Fehler mit Macron:

[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

im Hauptmenu:./00mainMenu.sh

admin@OttisRaspiBlitz:~ $ ./00mainMenu.sh
Starting the main menu ...
setup is done - loading config data
checking bitcoind - please wait ..
can take longer if device was off or first time
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p prompt] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p prompt] [-u user] file ...
tail: '/mnt/hdd/lnd/logs//net/lnd.log' kann nicht zum Lesen geöffnet werden: Datei oder Verzeichnis nicht gefunden
tail: '/mnt/hdd/lnd/logs//net/lnd.log' kann nicht zum Lesen geöffnet werden: Datei oder Verzeichnis nicht gefunden
tail: '/mnt/hdd/lnd/logs//net/lnd.log' kann nicht zum Lesen geöffnet werden: Datei oder Verzeichnis nicht gefunden
(standard_in) 1: syntax error
tail: '/mnt/hdd/lnd/logs//net/lnd.log' kann nicht zum Lesen geöffnet werden: Datei oder Verzeichnis nicht gefunden
^C
admin@OttisRaspiBlitz:~ $

Otti24xx, 2019-02-09:
admin@OttisRaspiBlitz:~ $ nano /home/bitcoin/.lnd/lnd.conf
  GNU nano 2.7.4                                                                Datei: /home/bitcoin/.lnd/lnd.conf

# RaspiBolt: lnd configuration
# /home/bitcoin/.lnd/lnd.conf

[Application Options]
debuglevel=info
maxpendingchannels=5
alias=OttisRaspiBlitz
color=#68F442
#nat=true

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# REST open to all connections on Port 8080
restlisten=0.0.0.0:8080
# Domain, could use https://freedns.afraid.org
#tlsextradomain=lightning.yourhost.com
#tlsextraip=0.0.0.0

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

[Bitcoind]
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=xxx
bitcoind.rpchost=127.0.0.1
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6

Otti24xx, 2019-02-12:
Zwischenlösung gefunden - freilich unter Verzicht des Macaronen-Sicherheitsfeature: 

LND Wallet lässt sich erstellen und öffnen als Bitcoinnutzer mit:

$ lncli --no-macaroons create
$ lncli --no-macaroons unlock

danach sollte LND vollständig starten ...

$ sudo journalctl -f -u lnd

Otti24xx, 2019-02-13:
Habe 1.0 installiert, nun läuft alles... Hab Fehler gefunden, ich hatte versehentlich nur 7 Zeichen im LND Passwort (C), wahrscheinlich war das der Fehler - nun mit 8. Zeichen läuft alles wunderbar

rootzoll, 2019-02-13:
Super :) - ich schließe das Ticket dann erstmal.

closed 2019-02-13 by rootzoll