raspiblitz/raspiblitz #311: Unable to cashout

DaReaLDeviL, 2019-02-10:
Hi rootzoll,

just tried to cashout but I run into issues with it:

> TRY #7 ---> with max fee 7000 sat:
> lncli --chain=bitcoin --network=mainnet sendcoins --addr 3xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx --amt 8616493 --conf_target 3
> FAIL: [lncli] rpc error: code = Unknown desc = -25: Missing inputs
> 
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> FINAL FAIL --> Was not able to send transaction (see error above)
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> Press ENTER to return to main menu.

What whould be your advice now?

rootzoll, 2019-02-10:
"Missing Inputs" is a LND error code I havent seen before. Would be best to head over to the LND Dev Slack and get more info on that: https://dev.lightning.community ... please report back your findings

DaReaLDeviL, 2019-02-11:
They point out to do a rescan of the wallet... some db issue

this tool can do it:
https://github.com/btcsuite/btcwallet/tree/master/cmd/dropwtxmgr

this is some install:
https://github.com/lightningnetwork/lnd/issues/1494#issuecomment-421908735

but I don't know how to setup the right ~/go path for the install?
Could I just run
> go get -u github.com/btcsuite/btcwallet/cmd/dropwtxmgr
or do I have to create something first?

rootzoll, 2019-02-11:
Much thanks for the reasearch. How to run something in go on the RaspiBlitz check the 97addMobileWalletZap.sh script as an example. Please share your solution if others run into the same problem.

rootzoll, 2019-02-11:
I will close this issue ... because its an LND error - not with RaspiBlitz.
But please keep posting your findings on the closed issue.

closed 2019-02-11 by rootzoll