raspiblitz/raspiblitz #314: Zeichensatz

Otti24xx, 2019-02-13:
Habe SD Karte mit Version 1.0 erstellt - läuft alles super. Frage zum Zeichensatz, bei mir zeigt es Buchstaben am Rand statt einer Linie an:

                 lqqqqqqqqqqqq bitcoin / main qqqqqqqqqqqqqqk
                 x  Lightning Catching-Up Blockchain        x
                 x  Progress: 100000/562855                 x
                 x  re-rescan every start until funding     x

rootzoll, 2019-02-13:
hmm ... bei meinem Testbuild scheint es soweit gut ... doch jeder build ist anders. Ganz am Anfang beim build script werden die Zeichensätze installiert ... ggf. gabs da nen Aussetzer.

Otti24xx, 2019-02-14:
Hallo Christian, das Einrichten lief ohne Stocken durch, ja am Anfang erinnere ich mich an drei Zeichensatzdateien, ich glaube US und DE... kommt das evtl. vom Putty, kann man da auch noch was einstellen? Aber Super Arbeit! Danke! läuft alles klasse

rootzoll, 2019-02-14:
Danke :) Mit Putty hab ich keine aktuellen Erfahrungen - falls jemand nen Tip hat, welchen Zeichensatz man einbinden muss, damit es mit Putty out-of-the-box korrekt aussieht bitte gerne. 

Damit ist der issue für 1.0 release nicht kritisch. Ich lass ihn mal als enhancement offen.

frennkie, 2019-10-03:
The - non default (?!) - settings I am using on Putty are:

* Terminal -> Keyboard -> Function keys and keypad: Xterm R6
* Windows -> Translation -> Remote Char Set: UTF-8

@Otti24xx Please test and let us know whether this works for you.

frennkie, 2019-10-17:
Hm.. I guess this is also needed: 

```
Windows -> Translation -> [X] Enable VT 100 line drawing even in UTF-8 mode
```

closed 2019-11-20 by rootzoll

M73, 2024-03-19:
These 3 worked for me:

> The - non default (?!) - settings I am using on Putty are:
> 
> * Terminal -> Keyboard -> Function keys and keypad: Xterm R6
> * Windows -> Translation -> Remote Char Set: UTF-8
> 
> Hm.. I guess this is also needed:
> ```
> Windows -> Translation -> [X] Enable VT 100 line drawing even in UTF-8 mode
> ```

Thanks @frennkie