raspiblitz/raspiblitz #319: LND not syncing / starting anymore

wuestenigel, 2019-02-16:
Hey, after my vacation my raspiblitz was offline and I wanted to get it back working. Unfortunately I got stuck here:

`Feb 16 03:41:44 wuestenigel lnd[29660]: 2019-02-16 03:41:44.844 [INF] CRTR: Syncing channel graph from height=561961 (hash=0000000000000000000e64331008e9d6033f8cf07a488e907c465633430c7bab) to height=563215 (hash=00000000000000000001faa205cc4679f1a08a17e5b3e0dccd7f48aa3d832a23)
Feb 16 03:41:45 wuestenigel lnd[29660]: 2019-02-16 03:41:45.649 [ERR] SRVR: unable to start server: unable to fetch node: 02f3067188240c75beb36477db63771f66c976f7e5bdd6f6ace5508396aba24815, unable to find node
Feb 16 03:41:45 wuestenigel systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Feb 16 03:41:45 wuestenigel systemd[1]: lnd.service: Unit entered failed state.
Feb 16 03:41:45 wuestenigel systemd[1]: lnd.service: Failed with result 'exit-code'.
`

any ideas how I can solve this? Should I try to update to v1.0?

wuestenigel, 2019-02-16:
Played around a little bit, now this: Do I just have to wait or can I do sth?

[image]

rootzoll, 2019-02-16:
I think a move to v1.0 would be best: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#2-lnd-channel-state-backup aqnd here is preview image of version 1.0: wiki.fulmo.org/downloads/raspiblitz-v1.0-2019-02-14.img.zip

wuestenigel, 2019-02-16:
Thanks, any ideas how to solve this permission thing? sorry, not 100% familiar with terminal:

> Macintosh:~ marcoverch$ scp -r admin@192.168.178.65:/mnt/hdd/lnd/  /Users/marco/Dropbox/raspiblitz_backup/vor_1_0
admin@192.168.178.65's password: 
tls.cert                                                                                                                                                                              100%  774   348.5KB/s   00:00    
scp: /mnt/hdd/lnd/tls.key: Permission denied

rootzoll, 2019-02-16:
the admin user does not own the lnd files and gets a "permission denied".

use the bitcoin user - thats the owner of the LND data files:
scp -r bitcoin@192.168.178.65:/mnt/hdd/lnd/ /Users/marco/Dropbox

wuestenigel, 2019-02-18:
Great, will do!

Is it normal that progress after 24 hours is about 10%

[image]

rootzoll, 2019-02-18:
Yeah - reindexing is a measure of last resort and will take a very long time.

With the v1.0 release to day there will be two alternatives to speed this up (replacing a corrupt blockchain). So can go to terminal and call either "/home/admin/50copyHDD.sh" to replace blockchain with fresh data from another computer by SCP or you can call "/home/admin/50torrentHDD.sh" to download a fresh prepared blockchain by torrent - once torrent is finished it will replace old data.

wuestenigel, 2019-02-28:
Great, is it normal that it took 10 days to reach 40% right now? The HDD contained the Blockchain already.

rootzoll, 2019-02-28:
re-index means there was a problem/data-corruption with your blockchain and this is the repair fallback modus. Yeah and it really takes so long - it basically validating the whole blockchain again.

There will be more options in the next release v 1.1 to speed this up .. see my last post which experimental ways you have already in v1.0 to speed this up.

wuestenigel, 2019-03-15:
Okay, what is the reason here? My HDD had the full blockchain on it before upgrading to v1.0 

I'm now on 75% after waiting 25 days. ;)

Does it mean I have to wait one month after each version update to 1.1, 1.2 and so on? ;-)

rootzoll, 2019-03-19:
Normally blockchain data will be kept thru updates.

Version 1.0 had the build in discovery that bitcoind is requesting a reindex and then started the reindex. Not sure if that triggered anything during your update to 1.0 - because there was an error in the past with your pre-1.0 version.

Version v1.1 now is out and offers TORRENT & COPY option in case of reindex situation - this should speed up repairs. If you update during a re-index. It should offer you those new options ... not tested.

I am closing this issue. But feel free to give feeback if this was a fix for you.

closed 2019-03-19 by rootzoll