raspiblitz/raspiblitz #322: RTL installation failed (Cannot find module 'express')

raumi75, 2019-02-17:
I installed RTL by checking it in the service menu. Even after reboot it did not start up.

```
admin@raumiblitz:~ $ service RTL status
● RTL.service - RTL daemon
   Loaded: loaded (/etc/systemd/system/RTL.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2019-02-17 03:57:17 GMT; 1s ago
  Process: 8989 ExecStart=/usr/bin/node /home/admin/RTL/rtl --lndir /home/admin/.lnd/data/chain/bitcoin/mainnet (code=exited, status=1/FAILURE)
 Main PID: 8989 (code=exited, status=1/FAILURE)
```

Starting it manually produces this error:

```
admin@raumiblitz:/var/log $ sudo /usr/bin/node /home/admin/RTL/rtl --lndir /home/admin/.lnd/data/chain/bitcoin/mainnet
module.js:549
    throw err;
    ^

Error: Cannot find module 'express'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/admin/RTL/app.js:2:17)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
```

rootzoll, 2019-02-18:
Was this the v1.0 preview? ... in older code bases there was a problem I fixed. I checked on the latest code base install works - I even updated RTL to v.0.2.1 now, looks good. 

I am closing this issue ... if this was v1.0 preview, I will reopen.

raumi75, 2019-02-18:
Yes. This is v1.0 preview. I used the latest img-file I downloaded last week.

rootzoll, 2019-02-18:
OK I will keep it open and test some more. If I am not able to reproduce .. I will keep open, but not delay v1.0

raumi75, 2019-02-20:
I can confirm this error persists after a complete reinstall with the official v1.0 image.

rootzoll, 2019-02-21:
I was not able to reproduce it .. but its on the list for the next release to look deeper into.

raumi75, 2019-02-24:
Of course you couldn't reproduce it. It was not Raspiblitz. It was MY network. The dns server did not resolve the tor-server properly resulting in an ssl-error that aborted the installation process.

```
sudo apt-get update
...
https://deb.torproject.org/torproject.org stretch/main Sources
server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
```

I fixed it by overruling the dns-server in `/etc/dhcpcd.conf` (added this line:)

```
static domain_name_servers=1.1.1.1 8.8.8.8
```

and ran `systemctl daemon-reload`.

After that the installation worked fine.

The only lesson learned should be that we should **check if the installation of rtl was successful** before activating RTL and rebooting, because the user can't see the error messages when the dialog appears announcing that RTL will be available after reboot.

thelwyn, 2019-02-25:
I freshly installed Raspiblitz 1.0 and was excited to try the RTL interface but nothing was showing up.

When activating the service , it started the installation and I could see that at some point it couldn't reach some TOR repo, but the installation went on with a final message saying it was going to restart and to try port 3000 after that.

But nothing shows up. Then checking the RTL service log, it says it can't reach the directory /home/admin/RTL/ ... and indeed, it doesn't exist!

I retry disabling/enabling the service after that, and during the installation I monitored the /home/admin/ directory from time to time: at some point, an RTL directory appears, but by the end of the unsuccessful installation, this directory had disappeared.

- Is TOR installation necessary for this RTL?
- If yes, the installation should clearly states that it failed, and not suggesting that after reboot just need to check local-ip:3000.

I will double-check tonight after @raumi75 comment and see if it was a DNS server resolution problem too (I'm in China, why I am not suprised that TOR repo isn't reachable at first...)

raumi75, 2019-02-25:
Tor is not necessary, but the installation process aborts when the tor repository can't be reached.

I just had a simple DNS problem. Debugging ist is easier if you don't run the services menu, but simply try `apt update` in your terminal. You will see where it fails.

thelwyn, 2019-02-27:
Yep that was it, the tor repo was unreachable, now my Raspiblitz is behind VPN, it can reach it. Will try reenabling RTL tonight.

Agree with @raumi75 comment above, the success of RTL installation should be checked and a feedback given to the user.

raumi75, 2019-02-27:
Maybe just check for the presence of `/home/admin/RTL/rtl.js` or like Line 35 check if the service exists before rebooting. If that file is not there, then something went wrong and there is no use activating the systemd service and rebooting.

rootzoll, 2019-03-13:
TODO v1.1: Make that check if the Install ran thru
TODO v1.1: Update to new RTL version: https://twitter.com/RTL_App/status/1105657165279318016

rootzoll, 2019-03-14:
OK RTL version is updated and RTL got more checks on install. Will be part of v1.1 realease. Now in final testing.

rootzoll, 2019-03-17:
RTL runs fine so far. Closing issue for v1.1 release.

closed 2019-03-17 by rootzoll