raspiblitz/raspiblitz #324: Dynamic DNS deleting tls without questioning

raumi75, 2019-02-17:
After upgrading to version 1.0, my Shango connection was working fine. Then I noticed that there was no dynamic dns in the services section. So I added that again, just to have a proper setup.

What I did not realize was that it generates a new tls certificate and throws away the old one. In my case that was not necessary because I was using the same domain.

Is there a reason for that behavior? Could we move the old tls.* files somewhere instead of deleting them? In my case, I recovered the keys from a backup because I did not want to reconnect my clients.

```
/home/admin/config.scripts/lnd.newtlscert.sh 
```

rootzoll, 2019-02-20:
Thats right .. on the recover/update if might do that. I will add this to the next 1.1 release.

rootzoll, 2019-03-14:
TODO: Check if dyndns stay stable after update process.

rootzoll, 2019-03-15:
To prevent TLS renew on update I havent found an elegant solution. For now after an update you most prob need to re-connected your devices. 

But I changed the delete to a move like this:
```
echo "keep old tls data as backup"
sudo mv /mnt/hdd/lnd/tls.cert /mnt/hdd/lnd/tls.cert.old 
sudo mv /mnt/hdd/lnd/tls.key /mnt/hdd/lnd/tls.key.old 
```

But it could even happen that on a update its called twice ... so this is not a solution yet, just a first tweak.

Maybe before an update/recovery begins it should backup the TLS and Macroons and when its done setting everything up it will copy them back - not sure.

rootzoll, 2019-03-17:
Moving for a more enhenced solution to fure release milestone.

rootzoll, 2019-04-09:
On the beginning of a update TLS certs and macaroons are getting now backuped and replayed at the end of a backup - so after a update connections to external devices/wallets should stay stable.

TODO: Test on update.

rootzoll, 2019-04-27:
TODO: Still found the bug, that when AUTO-UNLOCK is active before update. The re-activating of that feature during final recovery setup dialogs is still changening the TLS certs again.

rootzoll, 2019-04-27:
OK now tested and works.

closed 2019-04-27 by rootzoll

openoms, 2019-09-03:
Users have been reporting that the they could not connect Zap after chnaging their IP.
Manually resetting the TLS.cert helped.

Should this be automated every time when the publicIP is triggered to changed in lnd.conf
by the 
```
  # RECHECK PUBLIC IP
  # when public IP changes, restart LND with new IP
```
function of _background.sh?