raspiblitz/raspiblitz #328: LCD dosn't work. Black screen with white lines

Lit3block, 2019-02-18:
Hi guys, I'm trying to install and run my own raspiblitz but I've an issue with my lcd screen.
When the raspberry starts, the LCD starts, but the screen doesn't turn blue. It remains stuck on a black screen with white lines (similar to the terminal (see screenshot below )). How can I solve?

I tryed to reboot but nothing is change. Can enyone help me?

[image]

openoms, 2019-02-18:
@Lit3block your LCD seems to work well, but it has stuck with the setup process.  Did you have it set up fully previously? I would suggest restarting with a freshly burnt SDcard.

Lit3block, 2019-02-18:
HI @openoms; First of all. thanks for your reply.

> Did you have it set up fully previously?

yes I have already set it previously and it worked well. 

> I would suggest restarting with a freshly burnt SDcard.

Ok... so, should I start all over again (blockchain ftp transfer included?) ? 
I can do it, but if there are other possibilities I would prefer.

openoms, 2019-02-18:
can you access the RaspiBlitz through the terminal with ssh?
Changing only the SDcard would not require to download the blockchain again. It is the same as the update process: https://github.com/rootzoll/raspiblitz/blob/master/README.md#updating-to-new-version

Lit3block, 2019-02-19:
Upstate: I've finished to copy and sync the blockchain. i've reboot my rasp and the screen issue is disappear. Now, everything work well! Thanks @openoms.

openoms, 2019-02-19:
No problem, glad I could help, but it was you who have sorted it 👍

openoms, 2019-03-20:
I think this can be closed @rootzoll

closed 2019-03-22 by rootzoll