raspiblitz/raspiblitz #331: After updating to 1.0, Litecoin mainnet, no prompt to unlock wallet

goodc0re, 2019-02-19:
After updating to 1.0, Litecoin mainnet, no prompt to unlock wallet, instead this: 

[image] 

After exiting this screen with control-c, the prompt to unlock lnd appears, some users might not know how to proceed though.

rootzoll, 2019-02-19:
jikes - other litecoin mainnet updates were reported working - so not sure where this is coming from.

Would recommend the hard way of updating .. make backup of LND data dir, setup fresh Litecoin Blitz 1.0 and then play the backup back - see FAQ: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-backup-my-lightning-node

nyxnor, 2021-05-03:
pls close this

closed 2021-05-03 by rootzoll