raspiblitz/raspiblitz #332: Updating 1.0 version from 0.98

lucbran, 2019-02-19:
Hi,
downloaded new version and burnt sd card but unable to make it work:.
After I am asked for wallet pwd, I get to the LND synch page with no progress shown.After a while sys asking again wallet pwd.

This is my sys status..
admin@S:~ $ sudo systemctl status lnd
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-02-19 17:04:37 GMT; 5min ago
 Main PID: 6477 (lnd)
   CGroup: /system.slice/lnd.service
           └─6477 /usr/local/bin/lnd --externalip=

Feb 19 17:04:37 S.P.Q.R.Nexus systemd[1]: Started LND Lightning Daemon.
Feb 19 17:04:37 S.P.Q.R.Nexus lnd[6477]: 2019-02-19 17:04:37.871 [INF] LTND: Version
Feb 19 17:04:37 S.P.Q.R.Nexus lnd[6477]: 2019-02-19 17:04:37.871 [INF] LTND: Active 
Feb 19 17:04:37 S.P.Q.R.Nexus lnd[6477]: 2019-02-19 17:04:37.873 [INF] CHDB: Checkin
Feb 19 17:04:37 S.P.Q.R.Nexus lnd[6477]: 2019-02-19 17:04:37.945 [INF] RPCS: passwor
Feb 19 17:04:37 S.P.Q.R.Nexus lnd[6477]: 2019-02-19 17:04:37.957 [INF] RPCS: passwor
Feb 19 17:04:37 S.P.Q.R.Nexus lnd[6477]: 2019-02-19 17:04:37.957 [INF] LTND: Waiting
lines 1-14/14 (END)

rootzoll, 2019-02-19:
Your externalip on lnd startup is empty ... can you post the content for your raspiblitz config:
cat /mnt/hdd/raspiblitz.conf

lucbran, 2019-02-19:
Hello
after some attempts and re-entering my wallet password, it looks like it is now in progress of filtering blockchain.. Anyway here is my config (strangely still shows version 0.98):
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='0.98'
hostname=S.P.Q.R.Nexus
network=bitcoin
chain=main
autoPilot=off
autoNatDiscovery=off
runBehindTor=off
rtlWebinterface=off

publicIP=

publicIP=87.1.....

rootzoll, 2019-02-19:
Looks like a leftover from an old bug. You need to clean that one up.

open file for edit with
nano /mnt/hdd/raspiblitz.conf 

Delete all the empty lines and the line with "publicIP=". 
Store with CTRL+o and exit with CTRL+x
The reboot with: sudo shutdown -r now

lucbran, 2019-02-20:
Hi Christian,
ok cleaned conf file, rtebooted  and now setup finalized and I can access Raspiblitz Menu. I am starting testing each feature.
Started with wallet autounlock, set my passowrd, but when restarting I am stuck at : enter  LND wallet password (no way to go further after entering password).
I cleaned again conf file setting Autolock : Off , and I can access again but looks like something is wrong with this feature...

lucbran, 2019-02-22:
Hi tryiong to connect to peer (bitrefill) but when opening channel I am stuck at follwoing point:
COMMAND LINE: 
lncli --chain=bitcoin --network=mainnet openchannel 030c3f19d742ca294a55c00376b3b355c3c90d61c6b6b39554dbc7ac19b141c14f 300000 0

RESULT:
FAIL
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
[lncli] rpc error: code = Code(199) desc = chan size of 0.003 BTC is below min chan size of 0.15 BTC

What does this mean? thanks

rootzoll, 2019-02-22:
You need to fund the channel with more then 0.15 BTC .. bitrefill seems only wants people to open channels with much funding - no small channels.

rootzoll, 2019-02-22:
BTW I checked the AutoLock - I was not able to reproduce the error. So I will close this issue for now.

closed 2019-02-22 by rootzoll