raspiblitz/raspiblitz #336: keeps asking for PASSWORD C

playak, 2019-02-20:
Installed the latest SD image. Had to manually add the blockchain to my HDD, as the auto copy from a second HDD didn't work (yes it had its own power, and did copy the blockchain, but then I had even more problems).

So I copied the whole HDD myself, partly from my Mac full node, and off course set the owner to bitcoin (1002.1002) for all files.

Now the system starts with a nice blue screen, saying:
                                                                           ┌──────────────── bitcoin / main ───────────────────┐
                                                                           │  Lightning Filtering Blockchain                   │  
                                                                           │  Progress: ?/563929                               │  
                                                                           │  Please wait - this can take some long time.      │  
                                                                           │  Its OK to close terminal and ssh back in later.

, but then after a minute or so, it switches to a balck screen again, saying:
****************************************************************************
Unlock LND Wallet --> lncli --chain=bitcoin unlock
****************************************************************************
HELP: Enter your PASSWORD C
You may wait some seconds until you get asked for password.
****************************************************************************
Input wallet password: 

lnd successfully unlocked!
please wait ... update to next screen can be slow

Then back to the blue screen, etc etc. Tail of the lnd log file says:

2019-02-20 19:34:49.666 [INF] LNWL: Opened wallet
2019-02-20 19:34:50.633 [INF] LTND: Primary chain is set to: bitcoin
2019-02-20 19:34:50.912 [INF] LTND: Shutdown complete
2019-02-20 19:35:51.376 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-02-20 19:35:51.376 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-02-20 19:35:51.377 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-02-20 19:35:51.382 [INF] RPCS: password RPC server listening on [::]:10009
2019-02-20 19:35:51.383 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-02-20 19:35:51.384 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

lncli unlock works fine though.

Any idea what causes this anyone?

rootzoll, 2019-02-20:
If you get a repeatly Password C but seeing that unlocking was successfull ... it means that LND is shutting down after unlock and gets restarted by systemd and asks you again tu unlock.

You can see the shutdown in the logs at 2019-02-20 19:34:50.912 ... but I can not see the reason why it is shutting down in the logs. 

Can you post the complete output of XXdebugLogs.sh ?

playak, 2019-02-20:
Thanks. Here is that complete output:

$ ./XXdebugLogs.sh

*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 22:49:38 up  6:19,  3 users,  load average: 0.25, 0.23, 0.21

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-02-20 16:30:02 GMT; 6h ago
  Process: 779 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 788 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─788 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Feb 20 16:30:01 raspihelx2 systemd[1]: Starting Bitcoin daemon...
Feb 20 16:30:02 raspihelx2 systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-02-20T20:02:53Z UpdateTip: new best=0000000000000000000b348ee3cfe241384a3b3923f9f6b3ea239ceb8c1f6619 height=563932 version=0x2000e000 log2_work=90.368195 tx=384966360 date='2019-02-20T20:02:43Z' progress=1.000000 cache=20.1MiB(180110txo) warning='39 of last 100 blocks have unexpected version'
2019-02-20T20:10:40Z UpdateTip: new best=0000000000000000001e6cf2c6888c297d6ed612555cf936ea24b452d669f0b3 height=563933 version=0x20400000 log2_work=90.368218 tx=384969254 date='2019-02-20T20:10:15Z' progress=1.000000 cache=20.6MiB(184312txo) warning='39 of last 100 blocks have unexpected version'
2019-02-20T20:20:59Z UpdateTip: new best=000000000000000000124e4f94e0921f33335cdd217dfebaeb0a2ff126bb321c height=563934 version=0x20800000 log2_work=90.368242 tx=384972100 date='2019-02-20T20:20:23Z' progress=1.000000 cache=21.1MiB(189159txo) warning='39 of last 100 blocks have unexpected version'
2019-02-20T20:39:24Z UpdateTip: new best=00000000000000000024d762fb720529ec5efeea510011a82d53858f0fa4543d height=563935 version=0x3fffe000 log2_work=90.368265 tx=384975299 date='2019-02-20T20:39:13Z' progress=1.000000 cache=21.8MiB(195851txo) warning='40 of last 100 blocks have unexpected version'
2019-02-20T20:43:23Z Pre-allocating up to position 0x1100000 in rev01535.dat
2019-02-20T20:43:23Z UpdateTip: new best=000000000000000000277f9eb5878056eb23badc33c43378de05af8b49078ef4 height=563936 version=0x20000000 log2_work=90.368289 tx=384978225 date='2019-02-20T20:42:58Z' progress=1.000000 cache=22.1MiB(198422txo) warning='39 of last 100 blocks have unexpected version'
2019-02-20T20:45:24Z UpdateTip: new best=00000000000000000022f8601a6ab772f60bd26557cc74458044f98297ddf0c8 height=563937 version=0x20c00000 log2_work=90.368312 tx=384980893 date='2019-02-20T20:45:03Z' progress=1.000000 cache=22.3MiB(200735txo) warning='40 of last 100 blocks have unexpected version'
2019-02-20T21:23:34Z Leaving block file 1535: CBlockFileInfo(blocks=113, size=133464615, heights=563754...563937, time=2019-02-19...2019-02-20)
2019-02-20T21:23:34Z Pre-allocating up to position 0x1000000 in blk01536.dat
2019-02-20T21:23:34Z Pre-allocating up to position 0x100000 in rev01536.dat
2019-02-20T21:23:34Z UpdateTip: new best=0000000000000000002aabdce5e5c53bf87fccb60d985e927dec2a9052c7405a height=563938 version=0x20000000 log2_work=90.368336 tx=384984102 date='2019-02-20T21:23:29Z' progress=1.000000 cache=23.6MiB(212566txo) warning='39 of last 100 blocks have unexpected version'
2019-02-20T21:31:48Z UpdateTip: new best=0000000000000000000b3b136f33727d421b49dcd4db0db29b8a56893214e334 height=563939 version=0x20000000 log2_work=90.368359 tx=384987193 date='2019-02-20T21:31:32Z' progress=1.000000 cache=25.4MiB(221250txo) warning='38 of last 100 blocks have unexpected version'
2019-02-20T21:32:26Z UpdateTip: new best=000000000000000000086b45d70f3c775af5725bc378ed265bac235493d898df height=563940 version=0x20000000 log2_work=90.368383 tx=384989100 date='2019-02-20T21:32:20Z' progress=1.000000 cache=25.5MiB(222895txo) warning='38 of last 100 blocks have unexpected version'
2019-02-20T21:39:38Z UpdateTip: new best=00000000000000000028e1d6485078b31125c82ceb7a98e63282cf0e5a2f1fc7 height=563941 version=0x20000000 log2_work=90.368406 tx=384990576 date='2019-02-20T21:39:06Z' progress=1.000000 cache=26.1MiB(228470txo) warning='37 of last 100 blocks have unexpected version'
2019-02-20T21:56:14Z UpdateTip: new best=00000000000000000001d994d32e8081eb886c31b98944536647cd0fc031d636 height=563942 version=0x20000000 log2_work=90.36843 tx=384993382 date='2019-02-20T21:56:16Z' progress=1.000000 cache=26.8MiB(235144txo) warning='36 of last 100 blocks have unexpected version'
2019-02-20T22:09:39Z New outbound peer connected: version: 70015, blocks=563942, peer=36
2019-02-20T22:20:26Z Pre-allocating up to position 0x200000 in rev01536.dat
2019-02-20T22:20:27Z UpdateTip: new best=00000000000000000010fd10a17fa358d05c2a86f98faf65e19d66b0dde0c9ce height=563943 version=0x20000000 log2_work=90.368453 tx=384996230 date='2019-02-20T22:20:14Z' progress=1.000000 cache=27.6MiB(242487txo) warning='36 of last 100 blocks have unexpected version'
2019-02-20T22:24:00Z UpdateTip: new best=0000000000000000000878c20e41fcb8ddaec29cf88a868d47f024f3aefe2ae9 height=563944 version=0x20000000 log2_work=90.368477 tx=384999191 date='2019-02-20T22:23:30Z' progress=1.000000 cache=27.9MiB(245059txo) warning='36 of last 100 blocks have unexpected version'
[... cut ...]
Feb 20 19:35:51 raspihelx2 lnd[27479]: 2019-02-20 19:35:51.382 [INF] RPCS: password RPC server listening on [::]:10009
Feb 20 19:35:51 raspihelx2 lnd[27479]: 2019-02-20 19:35:51.383 [INF] RPCS: password gRPC proxy started at [::]:8080
Feb 20 19:35:51 raspihelx2 lnd[27479]: 2019-02-20 19:35:51.384 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config

rootzoll, 2019-02-21:
OK this seems to be the point where LND shutsdown

> Feb 20 19:34:50 raspihelx2 lnd[21174]: unable to create chain control: unable to connect to bitcoind: status code: 401, response: ""
> Feb 20 19:34:50 raspihelx2 lnd[21174]: 2019-02-20 19:34:50.912 [...]

LND cannot connect to bitdoind .. please check if both configs contain the same RPC password:
´sudo cat /mnt/hdd/bitcoin/bitcoin.conf´
´sudo cat /mnt/hdd/lnd/lnd.conf´

playak, 2019-02-21:
Great, thanks a ton! The manual copy of /mnt/hdd/bitcoin/bitcoin.conf from my Mac obviously still had a non matching config file in it, but I had made a backup of the blockchain directory that was generated by Raspiblitz. so I still had the correct config file to restore. LN catching up nicely now.

bhavman2, 2019-04-02:
[image]

Having a similar issue where it prompts for PASSWORD C but does nothing after entering the password.. requires you to quit to cmd line and manually launch main menu.

fluidvoice, 2019-04-03:
> [image]
> 
> Having a similar issue where it prompts for PASSWORD C but does nothing after entering the password.. requires you to quit to cmd line and manually launch main menu.

Better than just reporting your problem would be your results from doing the things done above, including providing the debug log output.

vmavromatis, 2020-05-30:
I am having the same issue (keeps asking for password C):

```

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.5
chainnetwork: bitcoin / main
 22:26:35 up  1:43,  2 users,  load average: 3.46, 3.19, 3.10

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-05-30 20:46:12 BST; 1h 40min ago
  Process: 3077 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 3084 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 3192 (bitcoind)
    Tasks: 18 (limit: 2200)
   Memory: 412.8M
   CGroup: /system.slice/bitcoind.service
           └─3192 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

May 30 20:46:12 vasilightning systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory
May 30 20:46:12 vasilightning systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sat 2020-05-30 20:42:34 BST, end at Sat 2020-05-30 22:26:36 BST. --
May 30 20:46:12 vasilightning systemd[1]: Starting Bitcoin daemon...
May 30 20:46:12 vasilightning systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory
May 30 20:46:12 vasilightning systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-05-30T21:16:31Z UpdateTip: new best=00000000000000000005ad98ca19760e32e7bcd1c23a874369461073d5c00954 height=584568 version=0x24ebc000 log2_work=90.825584 tx=433067165 date='2019-07-08T22:51:25Z' progress=0.802359 cache=58.0MiB(510053txo)
2020-05-30T21:16:32Z New outbound peer connected: version: 70015, blocks=632373, peer=50 (full-relay)
2020-05-30T21:16:33Z Pre-allocating up to position 0x5000000 in blk01716.dat
2020-05-30T21:16:42Z Synchronizing blockheaders, height: 628533 (~99.35%)
2020-05-30T21:16:43Z Pre-allocating up to position 0x6000000 in blk01716.dat
2020-05-30T21:16:46Z Pre-allocating up to position 0x7000000 in blk01716.dat
2020-05-30T21:18:17Z UpdateTip: new best=00000000000000000006b7d1aedc80a0cecb75088a735dbebb6f4b76348d74a7 height=584569 version=0x2000e000 log2_work=90.825606 tx=433069033 date='2019-07-08T22:59:06Z' progress=0.802362 cache=58.6MiB(515272txo)
2020-05-30T21:18:23Z socket recv error Connection reset by peer (104)
2020-05-30T21:19:41Z UpdateTip: new best=0000000000000000000e7d6897e61059a8025a2eb066a8c10f5eff513009081c height=584570 version=0x20000000 log2_work=90.825629 tx=433070185 date='2019-07-08T23:03:01Z' progress=0.802363 cache=59.2MiB(520734txo)
2020-05-30T21:20:15Z UpdateTip: new best=0000000000000000001ad7b7f95b7968b3107d84590a5ec6d49bc0ab606d02b8 height=584571 version=0x20c00000 log2_work=90.825651 tx=433073038 date='2019-07-08T23:14:26Z' progress=0.802368 cache=59.7MiB(525334txo)
2020-05-30T21:21:13Z UpdateTip: new best=000000000000000000003041ed8a0ec7ce189c962a941b76c8a7d26a9738b4ae height=584572 version=0x20800000 log2_work=90.825674 tx=433074824 date='2019-07-08T23:19:26Z' progress=0.802371 cache=60.1MiB(529715txo)
2020-05-30T21:22:11Z Pre-allocating up to position 0xb00000 in rev01713.dat
2020-05-30T21:22:12Z UpdateTip: new best=0000000000000000001eeb540694769505e222c1fea190ec1825fb2e11f21507 height=584573 version=0x20000000 log2_work=90.825696 tx=433077780 date='2019-07-08T23:38:36Z' progress=0.802376 cache=60.7MiB(535348txo)
2020-05-30T21:22:55Z UpdateTip: new best=0000000000000000000fb319250dbc9d784b08ab54bc9a72fabf1be4d4053f74 height=584574 version=0x20800000 log2_work=90.825718 tx=433080620 date='2019-07-08T23:44:56Z' progress=0.802382 cache=61.2MiB(540028txo)
2020-05-30T21:23:30Z UpdateTip: new best=00000000000000000007898911b7ee2895ab8e7beabb79b54f864e0b6f4d62e6 height=584575 version=0x20c00000 log2_work=90.825741 tx=433083366 date='2019-07-09T00:02:05Z' progress=0.802386 cache=61.7MiB(544118txo)
2020-05-30T21:24:00Z UpdateTip: new best=00000000000000000009e19831516aa0fc869f43e37e2aa162bbcd65291a7fd7 height=584576 version=0x20c00000 log2_work=90.825763 tx=433086305 date='2019-07-09T00:07:02Z' progress=0.802392 cache=62.1MiB(547801txo)
2020-05-30T21:24:35Z UpdateTip: new best=00000000000000000003bfa6dd0d24ec6c785ce31c406b9c0f11dd87d49e9283 height=584577 version=0x20c00000 log2_work=90.825786 tx=433089475 date='2019-07-09T00:26:00Z' progress=0.802397 cache=62.7MiB(553540txo)
2020-05-30T21:25:09Z UpdateTip: new best=000000000000000000225ddda6f87f10ea5c3738e6cefd992b8af3a832708f83 height=584578 version=0x20000000 log2_work=90.825808 tx=433092485 date='2019-07-09T00:44:02Z' progress=0.802403 cache=63.2MiB(557810txo)
2020-05-30T21:25:09Z ping timeout: 1224.469318s
2020-05-30T21:25:55Z UpdateTip: new best=0000000000000000000aebcd9a561a86af433e6c1fcb0fc9f5b804fb097e616f height=584579 version=0x20000000 log2_work=90.82583 tx=433095596 date='2019-07-09T00:53:16Z' progress=0.802408 cache=63.5MiB(561350txo)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
[... cut ...]
syncProgress=80.24 
startcountLightning=1
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=111

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999
```

It's my old raspiblitz disk (1.1 or something?). I just flashed the 1.5 version and plugged it.

Any ideas??

rootzoll, 2020-06-04:
"Keeps asking for Password C" means that the LND has problems running and is always stopping, restarts and needs you to unlock the wallet again. So lets take a look what problem LND has in the logs.

The uptime of LND is "1h 40min" in your logs ... so the restart seems not that often. In the logs I can see no Error messages .. (because it seems not in the last 30 messages). I would need you to monitor the LND debug logs a bit more closer while the error is happening ... have a second SSH terminal open and run: `sudo tail /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log`

LukeDamn, 2020-06-04:
OMG
I.m having the same problem. Yesterday, after updating to v1.5.1 by Patch, the reboot was ok. Then I changed the Touchscreen settings to ON and Password C was not OK anymore. I tried everyting and conacted the Telegram group: no solution. 
The reason to patch to v 1.5.1. was because of the Dropbox error. I still don't see backup files in my Apps folder in dropbox. So no Static-Channel-Backup.... I think.

My debug file:   https://termbin.com/fwo3

[image]

After running: sudo tail /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
[image]

rootzoll, 2020-06-04:
@LukeDamn so your LND is waiting for an unlock. And its says that the password you tried was wrong. So this looks first of all like normal "wrong password" situation ... no error that makes LND restart. Turning on the Touchscreen is not able to change your password C.

Please try to type in the password on the command line (after CRTL+c) in the clear just once to see if there is any change on keyboard layout or setting to make sure its really stull the same password c like you were using during setting up your wallet.

Just a side note: Did you had on the v1.5 made the optional update to LND 0.10.0 before doing the patch?

About the DropBox Backup not working - would like to check this once your password c is working. But to be safe I would recommend doing a LND Rescue-File backup and download it to your laptop: run command `menu` and then REPAIR > BACKUP-LND

LukeDamn, 2020-06-04:
Hello @rootzoll,
Thank you for your fast awnser.
Backup went wel but I can't locate the backup file (tar.gz). I was not promed to save it somewhere. (I´m not trained in terminal use...:)

I did not dit the optional update to LND 0.10.0 

I´m really sure it the correct password. EXCEPT I was struggling with a German laptop (Dell XPS 9360) and switched the keyboard settings US to GERMAN. But I´m not sure it´s the reason .... when I type ´visible´ its the correct password. I tried several language and keyboard settings.... not found the one used when I made the passwords. But I have a feeling it has to be something stupid as this.

rootzoll, 2020-06-04:
changing keyboard layouts can be a critical. If you are not able to reconstruct the password I way to recover would be:

- unpack the backup file you downloaded (with a unzipper) - keep original backup file
- you find the `channel.backup` file then under `mnt/hdd/lnd/data/chain/bitcoin/mainnet`
- then reset LND: command `menu` > REPAIR > RESET-LND
- Choose "old wallet" from SEED+SCB and use your seed words and your channel.backup file
- All your channels wil be closed but your funds should get recovered that way

LukeDamn, 2020-06-05:
Thank you! I recovered LND.

I´m really not sure it has anything to do with the keyboard layout. I used the following sequence:

- typed the password on the .pdf recovery sheet with .pdf-editor (so I could see the password)
- directly typed the password in the terminal when asked (twice)
- used that password multiple times on 2 different PC´s and phone by terminal and RTL. Always able to login with the password

So it´s very unlikey that is ´only´ a keyboard issue. 

2 last questions: 
- The Seed words stay the same after this recovery? Or sould I generate a new phrase?
- The closed channels can be opened again? Do I need to fund them gain?

I own a Node for a 1,5 week now and I learned more of the Bitcoin and Lightning Network past week them the last 2 years. It is amazing!

rootzoll, 2020-06-05:
If you restore from SEED the words stay the same - but you can always change the Password C that is used by LND to encrypt your wallet.db for local data protection.

You can open new channels with the same nodes again if those nodes agree (what they normally do, but in the future some nodes might refuse to redo channels nodes that had downtimes) The old funds you had in those channels (minus closing transaction fees) should be back in your LND on-chain wallet if you used the static channel backup file (its not a 100% garantueed method to recover, but very practical). If those fund are back confirmed on the blockchain you can use them to fund the new channels again.

> I own a Node for a 1,5 week now and I learned more of the Bitcoin and Lightning Network past week them the last 2 years. It is amazing!

Thanks. Enjoy the journey :)

closed 2020-07-04 by rootzoll

datascient, 2021-07-15:
Same issue for me. I tried the above methods and it doesn't work, keeps asking the Password c. The logs are below 
[H[2J[3J
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.0
chainnetwork: bitcoin / main
 05:28:28 up 27 min,  2 users,  load average: 1.67, 1.59, 1.42

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-07-15 05:01:34 BST; 26min ago
  Process: 1702 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 1710 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1733 (bitcoind)
    Tasks: 13 (limit: 4533)
   CGroup: /system.slice/bitcoind.service
           └─1733 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid

Jul 15 05:01:34 raspberrypi systemd[1]: Starting Bitcoin daemon...
Jul 15 05:01:34 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2021-07-15 05:00:08 BST, end at Thu 2021-07-15 05:28:28 BST. --
Jul 15 05:01:34 raspberrypi systemd[1]: Starting Bitcoin daemon...
Jul 15 05:01:34 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-07-15T04:28:13Z Rolling forward 0000000000000000000ba779bf5b3d3469c57902b27e49fa397eb039b5d37390 (672767)
2021-07-15T04:28:14Z Rolling forward 0000000000000000000621464723aa4ef3cad91b809d7a048102f7bc1c535efc (672768)
2021-07-15T04:28:14Z Rolling forward 0000000000000000000b2f54c048cd08a4cdecd7aee379ff2553aceab984fedf (672769)
2021-07-15T04:28:15Z Rolling forward 0000000000000000000ae490572c9574e32d1b59dc6ff5f80162aeb5d288dec6 (672770)
2021-07-15T04:28:16Z Rolling forward 000000000000000000047a513b3901b999ae4b8170f9be873c18bed3f346bb5d (672771)
2021-07-15T04:28:16Z Rolling forward 000000000000000000072d2452a78e826f8279903a9eaa130bbe491e9675f7c4 (672772)
2021-07-15T04:28:17Z Rolling forward 0000000000000000000a8a65dadc0605360d0462aeba93ec264248c56ebe98e9 (672773)
2021-07-15T04:28:18Z Rolling forward 00000000000000000005b475a526df6e73156f8372ec4241d2a4ef2d82148c12 (672774)
2021-07-15T04:28:19Z Rolling forward 0000000000000000000b5c1c43feb57079ad78497e1d5b978cdd85c41644569a (672775)
2021-07-15T04:28:20Z Rolling forward 0000000000000000000bdc675f0aed780ebd3f602899cdcfea4799e0f0820402 (672776)
2021-07-15T04:28:20Z Rolling forward 0000000000000000000ae846fd85130a3ddee57fe59c1b211df073f2893e3ee6 (672777)
2021-07-15T04:28:21Z Rolling forward 000000000000000000045d4b182109496ec7e2a01c7af3543db307eb26fd2061 (672778)
2021-07-15T04:28:22Z Rolling forward 0000000000000000000510aabc03de4338701444184c6b7f45342fcee4142846 (672779)
2021-07-15T04:28:23Z Rolling forward 0000000000000000000503f14a01ad44d1be620a4fbd76ea9156e2f453f4b401 (672780)
2021-07-15T04:28:24Z Rolling forward 00000000000000000002405808584c4fbf69701947f3b91d628acf6646fe4f18 (672781)
2021-07-15T04:28:25Z Rolling forward 00000000000000000006e1c3332135cc9daa80b407a1760f03ddaca5ee35b3cd (672782)
2021-07-15T04:28:25Z Rolling forward 0000000000000000000904317cb86ab157edee0d9f6e15eb0c75477f79cab23c (672783)
2021-07-15T04:28:26Z Rolling forward 000000000000000000044485496009755568e8738b8b4f09f9897ebb7ccb1ea3 (672784)
2021-07-15T04:28:27Z Rolling forward 00000000000000000003e66945606ff0900ce229bc21fc5c6be24270c36d4471 (672785)
2021-07-15T04:28:27Z Rolling forward 0000000000000000000132ab26426622f28235cca44bc61a3c68acc744ea7daa (672786)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-07-15 05:16:37 BST; 11min ago
  Process: 19691 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 19693 (lnd)
    Tasks: 10 (limit: 4533)
   CGroup: /system.slice/lnd.service
           └─19693 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --tor.socks=9070 --tor.control=9071 --listen=127.0.0.1:9735

Jul 15 05:16:37 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Jul 15 05:16:37 raspberrypi systemd[1]: Started LND Lightning Daemon.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Thu 2021-07-15 05:00:08 BST, end at Thu 2021-07-15 05:28:28 BST. --
Jul 15 05:10:25 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 2.
Jul 15 05:10:25 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
[... 194 lines cut ...]

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='10.39.219.162'
tempCelsius='47.2'
uptime=1680
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
bitcoinErrorShort='Loading block index'
bitcoinErrorFull='error code: -28 error message: Loading block index...'
startcountLightning=0
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

rw1018, 2021-07-27:
is there any solution for this? I restarted the raspiblitz at 97% downloaded and now I cannot finish!

rootzoll, 2021-08-17:
@datascient as long as your blockchain is on the "Rolling forward" the LND will not be ready and will keep asking for password C. This is a waiting situation. Please report back if you atre still stuck in that loop when there are no "Rolling forward" reports from bitcoind anymore.