raspiblitz/raspiblitz #337: Zeus mobile Wallet

rootzoll, 2019-02-21:
https://github.com/ZeusLN/zeus

OpenSource and works with Raspi LND ... if some likes to write a connect script like the ´97addMobileWalletShango.sh´ - that would be great :D

openoms, 2019-02-21:
There is no modification needed. It takes the macaroon as hex and we can export that from the menu already. Needs the REST port forwarded on the router if accessing from outside of the home network, https://twitter.com/openoms/status/1098562853483802624

openoms, 2019-02-21:
Showing the selected macaroon on the RaspiBlitzas as a QR code would be still great for future ease of use.

jochemin, 2019-02-27:
I wouldn't show admin.macaroon QR on the screen. Very sensitive file.

openoms, 2019-02-27:
You are right on that, but this could be only one of the options to export the macaroon. Not displayed continuously, only when requested specifically by a secure ssh login. At the moment we have options to export via secure copy (scp) or to be shown as a hex string in the terminal.

ZeusLN, 2019-02-28:
Looks like we can use lndconnect pretty easily. I'm going to investigate and report back.

kaloudis, 2019-03-03:
Zeus now has basic lndconnect support. We should be able to use the Zap connect shell file and just pass in --port 8080 to get the QR we need

kaloudis, 2019-03-03:
https://github.com/rootzoll/raspiblitz/pull/390

I need to make one tweak to how Zeus handle's lndconnect's macaroon format, but once that is fixed this should do the trick

rootzoll, 2019-03-04:
I merged #390 into dev branch ... to test you can set your plain raspiblitz v1.0 to dev code base witch calling on the terminal: `/home/admin/XXsyncScripts.sh dev` .. then you should see the ZEUS option in the Mobile App section of the raspiblitz.

TODO: Test before 1.1 release.

kaloudis, 2019-03-05:
@rootzoll please note that there is a bug in Zeus currently with how we handle the macaroons. I've fixed it with this PR: https://github.com/ZeusLN/zeus/pull/43

We plan to cut version 0.0.6 either Tuesday or Wednesday night to fix this.

openoms, 2019-03-06:
> I merged #390 into dev branch ... to test you can set your plain raspiblitz v1.0 to dev code base witch calling on the terminal: `/home/admin/XXsyncScripts.sh dev` .. then you should see the ZEUS option in the Mobile App section of the raspiblitz.

This worked great on my Odroid XU4 node on DietPi with the new Zeus: https://github.com/ZeusLN/zeus/releases/tag/v0.0.6

closed 2019-09-09 by rootzoll