raspiblitz/raspiblitz #340: IP address shown in red even though IP / port is reachable from the outside

rytisbitcoin, 2019-02-21:
I have forwarded ports 8333, 9735 and 10009 on my router. On the screen it says I have 20 connections (so my node is clearly accepting incoming connections), yet the IP address (and the LN node address) are shown in red.

vmavromatis, 2019-02-21:
Stupid question - is it actually red or orange? Mine was red, then I forwarded the ports 8333, 9735 and 10009 to 192.168.1.41 (local IP for raspi) and rebooted. After the reboot the address is orange but I have 5 peers, so I guess my node is open..? BTW, I couldn't find my node in 1ML.com so maybe I'm doing something wrong (using DynDNS).

rytisbitcoin, 2019-02-21:
It is actually red.

vmavromatis, 2019-02-21:
OK then I'm telling you the steps that worked for me in case it helps: forward those 3 ports to local IP, enable uPnP and reboot raspiblitz.

rytisbitcoin, 2019-02-21:
Do you mean enable uPnP in my router?

vmavromatis, 2019-02-21:
No I meant in the router - can't help more than that I'm afraid. Maybe someone else can.

rootzoll, 2019-02-22:
@rytisbitcoin the detection is not perfect yet .. see the FAQ topics on that: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#why-is-my-bitcoin-ip-on-the-display-red

The code on the detection can be found in `/home/admin/00infoBlitz.sh` - its that part thats dicides on the color:
```
   public_check=$(nc -z -w6 ${public_ip} 9735 2>/dev/null; echo $?)
  if [ $public_check = "0" ]; then
    # only set yellow/normal because netcat can only say that the port is open - not that it points to this device for sure
    ln_publicColor="${color_yellow}"
  else
    ln_publicColor="${color_red}"
  fi
```

So you can try on the terminal for example with your public ip `nc -z -w6 91.65.34.69 9735; echo $?` ... it will return the exit code of netcat. If anybody has an idea how to improve this, let me know.

rytisbitcoin, 2019-02-23:
The example you mentioned is related to the LN done IP, but the color is also red for Bitcoin node IP (i.e. where is says "Public ***.***.***.***:**** ** connections", so using the IP in your example it shows "Public 91.65.34.69:8333 20 connections". The words "Public" and "connections" are white, "20" is purple, "91.65.34.69:8333" is red. Because it has 20 connections, it is clearly reachable from the outside, so it should be green?

rytisbitcoin, 2019-02-23:
I have done some more testing.

Using the IP from your example (91.65.34.69), the command nc -z -w6 91.65.34.69 9735; echo $? returns "0" (i.e. the port shows as open), but using my own IP instead (nc -z -w6 xxx.xxx.xxx.xxx 9735; echo $?) it returns "1". However, when testing via a service like canyouseeme.org, all the relevant ports on my IP (8333, 9735, 10009) are shown as open. It seems that this is the issue with using netcat specifically to test if the ports are open, however I was not able to find a working alternative, and therefore do not have a solution for this issue.

rootzoll, 2019-02-24:
@rytisbitcoin much thanks for your research so far. 

If there is any working alternative showing up or an idea ... maybe another unix command that can be installed with apt-get, etc ... please add to this issue.

rytisbitcoin, 2019-02-24:
Did not solve the issue, but some more information to add: when trying nc -z -w6 xxx.xxx.xxx.xxx 9735; echo $? directly from the node itself, it returns a "connection refused" messaged immediately. When trying it from my computer (on the same network/IP), it returns "timed out" message after timing out.

rootzoll, 2019-02-27:
I added this issue to the v1.1 milestone to at least take some time in the comming weeks to see/research if there can be found a better solution to this.

CommanderPoe, 2019-02-27:
Having the complete oposite situation here... 
admin ip (internal) is green
Public ip (external) yellow 
Uri is yellow as well

But when i go to bitnodes and type my external ip it prints that my node is unreachable, however according to the pi info im getting 8 connections. Any ideas what could be the reason for that?

I opened the ports 8333 and 9735 on my router.

rootzoll, 2019-03-02:
TODO: Also check for correct port. When the port of LND in the config is set to 9736 ... the displaystill seems to check for 9735.

sirenak2, 2019-03-07:
Sorry to chime in on this but i'm having the exact same issue as Rytisbitcoin. Both bitcoin node and Lightning node are red. Bitcoin node has 23 connections, all ports are open using http://canyouseeme.org . I have inbound peers on RTL showing as true. nc -z -w6 xxx.xxx.xxx.xxx 9735; echo $? returns "1" . Thought i'd reply rather than make a new thread. Been holding off putting a photo on twitter of my completed Raspiblitz as I'm not sure it's all finished yet lol. I can't find my node using https://explorer.acinq.co/ . Anyyway, I love everything so far and hope I can get this last issue sorted soon!

rootzoll, 2019-03-07:
@sirenak2 if you know your ports are open and you dont want it red on the display you can simply hack the displayscript and set it green yourself:

Edit  `/home/admin/00infoBlitz.sh` - and repalce this code ...
```
   public_check=$(nc -z -w6 ${public_ip} 9735 2>/dev/null; echo $?)
  if [ $public_check = "0" ]; then
    # only set yellow/normal because netcat can only say that the port is open - not that it points to this device for sure
    ln_publicColor="${color_yellow}"
  else
    ln_publicColor="${color_red}"
  fi
```

with 

```
    ln_publicColor="${color_green}"
```

until we find a better solution ;)

sirenak2, 2019-03-08:
Haha, sorted. Thanks :)

jay653, 2019-03-10:
my info screen looks like the screen.

all ports are open ( https://canyouseeme.org/ ) [i checked 8333, 9735, 10009)

if i look up the ip at 1ml.com or https://explorer.acinq.co/ i cant find my node.

if i look up the ip at https://bitnodes.earn.com/ i find my node (  84.153.110.7:8333 /Satoshi:0.17.0.1/ with green checkmark) 

I was able to open a channel with my eclair mobile wallet now (so channel count is now 1/1), but it sucks a little since you cant recieve payments with mobile eclair yet.

but now to my question:
Why cant i find my node on 1ml.com or similar services?

[image]

sirenak2, 2019-03-10:
It took over 24 hours for mine to show on 1ml.com and a little longer for it to show on https://explorer.acinq.co.

fluidvoice, 2019-03-10:
This issue OP is not about 1ML. Open a new issue if you have other questions.

jay653, 2019-03-10:
my issue is also not about 1ML, i just used 1ML to check if my node is reachable from the outside (which is the OP's issue).

thx @sirenak2 for clearing this up. you guys can consider my issue solved.

justinsane, 2019-03-30:
Am I good to go per my screenshot? 1ML doesn't find anything but https://canyouseeme.org/ does show port 8333 open. I manually set port forwarding on my router open for 8333, 9735, and 10009.

Thanks in advance!

[image]

rootzoll, 2019-04-02:
@justinsane your address is shown in "yellow" not "red" - so verything looks good here.

TODO reminder to myself for future release:
Also check for correct port. When the port of LND in the config is set to 9736 ... the displaystill seems to check for 9735.

rootzoll, 2019-04-09:
LND port is know shown correct on the LCD from v1.2 on.

TODO for later release: write a python port scanner to check of the correct/reachble service is running on that port.

rytisbitcoin, 2019-05-20:
Another suggestion on how to fix this (at least for Bitcoin node IP address and port): if the number of Bitcoin node connections is >=9, then show it in green, because this can only happen if the IP and port are reachable from outside (this can be in addition to the current method for choosing color, and act as a failsafe).

intorid, 2019-06-08:
> Начиная с версии 1.2, порт LND отображается правильно на ЖК-дисплее.
> 
> TODO для более поздней версии: напишите сканер портов Python, чтобы проверить, работает ли на этом порту правильный / доступный сервис.

I think I found out the reason for the wrong definition shown in red. 
I would like to know which ports Netcat checks for status colors.
I used the command nc -z -w6 xxx.xxx.xxx.xxx PORT; echo $? for ports 8333 and 9735 what other ports are needed for status 0 

**The problem is related to the lack of support for the NAT function Loopback / Hairpin NAT**

I enabled the option Hairpin NAT on my router, now I check the status of the command in the ssh determines 0 if I start with the status of the screen, if I also check the status of the ssh then the screen is green. On RPi status is still red! o_O

rootzoll, 2019-06-28:
sorry because of time issue - moving back to future release

odudex, 2020-03-07:
I'm on V1.4, forwarded ports and I am able to create channels and use them, but only from outside my local network. Address is red, and running nc -z -w6 xxx.xxx.xxx.xxx 9735 from pi returns 1.
Running a wallet in my phone, I can use the channel from my node only with 4G and other's wifi. Once connected to my local network wifi channel goes offline.
Is there something I can do about it? Some configuration in my router?

radar24, 2020-03-11:
so is yellow good?? mine also shows yellow. all the ports are open and mine is reachable on bitnodes.io but yellow. I'd like to have it green though. Should I just manually replce the codes?

rootzoll, 2020-03-11:
yellow means - should work - see: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#why-is-my-node-address-on-the-display-yellow-not-green

odudex, 2020-03-13:
For those with same problem I showed above: Red address, can connect from outside LAN, but can't connect from within, this is due to lack of "Hairpinning" feature on router:
https://en.wikipedia.org/wiki/Hairpinning

rootzoll, 2020-03-16:
@odudex thanks - added the info about haqirpinning to the FAQ

In general I think the port forwarding will get more and more less important when we move the RaspiBlitz to TOR by default (TOR does not need any port forwarding).

NilByte, 2020-11-26:
I have a Thomson DOCSIS 3.0 cable modem from ISP that actually also have some some port forwarding and port triggering functionality. Connected to the cable modem I have a Linksys router and opening ports on the Linksys router didn't work as expected. 

Figured out that setting the Linksys router ip as DMZ host in cable modem made all the port forwarding in Linksys router work as expected and made both port 8333 and 9735 reachable.

So I finally set up port forwarding in the cable modem (8333, 9735, 10009, 8080) to Linksys router ip and also opened these ports in Linksys router. Then turned DMZ host off in cable modem again. 

Lessons learned for me is to check if the cable modem is causing the trouble on node not being reachable in spite all port have been opened in the router connected to the cable modem.

marxenegls, 2021-09-30:
i too have a new DOCSIS 3.0 cable modem from ISP with ipv6 support and Linksys +tp-link

i have been struggling for days over the same issue i've even lost connection to RTL Lit etc . i have eventually checked my dyndns with both opendns and noip ended up having to be reset, everything noip lost total access to local hosts, public ip and opendns had another ip.. i didn't check it, but i have a feeling it could have been an he.net tunnel, which i also ended up having to close. 

issues started as soon as i enabled wifi to my home router, which supports ipv6, my lab router doesn't, yet it's where i want to run the node off, although the idea of ipv6 is tempting for an hour i had it running and noticed connections doubled too- for now at least i 'll leave it at that, acclimatizing on lab stage. 
yes i had added dhclient.conf entry for wlan0. checked isp gateway for ip fragmentation, upnp, firewalls and its own port forwarding rules..

 now that i've reset dns, both dyn dns services, reset tls certs by tlsautorefresh=1 - i can login RTL,  etc again, channels CSB enabled to USB, funds still okay -i have a green light, but the public ip is now showing as 0.0.0.0, if i enabled autonat it switches back to red, If i enter the public ip manually it turns red, if i merely enter lnd's ip, the host's ip as opposed to 0.0.0.0, it  goes red. I mean i could understand it if i still had both nics enabled but now it's linked via eth0 - there isn't a second ap.
i did the restart after the clean up... 

 Noip sees my public ip, and so does Opendns and linked hosts. but not lnd..

I'm at a loss - i've looked into hardware nat , autonat, nat etc I've even placed the node in a DMZ -even tried ip_forwarding, nothing, i was considering enabling tor to see what that gives, but it could just as well complicate things further -  if whatever prevents public ip discovery, took an even bigger issue with onion rings and torrc  : p

 I only forwarded 9735 both [for now local access only] via RTL mainly,still behind gateway and router, so forwarding port (8333, 9735, 10009, 8080) isn't necessary, plus the router is in a DMZ, so port forwarding shouldn't be all that critical. No mobile access yet, i tried hooking both Zeus and Zap, both fail- and testing purposes until this ip is sorted i don't particularly care to publicise the issue... wait what did i just say ? 
  OOhhh.. that's why i'm only here now

marxenegls, 2021-10-01:
autonat .. upnp  both causing issues-  
i have the same DocisIII gateway, all ports are forwarded - disabled filtering of ip grag upnp .. second router where the node is on sits in DMZ only has 9735 forwarded
disabling autonat still the same but at least it loads up with a yellow ip and logo

.even if i enter public ip or and lnd's, wait 30sec for dhcp,  it goes red from yellow but, still doesn't show public ip,  i end up with a 0.0.0.0 or 127.0.0.1 for an ip  -though my dyndns sees the public ip .
 tor is not yet enabled 
i 've compiled and installed noip's duc [ddns client] & private domain, had to no raspbian bin ..still doesn't see public ip 
....i'm at a loss

-
today;
i think i found the issue,  one possible reason would be - avahi-daemon's .local,  conflicts with domain check this out : https://unix.stackexchange.com/questions/352237/avahi-daemon-and-local-domain-issues