raspiblitz/raspiblitz #341: 3.2" Case vs 3.5" Screen?

aaronn, 2019-02-21:
The linked case says it snaps around a 3.2" screen, but the linked screen in the reference shopping list is 3.5". Just double checking – is this correct and fits despite what's listed?

raumi75, 2019-02-21:
You can‘t put the top frame on becaus the display is too large, but that‘s not really important. It will look like the photo in the readme.

aaronn, 2019-02-22:
Got it. Thank you.

closed 2019-02-22 by aaronn

dhniels, 2019-03-25:
The linked display in the guide is a 3.2", not a 3.5" and I can confirm the frame does fit around the 3.2" display. The one in the photos of the Readme appears to be the 3.5" model (frame doesnt fit, doesnt have the 3 buttons on the end.) As a result the text on the status screen wraps. [Here is a photo](http://magaimg.net/img/7kj4.jpg). Not sure if theres a way to remedy this?

rootzoll, 2019-03-25:
On testing I gave up on the 3.2" because of this text wrapp problem. 3.5" working fine and on the case just ignore the topping - looks good without it :)

openoms, 2019-03-26:
@dhniels

openoms, 2019-03-26:
you can also run `sudo  dpkg-reconfigure console-setup` to choose and keep the setting after reboot

dhniels, 2019-03-27:
@openoms brilliant, thank you! the smallest font I could see was `Uni3-Terminus12x6.psf.gz` which still wraps just a couple characters, but at least the status screen is readable now.

openoms, 2019-03-27:
This is open source, so amything is possible :) . There should be plenty of other fonts available.
The easiest is still probably what you suggest is to edit the screen contents.
You need to edit the `00infoBlitz.sh`.

Run the inbuilt text editor with:
 `sudo nano 00infoBlitz.sh`
press the down arrow to scroll to the bottom and you will find the ASCII art.
To save press: CTRL+O, CTRL+X to exit.
Might need to restart to take effect.

dhniels, 2019-03-28:
Thanks for pointing me to the file! I was able to get it to fit on the 3.2” display by changing to the Uni3-Terminus12x6 font, removed 3 characters from the dashed line, removed the spaces between the dash next to “free mem”, and removing “net” from the bitcoin line. Here is a photo of [my 3.2” RaspiBlitz](http://magaimg.net/img/7lvt.jpg) for kicks!