raspiblitz/raspiblitz #342: BUILD FAILED --> Bitcoin download PGP author not OK

StifflersMoM69, 2019-02-21:
Get this error when running build_sdcard.sh

rootzoll, 2019-02-22:
Can you post the lines before that to see what failed ... sometimes some download is failing and then the failsafe is triggered.

StifflersMoM69, 2019-02-22:
Seems all fine to me.
`*** BITCOIN ***
mkdir: cannot create directory ‘/home/admin/download’: File exists
--2019-02-22 19:53:27--  https://bitcoin.org/bin/bitcoin-core-0.17.0.1/bitcoin-0                                                          .17.0.1-arm-linux-gnueabihf.tar.gz
Resolving bitcoin.org (bitcoin.org)... 138.68.248.245
Connecting to bitcoin.org (bitcoin.org)|138.68.248.245|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 24306145 (23M) [application/octet-stream]
Saving to: ‘bitcoin-0.17.0.1-arm-linux-gnueabihf.tar.gz’

bitcoin-0.17.0.1-arm-linux-gnueabi 100%[================================================================>]  23.18M   206KB/s   in 1m 59s

2019-02-22 19:55:27 (200 KB/s) - ‘bitcoin-0.17.0.1-arm-linux-gnueabihf.tar.gz’ saved [24306145/24306145]

--2019-02-22 19:55:27--  https://bitcoin.org/laanwj-releases.asc
Resolving bitcoin.org (bitcoin.org)... 138.68.248.245
Connecting to bitcoin.org (bitcoin.org)|138.68.248.245|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17940 (18K) [application/octet-stream]
Saving to: ‘laanwj-releases.asc’

laanwj-releases.asc                100%[================================================================>]  17.52K  --.-KB/s   in 0s

2019-02-22 19:55:28 (34.9 MB/s) - ‘laanwj-releases.asc’ saved [17940/17940]
`

rootzoll, 2019-02-22:
I just did a build of an sd card to test this again ... it worked on my machine - the basic script seems ok.

This is the part of the code where the error gets created.
```
laanwjPGP="01EA5486DE18A882D4C2684590C8019E36C2E964"
[....]
fingerprint=$(gpg ./laanwj-releases.asc 2>/dev/null | grep "${laanwjPGP}" -c)
if [ ${fingerprint} -lt 1 ]; then
  echo ""
  echo "!!! BUILD FAILED --> Bitcoin download PGP author not OK"
  exit 1
fi
```

So can you try the command on the terminal and let me know the output?
``gpg ./laanwj-releases.asc``

StifflersMoM69, 2019-02-22:
Looks also fine to me

`pub  4096R/36C2E964 2015-06-24 Wladimir J. van der Laan (Bitcoin Core binary rellsease signing key) <laanwj@gmail.com>`

rootzoll, 2019-02-22:
On my testmachine its:
```
pub   rsa4096 2015-06-24 [SC] [expired: 2019-02-14]
      01EA5486DE18A882D4C2684590C8019E36C2E964
uid           Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>
```

dont know with that output is different. 

I made the continue of the build script optional on the dev branch - it will still make a warning, but offer you to continue ... so use this command to run the build on a fresh raspbian:
``wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh && sudo bash build_sdcard.sh 'dev' ``

Let me know if that worked for you.

StifflersMoM69, 2019-02-23:
Hi! 
Almost.

!!! BUILD WARNING --> Bitcoin PGP author not as expected
Should contain laanwjPGP: 01EA5486DE18A882D4C2684590C8019E36C2E964
PRESS ENTER to TAKE THE RISK if you think all is OK

gpg: /root/.gnupg/trustdb.gpg: trustdb created
gpg: key 36C2E964: public key "Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
gpg: no ultimately trusted keys found
--2019-02-23 08:24:10--  https://bitcoin.org/bin/bitcoin-core-0.17.0.1/SHA256SUMS.asc
Resolving bitcoin.org (bitcoin.org)... 138.68.248.245
Connecting to bitcoin.org (bitcoin.org)|138.68.248.245|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1979 (1.9K) [application/octet-stream]
Saving to: ‘SHA256SUMS.asc’

SHA256SUMS.asc      100%[=====================>]   1.93K  --.-KB/s   in 0s

2019-02-23 08:24:10 (6.73 MB/s) - ‘SHA256SUMS.asc’ saved [1979/1979]

goodSignature(1)
correctKey(0)

!!! BUILD FAILED --> LND PGP Verify not OK / signatute(1) verify(0)

rootzoll, 2019-02-23:
hmm thats strange - are you using the following base image to build on:

Raspbian Stretch with desktop
Image with desktop based on Debian Stretch
Version: November 2018
Release date: 2018-11-13

It seems that there is that "gpg: no ultimately trusted keys found" problem .. but its a bit strange, because on my test machine (and also others in the past) this did not happend.

Ca you try manually (befor starting build script) this solution?
https://trog.qgl.org/20091030/troubleshooting-gnupg-gpg-no-ultimately-trusted-keys-found/

StifflersMoM69, 2019-02-23:
Honestly i am not sure anymore. I guess used the LITE-Version of Raspbian long time ago. I will begin from scratch.
I couldnt try the solution, because i was not able to create a key and change the trustlevel.

Thanks for your help!

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll