raspiblitz/raspiblitz #343: LCD Screen sold out need alternative

shatterreality, 2019-02-21:
LCD Screen sold out need alternative , US

rootzoll, 2019-02-22:
Does anbody from the US can propose a alternative/compatible shopping item on the US amazon or any other US store to order from? You can post here and I can update the US shopping list.

rootzoll, 2019-02-22:
You can test if compatible simply replacing the actual LCD and boot up RaspiBlitz again.

rootzoll, 2019-02-22:
As a TODO for version 1.1 release: Make LCD more flexible by config script: Switch easy between HDMI and LCD and maybe install/switch different driver that can be executed on first login to the device.

PapaCoderNit, 2019-02-22:
i bought 2 of these.. and working perfectly find with 1.0 https://www.amazon.com/gp/product/B01CNJVG8K/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1

PapaCoderNit, 2019-02-22:
> Does anbody from the US can propose a alternative/compatible shopping item on the US amazon or any other US store to order from? You can post here and I can update the US shopping list.
i bought 2 of these.. and working perfectly find with 1.0 https://www.amazon.com/gp/product/B01CNJVG8K/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1

rootzoll, 2019-02-22:
@PapaCoderNit much thanks ... added the link to the US shopping list

fcruz00, 2019-02-22:
I bought the Quimat version in amazon France. Found a similar one in US

https://www.amazon.com/Quimat-320x480-Resolution-Protective-Raspberry/dp/B06X191RX7/ref=sr_1_1?ie=UTF8&qid=1550850793&sr=8-1&keywords=Raspberry%2BPi%2BQuimat%2BTouch%2BScreen&th=1

Original link from amazon.fr

https://www.amazon.fr/gp/product/B06X191RX7/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

rootzoll, 2019-02-22:
@fcruz00 that looks good :) I added it as a third option. 

I think with 3 options on the LCD - I think we can close this issue. Much thanks for all that input :D

closed 2019-02-22 by rootzoll

BigFish47, 2019-02-22:
I bought this one from ebay this morning.
https://www.ebay.com/itm/3-5-Inch-320x480-Raspberry-Pi-TFT-Touch-Screen-LCD-Display-Monitor-Black-Case/253941862653?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

thehapax, 2019-02-23:
@rootzoll  my preferred electronics vendor online is sparkfun. While you can get parts from amazon, you don't have the quality of service, support and documentation if something goes wrong. I have not yet purchased this LCD, but it looks compatible https://www.sparkfun.com/products/14776

rootzoll, 2019-02-23:
@thehapax yepp that LCD looks like the waveshare one ... i will add it to the list. Much thanks.