raspiblitz/raspiblitz #344: Number of connections not shown on the screen anymore if dynamic DNS is enabled

rytisbitcoin, 2019-02-21:
The line with IP address of the node and the number of connections the node has, gets replaced by the address of the dynamic DNS. At least the number of connections the node has should still be shown.

rootzoll, 2019-02-22:
It was a quick decision to make room for long dnydns strings and not break the layout of the screen. An improvement could be for version 1.1 to check on the actual length of the dyndns string and only to remove the num of connections when dyndns string needs the space. Feel free to send a PR on this.

raumi75, 2019-02-22:
I'm too dumb to create a pull request for this commit: https://github.com/raumi75/raspiblitz/commit/3ca18633bf39d176916b4631260db57eb1843d70

Maybe it is because my previous pull request is not merged, yet.

Anyhow, that should fix the issue. The code to shorten that line was already there, but it was only used for IPv6 addresses. My fix used it for dyndns addresses, too.

rytisbitcoin, 2019-05-20:
I tried it on v1.2, but it still behaves as I described, while this says it should have been fixed in v1.1?

rootzoll, 2019-05-27:
@rytisbitcoin The fixes that get commited here may go into a future realease. Please check again after v1.3 release - this issue is currently on the list (but may get pushed).

rytisbitcoin, 2019-06-07:
A quick update - I used a shorter dynamic DNS address, and now it does show the number of connections as well. So it seems whether the number of connections is shown is dependent on the length of the dynamic DNS address.

rootzoll, 2019-06-18:
The fix from raumi is part of the v1.3 ... looks good on my side. Closing issue.

closed 2019-06-18 by rootzoll