raspiblitz/raspiblitz #345: Pressing "Esc" in the "Services" window removes all enabled services

rytisbitcoin, 2019-02-21:
When the "Services" window is opened (when connected to the node via SSH), clicking "Esc" on the keyboard not only deselects all services, but also confirms the "decision", and restarts the node with all services disabled. It should definitely not do that just by a single click without any confirmations. (I clicked it expecting it would cancel / close the window, and instead lost all the configuration of all the services I have previously set up)

rootzoll, 2019-02-22:
Thanks for the testing. I use the "dialog" tool/command for the service screen.
https://www.linuxjournal.com/article/2807

The ESC situation is not optimal. Also a lot of people were confused that the single services need to be selected with SPACE first before you hit ENTER.

There is a alternative to "dialog" called "whiptail" ... for the version 1.1 release it would be an idea to quick check if a change on the services to whiptail can ease some pain:
https://www.techrepublic.com/blog/linux-and-open-source/how-to-use-whiptail-to-write-interactive-shell-scripts/

raumi75, 2019-02-23:
Maybe a problem is: "When the user confirms their selections, a list of the choices is printed to stderr." (whiptail and dialog behave like this)

You can't tell if there is an empty list or if the user exited without confirming.

Maybe instead of checklist, we need a regular menu that says "install RTL" / "uninstall RTL" depending on the current state. This makes the action a little more clear to the user while at the same time communicating if it is installed or configured.

openoms, 2021-11-08:
this was solved. The dynamic choices are also implemented in some cases, but UX focus moved to the webUI

closed 2021-11-08 by openoms