raspiblitz/raspiblitz #348: progress during reindex shows 2.669697476678993e-09

goodc0re, 2019-02-23:
I shut down my mainnet bitcoin raspiblitz via the main menu Power Off (with open channels) and made sure to wait until it was powered down. 

Prepared the sd card with v 1.0 and after starting the raspiblitz again I noticed:

```
!! DETECTED NEED FOR RE-INDEX in debug.log ... starting repair script.
making sure services are not running ..
```

then

```
*************************
REINDEXING BLOCKCHAIN
*************************
Sat Feb 23 11:44:46 GMT 2019
THIS CAN TAKE SOME VERY LONG TIME
See Raspiblitz FAQ: https://github.com/rootzoll/raspiblitz
On question: My blockchain data is corrupted - what can I do?
If you dont see any progress after 24h keep X pressed to stop.
error code: -28
error message:
Loading block index...

RUNNING: 
PROGRESS: 

You can close terminal while reindex is running..
But you have to login again to check if ready.
```

tail -n1000 -f raspiblitz.log shows

```
***********************************************
Running RaspiBlitz Bootstrap 1.0
Sat 23 Feb 11:42:50 GMT 2019
***********************************************
Check if HDD contains configuration ..
load and update publicIP
update value (77.22.46.135)
DONE BOOTSTRAP
```

Now when I start the main menu I see: 

```
*************************
REINDEXING BLOCKCHAIN
*************************
Sat Feb 23 11:55:06 GMT 2019
THIS CAN TAKE SOME VERY LONG TIME
See Raspiblitz FAQ: https://github.com/rootzoll/raspiblitz
On question: My blockchain data is corrupted - what can I do?
If you dont see any progress after 24h keep X pressed to stop.

RUNNING: true
PROGRESS: 2.669695697709938e-09

You can close terminal while reindex is running..
But you have to login again to check if ready.

```

The progress doesn't look right. 

Any advice on how to proceed? I should get the node up and running again soon because of the funds in the channels.

rootzoll, 2019-02-23:
Not sure why your bitcoind demanding a re-index .. thats really a long procedure.

There are faster options to repair this. Did you check on the FAQ: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#my-blockchain-data-is-corrupted---what-can-i-do

tryolly, 2019-02-24:
About 5 hours ago the exact same thing happened to me as well. I just wiped everything completely and started downloading the chain from the torrent again. :(

rootzoll, 2019-02-26:
@goodc0re how dit it worked out?

goodc0re, 2019-02-26:
I just let it run, it's at 20% now, I don't have another copy of the chain data.

rootzoll, 2019-02-26:
There is the expiremental way to redownload the chain data by torrent. You could stop re-index and call just call the torrent script: `/home/admin/50torrentHDD.sh` - test feed back would be welcome to decide if I put this as standard re-cover option to version 1.1

Smiggel, 2019-03-02:
Same here. Ran version 0.99.

I powered off my node using the menu option. Then prepared the sd card with the v1 version and booted again. Set my new password and restored my LN wallet with the original password.

First the reminded was at 2.64xxx and now it’s at 0.005. :-( So it’s reindexing everything again. :-(

gabrieldov, 2019-03-03:
> There is the expiremental way to redownload the chain data by torrent. You could stop re-index and call just call the torrent script: `/home/admin/50torrentHDD.sh` - test-feedback would be welcome to decide if I put this as standard re-cover option to version 1.1

This worked for me, had same issue when upgrading to 1.0 - you will probably need to verify your power source and go dig a bit more about it around here because your HDD should have failed/corrupted due to power issues. Hopefully we'll find a good and cheap solution soon - I'm waiting on some components to test

Smiggel, 2019-03-04:
What worked for me:
I downloaded the blockchain on my Mac. When it was done (within a day), I SFTP-ed into my node as Bitcoin with my password. Then I removed the blocks, chainstate and indexes folders. I uploaded the folders from my Mac and then rebooted. I left everything else as it was before. Then Bitcoin cached up and LN did too. Now it runs fine again.

Elmit2015, 2019-03-11:
I arrived here too. 
What does the progress number mean? it seeps to count down.

Elmit2015, 2019-03-11:
> What worked for me:
> I downloaded the blockchain on my Mac. When it was done (within a day), I SFTP-ed into my node as Bitcoin with my password. Then I removed the blocks, chainstate and indexes [...]

How do you do download on another computer?

Elmit2015, 2019-03-11:
That also creates the question for me about a backup of data to recover my funds!!!

Elmit2015, 2019-03-11:
I stopped to re-index and went straight to run /home/admin/50torrentHDD.sh to fetch a new block chain. 
It shows me now:
```
*** 1) Status Torrent 'blockchain':
cat: .blockchain.out: No such file of directory
... started ...

*** 2) Status Torrent 'update':
cat: .update.out: No such file of directory
... started ...
```
That picture is now already for a long time. What should I do?

fluidvoice, 2019-03-11:
try...
`sudo -u bitcoin /home/admin/50torrentHDD.sh`

Elmit2015, 2019-03-12:
Disappointing, .... 
Since I did not get an answer what to do, I started 
/home/admin/50downloadHDD.sh

after 35 hours it finished. 
I restarted the raspberry.

It had created the directories and filled
```
/mnt/hdd/bitcoin/blocks
/mnt/hdd/bitcoin/chainstate
/mnt/hdd/bitcoin/indexes

```with hundreds of files

I logged in a second time
```
**************************
REINDEXING BLICKCHAIN
**************************

Mon Mar 11 02:29:09 GMT 2019
THIS CAN TAKE SOME VERY LONG TIME
See Raspiblitz FAQ: https://github.com/rootzoll/raspiblitz
On question: My blockchain data is corrupted - what can I do?
If you dont see any progress after 24h keep X pressed to stop.
error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port

RUNNING:
PROGRESS:
```

I logged in a second time:
ps aux|grep bitcoin
```
root       807  0.5  3.8 108904 36336 ?        Ssl  22:39   0:04 /usr/bin/node /home/admin/RTL/rtl --lndir /home/admin/.lnd/data/chain/bitcoin/mainnet
admin    14157  0.0  0.0   4676   584 pts/1    S+   22:53   0:00 grep --color=auto bitcoin

```
What should I do now????

Elmit2015, 2019-03-12:
Maybe that can help to figure out more (last section of debug.log:

```
2019-03-12T22:41:11Z Bitcoin Core version v0.17.0.1 (release build)
2019-03-12T22:41:11Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-03-12T22:41:11Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-03-12T22:41:11Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-03-12T22:41:11Z Using the 'standard' SHA256 implementation
2019-03-12T22:41:11Z Default data directory /home/bitcoin/.bitcoin
2019-03-12T22:41:11Z Using data directory /home/bitcoin/.bitcoin
2019-03-12T22:41:11Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-03-12T22:41:11Z Using at most 40 automatic connections (128000 file descriptors available)
2019-03-12T22:41:11Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-03-12T22:41:12Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-03-12T22:41:12Z Using 4 threads for script verification
2019-03-12T22:41:12Z scheduler thread start
2019-03-12T22:41:12Z HTTP: creating work queue of depth 16
2019-03-12T22:41:12Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2019-03-12T22:41:12Z HTTP: starting 4 worker threads
2019-03-12T22:41:12Z Cache configuration:
2019-03-12T22:41:12Z * Using 2.0MiB for block index database
2019-03-12T22:41:12Z * Using 12.2MiB for transaction index database
2019-03-12T22:41:12Z * Using 8.0MiB for chain state database
2019-03-12T22:41:12Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-03-12T22:41:12Z init message: Loading block index...
2019-03-12T22:41:12Z Wiping LevelDB in /home/bitcoin/.bitcoin/blocks/index
2019-03-12T22:41:12Z boost::filesystem::status: Bad message: "/home/bitcoin/.bitcoin/blocks/index"
2019-03-12T22:41:12Z Error: Error opening block database
2019-03-12T22:41:12Z Shutdown: In progress...
2019-03-12T22:41:12Z scheduler thread interrupt
2019-03-12T22:41:12Z Shutdown: done

```

Elmit2015, 2019-03-13:
I tried now to start bitcoind by 
sudo /usr/local/bin/bitcoind

bitcoind started, but I see may lines with 
2019-03-13T00:01:29Z ThreadRPCServer incorrect password attempt from 127.0.0.1:43586

fluidvoice, 2019-03-13:
> I tried now to start bitcoind by
> sudo /usr/local/bin/bitcoind
> 
> bitcoind started, but I see many lines with
> 2019-03-13T00:01:29Z ThreadRPCServer incorrect password attempt from 127.0.0.1:43586

You cannot start bitcoind that way. Reboot and when it has started, SSH into it then ctrl-c to get to a command prompt then run `./XXdebugLogs.sh` and post the output here.

Elmit2015, 2019-03-13:
something is sever wrong now. The space on the SD card is getting smaller and smaller. I installed mc to get around faster, and found that I got now a directory /root/.bitcoin with the three direcotries blocks chainstate wallets.
I also got on /mnt/hdd/bitcoin directories:
blocks (March 10), chainstate (March 10), indexes (March 4), testnet3 (March 10) and w0189ab.kasserver.com (March 12) with again blocks (March 12), chainstate (March 12), indexes (March 12), and testnet3 (March 12)

Elmit2015, 2019-03-13:
admin@mystore:~ $ ./XXdebugLogs.sh

*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 02:43:18 up 0 min,  2 users,  load average: 0.50, 0.14, 0.05

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-03-13 02:42:53 GMT; 25s ago
  Process: 838 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 846 (code=exited, status=1/FAILURE)

Mar 13 02:42:53 mystore systemd[1]: bitcoind.service: Unit entered failed state.
Mar 13 02:42:53 mystore systemd[1]: bitcoind.service: Failed with result 'exit-code'.

*** LAST 20 CHAINNETWORK LOGS ***
2019-03-13T02:42:52Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-03-13T02:42:52Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-03-13T02:42:52Z Using 4 threads for script verification
2019-03-13T02:42:52Z scheduler thread start
2019-03-13T02:42:52Z HTTP: creating work queue of depth 16
2019-03-13T02:42:52Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2019-03-13T02:42:52Z HTTP: starting 4 worker threads
2019-03-13T02:42:52Z Cache configuration:
2019-03-13T02:42:52Z * Using 2.0MiB for block index database
2019-03-13T02:42:52Z * Using 12.2MiB for transaction index database
2019-03-13T02:42:52Z * Using 8.0MiB for chain state database
2019-03-13T02:42:52Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-03-13T02:42:52Z init message: Loading block index...
2019-03-13T02:42:52Z boost::filesystem::status: Bad message: "/mnt/hdd/bitcoin/blocks/index"
2019-03-13T02:42:52Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2019-03-13T02:42:52Z Aborted block database rebuild. Exiting.
2019-03-13T02:42:52Z Shutdown: In progress...
2019-03-13T02:42:53Z scheduler thread interrupt
2019-03-13T02:42:53Z Shutdown: done

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-03-13 02:42:52 GMT; 25s ago
 Main PID: 847 (lnd)
   CGroup: /system.slice/lnd.service
           └─847 /usr/local/bin/lnd --externalip=220.133.223.206:9736

Mar 13 02:43:05 mystore lnd[847]: 2019-03-13 02:43:05.544 [INF] RPCS: password gRPC proxy started at [::]:8080
Mar 13 02:43:05 mystore lnd[847]: 2019-03-13 02:43:05.545 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock… and unlock it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Wed 2019-03-13 02:42:19 GMT, end at Wed 2019-03-13 02:43:18 GMT. --
Mar 13 02:42:52 mystore systemd[1]: Started LND Lightning Daemon.
Mar 13 02:43:05 mystore lnd[847]: 2019-03-13 02:43:05.484 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
Mar 13 02:43:05 mystore lnd[847]: 2019-03-13 02:43:05.484 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Mar 13 02:43:05 mystore lnd[847]: 2019-03-13 02:43:05.529 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
Mar 13 02:43:05 mystore lnd[847]: 2019-03-13 02:43:05.543 [INF] RPCS: password RPC server listening on [::]:10009
Mar 13 02:43:05 mystore lnd[847]: 2019-03-13 02:43:05.544 [INF] RPCS: password gRPC proxy started at [::]:8080
Mar 13 02:43:05 mystore lnd[847]: 2019-03-13 02:43:05.545 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

*** LAST 20 RTL LOGS ***
-- Logs begin at Wed 2019-03-13 02:42:19 GMT, end at Wed 2019-03-13 02:43:18 GMT. --
Mar 13 02:42:52 mystore systemd[1]: Started RTL daemon.
Mar 13 02:43:12 mystore node[848]: Server is up and running, please open the UI at http://localhost:3000

Elmit2015, 2019-03-13:
Any chance to get an answer how to continue????

Elmit2015, 2019-03-13:
Latest try now is that I have logged in again:
cd /mnt/hdd/bitcoin

sudo su -

mkdir original
mv blocks  original/
mv chainstate  original/
mv indexes  original/
mv testnet3  original/

mv w0189aba.kasserver.com/blocks .
mv w0189aba.kasserver.com/chainstate .
mv w0189aba.kasserver.com/indexes .
mv w0189aba.kasserver.com/testnet3 .

shutdown -r now

---
login again and it is REINDEXING BLOCKCHAIN again
RUNNING: true
PROGRESS: 2.6433793831...e-09

Am I on the right track now????
After/if it will ever finish, I think I should remove /root/.bitcoind

fluidvoice, 2019-03-13:
yes, remove **/root/.bitcoind** and /admin/.bitcoind if they exist.
then run...
`sudo /home/admin/config.scripts/network.reindex.sh`
and wait for reindex to complete then reboot.

Elmit2015, 2019-03-13:
That resulted in the exactly the same picture ;-(
I cannot figure out what the PROGRESS shows.

However, sudo tail  /mnt/hdd/bitcoin/debug.log shows a progress of Reindexing file blk00144.dat to file blk00231.dat within exactly 30 minutes.

fluidvoice, 2019-03-13:
> That resulted in the exactly the same picture ;-(
> I cannot figure out what the PROGRESS shows.
> 
> However, sudo tail /mnt/hdd/bitcoin/debug.log shows a progress of Reindexing file blk00144.dat to file blk00231.dat within exactly 30 minutes.

The progress is likely a bug in the script. don't worry about it.
just run `sudo tail -f /mnt/hdd/bitcoin/debug.log` and wait till it's finished. Will take a LONG time.

Elmit2015, 2019-03-14:
What should I do now? After several hours/days of syncing, try tor, try download and sync again:

admin@mystore:/var/log$ date;sudo tail -f /mnt/hdd/bitcoin/debug.log 
```
Thu Mar 14 02:05:51 GMT 2019
2019-03-14T02:04:57Z UpdateTip: new best=000000000000065748bacb10a856cf95c2ed5f1bd8d3db1b99b9156c00993b05 height=135144 version=0x00000001 log2_work=65.039362 tx=970812 date='2011-07-07T08:05:33Z' progress=0.002565 cache=69.0MiB(581685txo)
2019-03-14T02:04:57Z Fatal LevelDB error: Corruption: block checksum mismatch: /home/bitcoin/.bitcoin/indexes/txindex/000045.ldb
2019-03-14T02:04:57Z You can use -debug=leveldb to get more complete diagnostic messages
2019-03-14T02:04:57Z 

************************
EXCEPTION: 15dbwrapper_error       
Fatal LevelDB error: Corruption: block checksum mismatch: /home/bitcoin/.bitcoin/indexes/txindex/000045.ldb       
bitcoin in scheduler       

```
... and here it stucks!!!

Elmit2015, 2019-03-14:
Anybody????

rootzoll, 2019-03-14:
@Elmit2015 Going with the "COPY" option to replace the blockchain data seems maybe the best option for you. But thats a bit more hand work and you would need to prepare a synced blockchain on your laptop first:

- download latest bitcoin core: https://bitcoincore.org/en/download/
- before starting it set `txindex=1` for that bitcoin core in `bitcoin.conf` (maybe needs to be created)
- then let bitcoind sync the blockchain (takes time but should be faster on your laptop)
- Wen sync is ready on your laptop start on your RaspiBlitz the `/home/admin/50copyHDD.sh` and follow the instructions

fluidvoice, 2019-03-14:
> @Elmit2015 Going with the "COPY" option to replace the blockchain data seems maybe the best option for you. But thats a bit more hand work and you would need to prepare a synced blockchain on your laptop first:
> 
> * download latest bitcoin core: https://bitcoincore.org/en/download/
> * before starting it set `txindex=1` for that bitcoin core in `bitcoin.conf` (maybe needs to be created)
> * then let bitcoind sync the blockchain (takes time but should be faster on your laptop)
> * Wen sync is ready on your laptop start on your RaspiBlitz the `/home/admin/50copyHDD.sh` and follow the instructions
> 
> The plus of that way: You have always a copy of the blockchain on your laptop then and can replace it in the most fast way possible in error situations.

Why is `txindex=1` being used?

Elmit2015, 2019-03-17:
I get more and more troubles, .... Kernel messages, ....  latest is that the Raspberry is not found on the network anymore. The LAN IP address isn't anymore, ... I purchased a new SD card, this time a better quality, I believe (SanDisk 32 GB).  I also prepared a powered USB hub, which might help to get enough power to the hard disk (just in case that was the problem for the clicking of the hard disk).

I read that upgrade is just a replacement of the SD card, and all data are on the hard disk. Assuming I get the new SD card to work, and find my Raspberry again, I expect that I can copy the bitcoin blockchain from my desktop over to the hard disk and that the Lightning is still in place to continue.

Is there anything I should take care of or be prepared for?

Smiggel, 2019-03-17:
@Elmit2015 Just make sure you download the blockchain on a different machine with in the config txindex=1.

When you have downloaded it, copy the folders blocks, chainstate and indexes it to your Pi as user “bitcoin”. When done, you can start Bitcoind. It can take a while before the Pi has cached up. Especially lnd can take a little while. Just be patient.

fluidvoice, 2019-03-17:
Perhaps connect your Pi to the router by ethernet instead of using WiFi just to remove that potential problem source. Maybe the IP address of your Pi changed. Use the following script to locate all nodes on your LAN, then you can see which are known (eg. router, notebook, Pi, etc.):
`nmap -v -sn 192.168.2.0/24 192.168.0.0/24 192.168.1.0/24`

fluidvoice, 2019-03-17:
> @Elmit2015 Just make sure you download the blockchain on a different machine with in the config txindex=1.
>
> When you have downloaded it, copy the folders blocks, chainstate and indexes it to your Pi [...]

I read this in the Readme:

> One requirement is that the blockchain is from another bitcoin-core client with version greater or equal to 0.17.1 with transaction index switched on (txindex=1 in the bitcoin.conf).

But do you know **WHY** txindex=1 is supposedly required?  My understanding is this is not required for LN.

Smiggel, 2019-03-17:
@fluidvoice for personal use you probable don’t need it. When you use your node as fullnode, it makes sure you download everything.

fluidvoice, 2019-03-17:
> @fluidvoice for personal use you probable don’t need it. When you use your node as fullnode, it makes sure you download everything.

I'm not sure it's even needed for full node.  Do you have a **recent** reference where it says this is required?

Elmit2015, 2019-03-18:
Now the next problem, .... 
I burned the new SD card, and the hard disk cannot be found.
I did backed up the directory /lnd before. I hope it is useful.

/lnd$ tree .
```
.
├── data
│   ├── chain
│   │   └── bitcoin
│   │       └── mainnet
│   │           ├── admin.macaroon
│   │           ├── invoice.macaroon
│   │           ├── macaroons.db
│   │           ├── readonly.macaroon
│   │           └── wallet.db
│   └── graph
│       └── mainnet
│           ├── channel.db
│           └── sphinxreplay.db
├── lnd.conf
├── logs
│   └── bitcoin
│       └── mainnet
│           ├── lnd.log
│           ├── lnd.log.494.gz
│           ├── lnd.log.495.gz
│           └── lnd.log.496.gz
├── tls.cert
└── tls.key

```
Am I right to do now:
1. start up with a new hard disk
2. use the same passwords as the first time
3. use the seeds (24 words)
4. copy the /lnd directory over to the hard disk

Do I need to take care of anything else?

Elmit2015, 2019-03-18:
Another followup glitch. Since the hard disk is not found, I cannot login (Connection refused).

Should I just re-burn the SD or can I reset it when I mount the SD card to my Linux system? I guess fail2ban locks me out now. Is there a way to whitelist my desktop, which is on the same LAN?

fluidvoice, 2019-03-18:
Connect the hard drive to another computer. Can you see it and re-initialize it (create a new **msdos** partition table)?  If not then change your USB cable, drive case, and/or HDD. If so, then you might have a bad Raspberry Pi as I cannot think of why it would not be able to recognize the HDD.
Yes, re-burning the SD card might help.

Elmit2015, 2019-03-19:
I could copy the data to the new hard disk, ....
Last what I saw was:
```
****************************************************************************
Helping Instructions --> for creating a new LND Wallet
****************************************************************************
A) For 'Wallet Password' use your PASSWORD C --> !! minimum 8 characters !!
B) Answere 'n' because you dont have a 'cipher seed mnemonic' (24 words) yet
C) For 'passphrase' to encrypt your 'cipher seed' use PASSWORD D (optional)
****************************************************************************

lncli --chain=bitcoin create
Input wallet password: xxxxx

Do you have an existing cipher seed mnemonic you want to use? (Enter y/n): ty
```
I accidentally entered 2 characters instead of just y to put into my seed words.
It resulted in a "ty not found" and stopped.

I tried to restart that dialog with "raspiblitz" which results in a picture like:
Lightning Filgering Blockchain
Progress: ?/567804
Please wait - this can take some long time.
Its OK to close termminal and ssh back in later.

It stucks there already an hour, and my blockchain was synced up to the last 30 records. 

admin@RaspiBlitz:/mnt/hdd $ sudo tail /mnt/hdd/bitcoin/debug.log
```
2019-03-19T10:05:31Z UpdateTip: new best=00000000000000000026851b483990550c0ea2e7ce6e951a0ebce3ebaac7b8cf height=567798 version=0x20000000 log2_work=90.456434 tx=393023395 date='2019-03-19T10:05:09Z' progress=1.000000 cache=31.7MiB(279955txo) warning='35 of last 100 blocks have unexpected version'
2019-03-19T10:07:01Z UpdateTip: new best=0000000000000000001ba2bb4be7b08573a4a189837612a00b34a8d3976fc0cf height=567799 version=0x20000000 log2_work=90.456456 tx=393023849 date='2019-03-19T10:06:44Z' progress=1.000000 cache=31.8MiB(281258txo) warning='34 of last 100 blocks have unexpected version'
2019-03-19T10:10:02Z UpdateTip: new best=000000000000000000261f5f8239e9d4e57c6b82491388bb439ac6cdb68a76a4 height=567800 version=0x20400000 log2_work=90.456478 tx=393024501 date='2019-03-19T10:09:16Z' progress=1.000000 cache=32.1MiB(283682txo) warning='34 of last 100 blocks have unexpected version'
2019-03-19T10:16:42Z Pre-allocating up to position 0x800000 in rev01565.dat
2019-03-19T10:16:42Z UpdateTip: new best=00000000000000000002e3061e059a31a57532494ac8ddf5831ad62da360c2d3 height=567801 version=0x20000000 log2_work=90.456501 tx=393025991 date='2019-03-19T10:16:42Z' progress=1.000000 cache=32.5MiB(287685txo) warning='34 of last 100 blocks have unexpected version'
2019-03-19T10:17:16Z UpdateTip: new best=0000000000000000001185c9dcb5e199136e3e2d4c56af3db3babdf8da93fdb0 height=567802 version=0x20c00000 log2_work=90.456523 tx=393026204 date='2019-03-19T10:17:01Z' progress=1.000000 cache=32.5MiB(287849txo) warning='34 of last 100 blocks have unexpected version'
2019-03-19T10:17:30Z UpdateTip: new best=0000000000000000001883999e66722c1457a04d0610ff0b3e1cc43b437bac48 height=567803 version=0x20000000 log2_work=90.456545 tx=393026268 date='2019-03-19T10:17:16Z' progress=1.000000 cache=32.5MiB(287938txo) warning='33 of last 100 blocks have unexpected version'
2019-03-19T10:37:52Z UpdateTip: new best=00000000000000000015b3bb044c856de72191f4f915a7b55df42b709522be4d height=567804 version=0x20000000 log2_work=90.456567 tx=393029154 date='2019-03-19T10:37:35Z' progress=1.000000 cache=33.1MiB(293547txo) warning='33 of last 100 blocks have unexpected version'
2019-03-19T11:14:08Z Potential stale tip detected, will try using extra outbound peer (last tip update: 2176 seconds ago)
2019-03-19T11:14:08Z New outbound peer connected: version: 70015, blocks=567804, peer=16

```

It seems it is synced up now, ... but why/where does it stuck? How can I enter my seeds again?

Elmit2015, 2019-03-19:
I figured out that I can start that again with /home/admin/70initLND.sh

but that ends with:
```
!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!

!!! FAIL !!! SOMETHING WENT WRONG:
[lncli] rpc error: code = Unknown desc = invalid passphrase
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Press ENTER to retry ... or CTRL-c to EXIT
```
If I try ENTER, ... it repeats, ...
What should I do now???

rootzoll, 2019-03-19:
@Elmit2015 its hard to get an overview to your case. When you have a backup of the LND data dir I would recommend to run `/home/admin/XXcleanHDD.sh` make a fresh sd card with the new released version 1.1 image and setup a fresh RaspiBlitz (no need to input old seed during setup). If you have a fresh Blitz working with zero funding and channels. Stop the LND and replace the LND data dir content completly with the old data (just dont delete `/mnt/hdd/lnd` directory itself to keep static link active). Once old LND data is replayed onto the HDD make reboot and the RaspiBlitz should run with your old LND data.

Elmit2015, 2019-03-20:
<rant>I am soon throwing away all. Sorry, it takes too long for your responds to get anything done.</rant>

Unfortunately I don't get any answer to my questions, so I am sure I do a lots of things wrong. 

I got so far that, RaspiBlitz is starting.
I copied over the backed up lnd directory. and restarted the box.

I tried then all command on the raspiblitz startup screen with these results:
Info:
```
               RaspiBlitz v1.0  mystore.elmit.com
               bitcoin Fullnode + Lightning Network 
        ,/     -------------------------------------------
      ,'/      load average: 2.04, 1.95, 1.93, CPU 64°C
    ,' /       Free Mem 541M / 927M  Free HDD 2.4T (88%)
  ,'  /_____,  ssh admin@192.168.178.185 ▼12.5MiB ▲6.7MiB
 .'____    ,'  
      /  ,'    bitcoin v0.17.0.1 mainnet Sync OK (100.00%)
     / ,'      Public 220.133.223.206:8333 8 connections
    /,'        
   /'          LND 0.5.2-beta wallet 816671 sat 
               0/14 Channels 1093731 sat 0 peers

037dda58c0e1b81237f1fecf4cea99e9aaa5918fdbd0fb87042219f0b008ad10c1@220.133.223.206:9735
```
0/14 Channels 1093731 sat 0 peers  seems odd
The node is the right one, as I had before. Seems the copy of lnd worked.

Funding
```
******************************
Fund your Blockchain Wallet
******************************

COMMAND LINE: 
lncli --chain=bitcoin --network=mainnet newaddress np2wkh

RESULT:
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
FAIL PRECHECK - lncli getinfo shows 'synced_to_chain': false - wait until chain is sync 

******************************
TODO
******************************
Send REAL Bitcoin to address --> 
Whats next? --> Wait for confirmations. You can use info on LCD to check if funds have arrived.
If you want your lighting node to open channels automatically, activate the 'Autopilot' under 'Activate/Deactivate Services'
```
No address!!!!

Services
I activated RTL Webinterface
While it installs I saw some errors:
```
Error: mkdir
/home/admin/RTL/node_modules/angular-devkit/build-angular/node_modules/node-sass/vendor

/home/admin/RTL/node_modules/node-sass/vendor

```

lnbalance
```
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"
[... 44 lines cut ...]
In a second ssh connection:
```
$ date;sudo tail -f /mnt/hdd/bitcoin/debug.log
Wed Mar 20 00:22:29 GMT 2019
2019-03-20T00:20:39Z 239 addresses found from DNS seeds
2019-03-20T00:20:39Z dnsseed thread exit
2019-03-20T00:20:53Z New outbound peer connected: version: 70015, blocks=567900, peer=2
2019-03-20T00:20:53Z New outbound peer connected: version: 70015, blocks=567900, peer=3
2019-03-20T00:20:54Z New outbound peer connected: version: 70015, blocks=567900, peer=4
2019-03-20T00:20:54Z New outbound peer connected: version: 70015, blocks=567900, peer=5
2019-03-20T00:20:56Z Imported mempool transactions from disk: 936 succeeded, 0 failed, 0 expired, 3 already there
2019-03-20T00:21:01Z New outbound peer connected: version: 70015, blocks=567900, peer=7
2019-03-20T00:21:02Z New outbound peer connected: version: 70015, blocks=567900, peer=8
2019-03-20T00:21:13Z New outbound peer connected: version: 70015, blocks=567900, peer=9
2019-03-20T00:22:34Z New outbound peer connected: version: 70015, blocks=567900, peer=10

```
What should I do next??? To burn sd and download again and again, seems not a solution. 
However, I would like to try 1.1 though.

Some suggestions for improvements:
1. if a script ends, put a note how to continue (maybe already on top)
2. if an invisible input (password, ...) - display stars or acknowledge "there was an input"
3. make an automatically switch, if there is no display, then display on the HDMI
4. get a Telegram group, or anything where we can faster tackle down a problem.

Elmit2015, 2019-03-20:
I made a new SD card. 
Login password was now raspiblitz

To start with it took a LONG time (about 50 minutes), then it rebooted
Now login password is neither raspiblitz nor my set password

ssh: connect to host xxxxxxx port 22: Connection refused

What should I do to get in and can continue with your answers for above message?

fluidvoice, 2019-03-20:
login via SSH, ctrl-c to get to command prompt, then run these and **post the resulting URL**:
`/home/admin/XXdebugLogs.sh > tempfile.txt`
'sudo tail -n 50 /mnt/hdd/bitcoin/debug.log >> tempfile.txt`
`sudo tail -n 500 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log >> tempfile.txt` 
`sudo cat tempfile.txt | nc termbin.com 9999 && rm tempfile.txt`

rootzoll, 2019-03-22:
@Elmit2015 Its sound like you ran a "Update/Recover" not a "Fresh Setup". 

BTW if you plan to go for a "store" (as I can see in the name of your RaspiBlitz) maybe a lightning node like the "Nodl" is a better fit for you: https://www.nodl.it .. it has better integration of BTCPay Server.

fluidvoice, 2019-03-22:
> @Elmit2015 Its sound like you ran a "Update/Recover" not a "Fresh Setup".
> 
> BTW if you plan to go for a "store" (as I can see in the name of your RaspiBlitz) maybe a lightning node like the "Nodl" is a better fit for you: https://www.nodl.it .. it has better integration of BTCPay Server.

Better integration or just more reliable?  ... the cheaper BTCPi (https://lightninginabox.co/product/btcpi/) is basically just a Pi that can run BTCPay Server and/or Raspiblitz with an external HDD.

rootzoll, 2019-03-22:
`reliable` in that sense, that you buy tested hardware from nodl.it - that reduces a lot of sources of errors. The BTCPi is also a good option if a BTCPayServer is the goal to run for a store.

fluidvoice, 2019-03-22:
> `reliable` in that sense, that you buy tested hardware from nodl.it - that reduces a lot of sources of errors. The BTCPi is also a good option if a BTCPayServer is the goal to run for a store.

from the Readme... "DietPi is an extremely lightweight Debian-based OS. With images starting at 400MB, that's 3x lighter than "Raspbian Lite". It is highly optimized for minimal CPU and RAM resource usage, ensuring your SBC always runs at its maximum potential."

openoms, 2021-03-17:
closing as outdated -> no torrent option any more, hardware updated to RPi4

closed 2021-03-17 by openoms