raspiblitz/raspiblitz #349: New install of V1.0 not getting very far.

OfficialBigAl, 2019-02-23:
Hi,

Just started from scratch, Clean HDD and new V1.0 image.
Upon startup, waits for 10 sec and displays a screen waiting for Bitcoind start-up.
Then message within the blue box states "./Home/bitcoin/" not found.
I gave it 10 minutes, the HDD is constantly showing activity but nothing is being written when I check it on another setup.

Could not connect via SHH as network connection refused - so could not get any logs.
Have reverted to V0.98 and it starts up fine?!?  Most strange.

Not really an issue, more of a FYI - as long as no one else is complaining.

Regards BigAl.

rootzoll, 2019-02-23:
Thanks for the heads up. 

How did you cleaned the HDD?

OfficialBigAl, 2019-02-23:
I removed and recreated the partition using Disk Manager on Windows10
Then reformat (FAT32) using GUIformat from Ridgecorp Consultants ltd.

rootzoll, 2019-02-26:
OK .. I will keep this issue open so that others can maybe report on it when experience the same. If I dont see a second report on that until 1.1 release I will close.

OfficialBigAl, 2019-03-03:
Tried again and all ok!  You can close this issue. Thank you.

closed 2019-03-13 by rootzoll