raspiblitz/raspiblitz #350: Static IP address?

CommanderPoe, 2019-02-24:
Used to have the Raspibolt config now playing with the raspiblitz I was wondering what happens with the internal IP address in case it changes for users that dont use LCD screen? 

I have not read anything in the guide regarding that issue.

rootzoll, 2019-02-24:
normally with the LCD its easy to see if local IP changes and with dynamic IP its more flexible because you can change networks easy. But I can add something about setting a static address to the FAQ?. What text would you recommend?

CommanderPoe, 2019-02-24:
Oh thanks a lot for the answer, unfortunately Im not a techie person, but that was one of the first things that came to my mind when i decided to play with this guide since i dont have an LCD screen, I have some friends who started using the raspiblitz configuration and they dont own an LCD screen but i didn't raised the question. I thought you were maybe the right person to ask this kind of questions in case someone ( like me and my friends ) decided to use the guide without the LCD screen.

rootzoll, 2019-02-24:
The RaspiBolt had a chapter of network settings ... including static IP:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#connecting-to-the-network

closed 2019-02-24 by CommanderPoe