raspiblitz/raspiblitz #353: Missing fonts or something

tryolly, 2019-02-24:
Hi, Not a major problem just slightly annoying. I'm accessing my raspiblitz from putty using windows 10 and the screen is never displayed properly. It has letters instead of borders. I'm sure I'm just missing the correct font for it or something, if so, which one do I need to get?
[image]

raumi75, 2019-02-24:
Looks like #314 which was submitted in German. Still looking for someone with putty-skills. Most people here seem to be using Mac and Linux.

tryolly, 2019-02-24:
thanks raumi, looks like the issue will get solved eventually. I can wait.

rootzoll, 2019-02-24:
@tryolly I dont have a putty win10 either to test - maybe try in putte settings to change the font of the terminal. Please rport back if you find anything.

OfficialBigAl, 2019-02-24:
You have to have the right emulation/dictionary ;)  Try these settings...
[image]

rootzoll, 2019-02-24:
I will add this to the FAQ soon - thanks :)

tryolly, 2019-02-24:
Thanks guys. That fixed it up perfectly.

closed 2019-10-03 by frennkie