raspiblitz/raspiblitz #354: FAIL: /mnt/hdd/raspiblitz.conf should exist at this point

gehrlein, 2019-02-24:
Hello everyone

I try to set up a raspiblitz where I copied the blockchain from my own PC to an external HDD (which is called BLOCKCHAIN in ext4). I gave both the ownership of the HDD and the "bitcoin" folder to "1002:1002".  

During the first step of building the sd_card I already get the error:

"cp: -r not specified; omitting directory '/home/admin/raspiblitz/home.admin/config.scripts'
/build_sdcard.sh: line 484: /home/admin/setup.sh: Permission denied
chmod: cannot access '/home/admin/setup.sh': No such file or directory"

However, everything still proceeds and I can set up my passwords, Afterwards,  I receive the error:

error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.
bitcoin still not running
ls: cannot access "/mnt/hdd/raspiblitz.conf": permission denied
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
FAIL: /mnt/hdd/raspiblitz.conf should exists at this point, but not found!
Please report to: ...  (the issue 293 which is already closed here)
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Press ENTER to EXIT

I looked into /mnt/hdd/ and I found the raspiblitz.conf but I had to change the permission to admin manually, since it belonged to 1002:1002.

I guess this issue is related to ownership? Did I make an error while setting up the HDD by giving ownership to 1002? Who needs it instead?

rootzoll, 2019-02-24:
Preparing a HDD for setup, runs too often into problems. Thats why there are two other options during the setup: 1) copy the blockchain data from another computer over the network 2) cloning the data from another (see README setup docu).

So I would recommend to format your HDD on another computer (FAT32) to have it clean, build a fresh sd card and then choose one of the standard option.

But I still wonder why you have those permission denied errors during the build .. let me know if that is still going on.

gehrlein, 2019-02-25:
Thanks for the reply. I changed all ownership of the HDD and subfolders to 777 and everything worked!

closed 2019-02-26 by rootzoll