raspiblitz/raspiblitz #357: ./indexes not located on my bitcoin folder

CommanderPoe, 2019-02-24:
Hey started fresh and i was about to copy the whole blockchain from another computer using the Option C of the guide it states that i should copy _./chainstate ./indexes ./testnet3 ./blocks bitcoin@192.168.1.53:/mnt/hdd/bitcoin_  but i only have within my bitcoin folder the chainstate and teh blocks directories the other 2 are missing... but does that even matter?

rootzoll, 2019-02-24:
is it a bitcoin version 0.17.0.1? And do you have txindex=1 in the config?
see FAQ - Chapter "I have the full blockchain on another computer. How do I copy it to the RaspiBlitz?": https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#i-have-the-full-blockchain-on-another-computer-how-do-i-copy-it-to-the-raspiblitz

You could use the blockchain you have, but it might take quite some time then for the raspiblitz to do the indexing and reversioning of the blockchain itself.

CommanderPoe, 2019-02-25:
Hey @rootzoll  i do have .17.1 version (for MAC) and i dont have either ( .indexes or .testnet3 ) i just finished copying the whole blockchain using that method (C) and received a message the copy was invalid or incomplete altho i made sure the blocks were up to date when i started copying... Any solutions to that problem? After the wait it asked me to clean the hdd and pick another method.

CommanderPoe, 2019-02-25:
just had to create the .conf on my bitcoin folder and few minutes lates the .indexes folder magically appeared

closed 2019-02-25 by CommanderPoe