raspiblitz/raspiblitz #358: v1.0 s error: Could not connect to the server 127.0.0.1:8332

osilloh, 2019-02-24:
First of all thank you all for putting this project together. Here is the background of what I have done. 

1. I had a version prior to .98 running partially months ago.
2. Downloaded 1.0 re-flashed SD
3. Wiped drive with clean script

Here are screen logs of my progress after the above. 

****************************************************
Monitoring Screen Session: Download
Progress: 100.32% (253827844 of 253000000)
If needed press key x to stop Download
NOTICE: This can take multiple hours or days !!
Its OK to close terminal now and SSH back in later.
****************************************************
w0189aba.kasserver. 100%[++++++++++++++++++++] 127.78M  --.-KB/s    in 0s
w0189aba.kasserver. 100%[++++++++++++++++++++] 127.92M  --.-KB/s    in 0s
w0189aba.kasserver. 100%[++++++++++++++++++++] 128.00M  --.-KB/s    in 0s
w0189aba.kasserver. 100%[++++++++++++++++++++] 127.82M  --.-KB/s    in 0s
w0189aba.kasserver. 100%[++++++++++++++++++++] 127.84M  --.-KB/s    in 0s
w0189aba.kasserver. 100%[++++++++++++++++++++] 127.97M  --.-KB/s    in 0s
w0189aba.kasserver. 100%[++++++++++++++++++++] 127.58M  --.-KB/s    in 0s
w0189aba.kasserver. 100%[++++++++++++++++++++] 127.95M  --.-KB/s    in 0s

Aborting Download
killing screen session PID(22301)
pkill: killing pid 22303 failed: Operation not permitted
proccesses killed
cleaning screen
path to downloaded data is w0189aba.kasserver.com/
final size is 253827844 of targeted size 253000000
*** Moving Files ***
OK

*** Checking HDD ***
OK - HDD is ready.

*** Prepare bitcoin ***
OK - bitcoin setup ready

20:52 on 2/23/19

*** Start bitcoin ***
This can take a while ..
Created symlink /etc/systemd/system/multi-user.target.wants/bitcoind.service → /etc/systemd/system/bitcoind.service.

09:23 on 2/24/19

error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.
bitcoin still not running

*** RASPIBLITZ LOGS ***
blitzversion: 1.0
chainnetwork: bitcoin / main
 19:23:14 up 1 day, 14:11,  2 users,  load average: 0.54, 0.48, 0.46

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST 20 CHAINNETWORK LOGS ***
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

*** LND SYSTEMD STATUS ***
Unit lnd.service could not be found.

*** LAST 20 LND LOGS ***
-- No entries --

- RTL is OFF by config
UNKOWN STATE - there is blockain data config, but blockchain service is not running
It seems that something went wrong during sync/download/copy of the blockchain.
Or something with the config is not correct.
Sometimes a reboot helps --> sudo shutdown -r now
admin@RaspiBlitz:~ $

rootzoll, 2019-02-26:
format your HDD on another computer to FAT(32) and naming it TEST and then try again fresh (new sd card image).

closed 2019-02-26 by rootzoll

osilloh, 2019-03-09:
Thanks, here was my path to where I am now. I did the above and I could not get the torrent option to work. Did the FTP option again, got stuck in the same place. Finally decided to open firewall torrent ports and ports to torrent trackers. Did the above steps again with firewall ports open.

[image]

This allowed the bittorrent option to work! The chain downloaded much faster than FTP, but I am now stuck on the Lightning Filtering blockchain issue with @merlinmason

rootzoll, 2019-03-10:
OK added the info about opening the ports tfor torrent o the issue to remove FTP from v1.1