raspiblitz/raspiblitz #360: fsck the harddisk

raumi75, 2019-02-25:
The hard drive `/dev/sda1` alias `/mnt/hdd` does not get checked routinely during boot because in `/etc/fstab` the line ends with `0 0`.

Is there a reason for that? Considering how important the data integrity is on that drive, would it make sense to have a menu item that stops all services, unmounts the disk and runs `fsck` interactively?

rootzoll, 2019-02-26:
I took this from the RaspiBolt tutorial: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#mounting-external-hard-disk

Maybe get in contact with @Stadicus3000 for background on that decision.

I am all for improving on data integrity and very open to any optimization - even it it costs a bit speed. Adding to issue to next release.

raumi75, 2019-02-26:
I'll contact @Straticus3000 on twitter. According to this:

https://wiki.ubuntuusers.de/Dateisystemcheck/#berpruefung-abschalten

the last number should be a `2` so the disk will get checked routinely (after root partition). I will try it out, talk to Straticus and report back.

raumi75, 2019-02-26:
According to @stadicus3000 there is no real reason for the setting.

For science, I changed mine to `0 2`. That did not force a fsck during reboot.

You can check the number of mounts by running `sudo tune2fs -l /dev/sda1`

It was set to `-1`, which means never.

Then I manually changed it by runnung `tune2fs -c 5 /dev/sda1` (meaning every 5th time it will get checked)

The actual checking was a matter of seconds and the display was on. There were no errors on the disk and lost+found was empty. A good feeling. 

[image]

rootzoll, 2019-02-26:
Even if it adds 20seconds on startup I have no problem with it if it adds data integrety. 

So help me for the take away here ... for v1.1 I would:
1. add the HDD with /etc/fstab ending on 2 
2. run a `sudo tune2fs -c 1 /dev/sda1` once
correct?

rootzoll, 2019-02-26:
BTW what will haven if the chack fails? How can the RaspiBlitz react to that situation?

raumi75, 2019-02-27:
Yes, that's exactly what I think you should do.

About failed checks I'm not 100% sure. To the best of my knowledge, the system will try to repair minor errors. E. g. place a broken file into `lost+found`

https://raspberrypi.stackexchange.com/questions/61723/raspberry-pi-3-and-raspbian-jessie-how-to-run-fsck-at-boot

If the hdd is so badly damaged that fsck and boot can't continue, you probably wouldn't want the system to boot anyhow. That a problem an administrator hates. As with every server, this is an exception that calls for a manual intervention. I'd recommend the user to take the disk and connect it to a Linux system and try to fsck it manually.

That gives me an idea. What files are really precious? The whole `lnd` folder is only 200 MB in my case. How about copying it to the boot sd when you shutdown the pi (00mainMenu.sh OFF)? Or do we only need the channels.db file?

rootzoll, 2019-02-27:
OK that sounds good. Will add this for v1.1. I think that will add alot - much thanks.

To protect the "precious data" is the aim of the additional USB-BTRFS-Raid .. so I think before we make extra copies to the sd we check if this already covers it.

BTW should the USB-BTRFS-Raid also have fsck switched on the same way as the HDD?

raumi75, 2019-02-27:
No. 

`btrfs scrub start /mnt/raid/`is better than fsck. Will start a check on a mounted partition. It's done in the background and checks each checksum of every block. If one is wrong, Btrfs will fix it by taking the block from the good disk and copy it to the bad one. You could run this periodically with cron or as a menu item.

rootzoll, 2019-03-17:
In v1.1 the HDD is now mounted with fsck - checking in every boot. It adds maybe 20secs to the startup, but that looks good. Closing the issue.

closed 2019-03-17 by rootzoll