raspiblitz/raspiblitz #361: Bitcoin FTP Stuck

LightningK0ala, 2019-02-25:
I left FTP download running overnight and it seems to have gotten stuck at 102.14%?

[image]

When I first ssh'd into the machine this morning it didn't go into the screen session, it took me to the wizard, asked me to continue with the setup and I had to choose FTP again to see the output in the screenshot.

Running raspibolt 1.0.0.

LightningK0ala, 2019-02-25:
[image]

Seems like it's still downloading stuff, but lots of smaller sized files?

LightningK0ala, 2019-02-25:
Might just be a progress indication issue.

LightningK0ala, 2019-02-25:
[image]

I think it's fine now.

LightningK0ala, 2019-02-25:
[image]

CommanderPoe, 2019-02-25:
How long did it take you to synch the blockchain with that option? I tried the 4th option "Getting the blockchain from another machine" and it failed to me either cuz the indexes folder was missing or the process got corrupted somehow during the procces.

rootzoll, 2019-02-26:
@LightningK0ala yikes - looks like the FTP download got corupted and the Blitz throws you back at the re-index-repair option. re-indexing is more like a last resort ... I would recommend to stop re-indexing and start the torrent download -> use script `/home/admin/50torrentHDD.sh`

@CommanderPoe did the SCP transfere got interrupted or was the data after SCP corrupted?

TODO for next release: Maybe remove the FTP download its not delivering the data quality needed and now with torrent well enough seeded its maybe better to remove.

CommanderPoe, 2019-02-26:
@rootzoll Definitely data corrupted, this added to the missing .Indexes folder which i finally managed to get modifying the bitcoin.conf on my bitcoin folder... Will give it a try to that method again tonight after i format the HDD in FAT32.. Thanks as usual.

LightningK0ala, 2019-02-26:
@CommanderPoe Not sure exactly how long it took, maybe 16 hours?

@LightningK0ala I gave up on FTP and went the torrent route. It worked flawlessly, took about a day to sync.

rootzoll, 2019-02-26:
@LightningK0ala I think that finalizes my plans to drop the FTP from the menu for next v1.1 release. This centralized shit aint no good anyways and will save me some hosting fees ;)

@CommanderPoe Ok .. I always thought SCP was kind of reliable on data transfer - maybe it it got corrupted during storage to HDD - not sure ... will check on how to optimize that for the future.

LightningK0ala, 2019-02-27:
@LightningK0ala Awesome 😄 . In that case I think we can close this issue.

rootzoll, 2019-02-27:
I will keep issue open as a TODO: remove the FTP from menu in version v1.1

fkochdev, 2019-03-01:
I tried the torrent but ended up with REINDEXING BLOCKCHAIN after downloads completed. Currently at 2.6 % - any idea?

rootzoll, 2019-03-01:
torrent is doing a check of all data at the end of download .. so if that passed there might be something more troublesome with your setup. I would recommend to start fresh:

* format hdd on another computer with fat(32) name test
* make a fresh sd card from, v1.0 image

This time make sure to give HDD some extra power. For example add a powered USB hub between Pi and HDD or add a second power source with a Y-cable: https://en.wikipedia.org/wiki/Y-cable#USB

fkochdev, 2019-03-01:
Thing is, I started with a fresh setup like recommended, I may try again next week.

fluidvoice, 2019-03-01:
> torrent is doing a check of all data at the end of download .. so if that passed there might be something more troublesome with your setup. I would recommend to start fresh:
>
> * format hdd on another [...]

Maybe the FTP download script should be changed and the file should be broken up into 4 or more pieces then check the downloaded file's MD5 hash. Or at least check the result even if it's just one big file.
`wget ftp://yourfile.zip && wget ftp://yourfile.zip.md5 && md5sum -c *.md5`

Also there is a boat load of options with `wget`. See the linux man pages. Perhaps the script can be made more robust to handle errors.

rootzoll, 2019-03-06:
@fluidvoice thanks for the idea to improve the FTP option. But now that we also have the COPY option and the Torrent is quite good seeded i really like to get rid of the FTP to reduce complexity, maintenace and those FTP server bill of 25 bucks a month ;)

rootzoll, 2019-03-10:
Just a side note: Add an info about the torrent if its not starting to connect that maybe opening of ports might help: https://github.com/rootzoll/raspiblitz/issues/358

rootzoll, 2019-03-14:
OK. FTP option is removed on v1.1. Closing issue.

closed 2019-03-14 by rootzoll