raspiblitz/raspiblitz #363: Getting permission denied when trying to sign back in with ssh while downloading blockchain

sirenak2, 2019-02-25:
Hi, ordered my kit yesterday. Followed the guide and managed to get the blockchain downloading. Shut down ssh as I was under the impression I could and went out for the evening. Came back and tried signing in using ssh admin@myip and my password and received permission denied. after many failed attempts thinking i messed up up my password in creation I wiped my hdd and m sd and re flashed and started again. Rather than just leaving it running I decided to shut down ssh and attempt to log bk in only to be confronted by the same issue, permission denied. Should I just leave my pc on for the blockchain to be downloaded or am I doing something wrong here? 

Win 10 using ssh to connect to raspiblitz via ethernet . thanks and sorry for what is most probably a noob question. I just don't want to wast another 8 hours

vmavromatis, 2019-02-25:
What do you mean "shut down ssh"? Closed the terminal window?

Also, what's the full error for "permission denied"? If it's complaining about your RSA fingerprint, then just do `rm .ssh/known_hosts`. 

Have you tried restarting? I don't see how syncing the blockchain will solve your login issues...

sirenak2, 2019-02-25:
Yes sorry, closed the terminal window. The full error is basically "admin@192.168.1.4's password:
Permission denied, please try again.". I've tried restarting multiple times and deleting users/me/.ssh/known_hosts

I just tried rm .ssh/know_hosts and received  
rm : Cannot find path 'C:\WINDOWS\system32\.ssh\known_hosts' because it does not exist.
At line:1 char:1
+ rm .ssh/known_hosts
+ ~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (C:\WINDOWS\system32\.ssh\known_hosts:String) [Remove-Item], ItemNotFoun
   dException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.RemoveItemCommand

vmavromatis, 2019-02-25:
OK the known hosts file location was for linux but nevermind, that doesn't see to be the issue.Not sure why you can't connect... Have you tried a program like PuTTY to ssh to your pi? https://www.putty.org/

sirenak2, 2019-02-25:
I was using power shell. I'lll try putty now and hope that has a better outcome. I can only imagine something is restricting access in windows. I have been using the A password, the first time I started I though I wrote it down wrong but the second time I didn't . fingers crossed with putty. And thanks, I'm hating myself for being in this situation for something so streamlined

closed 2019-02-26 by sirenak2