raspiblitz/raspiblitz #364: HDD format before being mounted?

CommanderPoe, 2019-02-25:
Hey, i mounted the hdd which it used to have some info mounted on it, i thought that would be a big deal since i was waiting for the raspi to reboot it as stated in the guide... Next thing i found out was the blue screen telling me to wait for the d to be mounted but it actually never happened. I then mounted the hdd on my desktop pc and there were some folders created and the name of hte disk changed to blockchain, i plugged in the sd card and the hdd back into the pi but this time i couldnt even ssh into the pi this time, i wasn't even receiving data from the pi this time. 
Im planning to format the hdd and the sd again and start fresh but i wonder what would be the best format for the hdd? I have been reading people talking about partitions?

rootzoll, 2019-02-25:
Yes a fresh start would be the best try here. Format the HDD with FAT(23) naming it "TEST" and make a fresh sd card again to start with.

If the blue screen is telling you again "waiting for d" please let me know ... because that sounds like an old bug that normally should be fixed. Also please make sure to the latest sd card image that is linked in the main page of the repo. Good luck sailor.

CommanderPoe, 2019-02-26:
@rootzoll First part done, fresh start with a FAT format worked like charm, now heading to the second part, trying to synch the whole blockchain from my mac using the method C with the .Indexes this time.

rootzoll, 2019-02-26:
so far so good 👍

CommanderPoe, 2019-02-26:
Im facing issues with the Fingerprint for the ECDSA key now and i think its because of the dynamic IP of the raspi. 
I freshly started with the new SD card, the FAT format of the disk worked perfectly and i thought everything was going smooth till i got to the part of the synching... When opening the 2nd Terminal Window and trying to copy the whole blockchain using the `sudo scp -r ./chainstate ./indexes ./blocks bitcoin@192.168.1.xx:/mnt/hdd/bitcoin` i keep getting: **WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED! ** The fingerprint for the ECDSA key sent by the remote host is....
[image]

That message was popping in altho i had already ssh'ed into the pi... ok i proceded and Remove the cached key for the raspberry pi IP on the local machine... nothing happened, when i tried `sudo scp -r ./chainstate ./indexes ./blocks bitcoin@192.168.1.xx:/mnt/hdd/bitcoin` i got the same.

I went further and updated the key manually adding it to ~/.ssh/known_hosts but this didnt help either... I got teh same Warning.

Logged in with both users admin and bitcoin and its all flowing normally except for when i use this command with 2 terminal windows open. Any ideas how to fix this?

rootzoll, 2019-02-26:
@CommanderPoe 

See on how to fix this FAQ "What to do when on SSH I see "WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!"": https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#what-to-do-when-on-ssh-i-see-warning-remote-host-identification-has-changed

rootzoll, 2019-02-26:
OK .. looks like thre is something more to it ...

rootzoll, 2019-02-26:
@CommanderPoe can you try to add the following parameter to the scp command: `-o StrictHostKeyChecking=no` should be added after the `-r` 

for backgroind on this see: https://linuxcommando.blogspot.com/2008/10/how-to-disable-ssh-host-key-checking.html

CommanderPoe, 2019-02-26:
@rootzoll your command `sudo scp -r -o StrictHostKeyChecking=no  ./chainstate ./indexes ./testnet3 ./blocks bitcoin@192.168.1.53:/mnt/hdd/bitcoin` didnt work, i suppose that i have to dig deeper into the page you shared to me... Before doing this i wonder... Isnt the same if i just copy .indexes .blocks .chainstate manually thru SFTP to the same location _/mnt/hdd/bitcoin_  using a software like cyberduck? The only question here is should i do it with the bitcoin or admin user for permissions?

rootzoll, 2019-02-26:
bitcoin is the user that needs to own the data on `/mnt/hdd/bitcoin`. 

SFTP could be an option if you know how to setup a SFTP server on the chain computer ... and on the RaspiBlitz you need a working ftp client ... you could try with: 

But I choosed SCP because if you have SSH to login to the RaspiBlitz you should have also SCP available and no server or other client software would be needed for the transfer ... so every feedback how to make that SCP transfere work on your setup is good to improve the process.

rootzoll, 2019-02-26:
BTW you are in your local network and chain data is not critical .. even FTP instead of SFTP should be fine.

CommanderPoe, 2019-02-26:
> bitcoin is the user that needs to own the data on `/mnt/hdd/bitcoin`.
> 
Perfect ...
> SFTP could be an option if you know how to setup a SFTP server on the chain computer ... and on the RaspiBlitz you need a working ftp client ... you could try with: wget
> 
My Desktop pc isnt connected to the pi directly so i most of the time use SFTP but im an amateur.

> But I choosed SCP because if you have SSH to login to the RaspiBlitz you should have also SCP available and no server or other client software would be needed for the transfer ... so every feedback how to make that SCP transfere work on your setup is good to improve the process.
>

You are totally right... any feedback is needed at this stage and you have done an outstanding job putting this guide together! I SALUTE YOU SIER! but in the meantime i started to copying those files using CYBERDUCK login with the bitcoin user... Hope data wont fail on me this time...  Thanks for all the help as usual. Tomorrow i will try to fix the issue with the second login via ssh... that is the only reason why i couldnt use the SCP, but the strangest thing here is that yesterday when i first started with the guide, the first synch of the blockchain ( the one corrupted in the end ) was using your method and nothing happened, no deal regarding the fingerprints... so i dont understand why this now..

CommanderPoe, 2019-02-27:
Update... Copyed like a charm and lightning is already installed and wallet recovered succesfully ... Now waiting for the final synch. It seems that im ready to shine like a lightining again! Thanks @rootzoll ... For my end this issue can now be closed!

closed 2019-02-27 by CommanderPoe