raspiblitz/raspiblitz #365: check all the shopping lists

rootzoll, 2019-02-26:
An issue to collect all things/tasks to check the shopping lists for before next release.

https://github.com/rootzoll/raspiblitz/commit/02b31c31d96b0967aa43ca03900353f27f988af0#commitcomment-32454603

jay653, 2019-03-05:
shopping list Germany:
3.5" display is shown but the case is only for maximum of 3.2" display size

rootzoll, 2019-03-05:
@jay653 thats ok, you just dont use the final topping piece.

rootzoll, 2019-03-10:
add heatsinks:  https://github.com/rootzoll/raspiblitz/issues/415

rootzoll, 2019-03-18:
recheck: power supplies

fluidvoice, 2019-03-18:
> recheck: power supplies

@rootzoll perhaps add this heatsink set. It's the lowest profile that I have seen and very cheap.
https://www.banggood.com/Raspberry-Pi-23-Copper-Heat-Sink-Heatsink-With-3M-Special-Thermal-Cooling-Paste-p-1045808.html

rootzoll, 2019-04-04:
US-Shopping List Power Supply Candidate

https://twitter.com/fluidvoice/status/1113655453127462912
https://www.adafruit.com/product/1995

Plus: Special designed for RBP power problems & AWG20
Minus: Not Amazon Shop (extra shipping)

Not tested yet.

rootzoll, 2019-04-04:
This is the "Leicke" Power Supply from the German Shopping List:
https://www.amazon.de/gp/product/B01JZE38QE

I made very good experience with it - it never printed a "Undervoltage" message. Its designed and distributed by a German company. Looks like a AWG20 cable.

Testresults in RBP with a small heatsink:

```
60.1'C 1200/1400 MHz 1.2500V
63.9'C 1200/1400 MHz 1.2500V
65.5'C 1200/1400 MHz 1.2500V
66.6'C 1200/1400 MHz 1.2500V
67.1'C 1200/1400 MHz 1.2500V
67.7'C 1200/1400 MHz 1.2500V
68.8'C 1200/1400 MHz 1.2500V
Done.

Maximum prime number checked in CPU test: 10000

Test execution summary:
    total time:                          37.9599s
    total number of events:              10000
    total time taken by event execution: 151.8098
    per-request statistics:
         min:                                 11.94ms
         avg:                                 15.18ms
         max:                                 80.64ms
         approx.  95 percentile:              20.90ms

Threads fairness:
    events (avg/stddev):           2500.0000/20.60
    execution time (avg/stddev):   37.9525/0.00

66.6'C 1200/1400 MHz 1.2500V
64.5'C  600/ 600 MHz 1.2500V
65.0'C 1200/1400 MHz 1.2500V
63.4'C  600/1400 MHz 1.2500V
63.4'C 1200/1400 MHz 1.2500V
62.3'C  600/ 600 MHz 1.2V
62.8'C 1200/1400 MHz 1.2500V
62.3'C 1200/1400 MHz 1.2500V
```

Looks really good to me.

This could be a Leike close in the US amazon store
https://www.amazon.com/Yelesley-Raspberry-Micro-USB-Charger-Raspberry-Pi/dp/B07F3QP7PQ

fluidvoice, 2019-04-04:
Looks like the **max voltage is 1.3750V** 
One guy on Twitter posted this (note: he's using PoE not standard PS) but I wonder if we should look for this as a signal of a VERY good PS.
https://twitter.com/thinkmassive/status/1113631126025719808

> ...with the lid on there's a bit of throttling, not bad:
> 
> Threads started!
> 48.3'C 1400 MHz  1.3750V
> 56.9'C 1400 MHz  1.3750V
> 59.6'C 1400 MHz  1.3750V
> 59.6'C 1200/1400 MHz 1.3750V
> 60.1'C 1400 MHz  1.3750V
> 60.1'C 1200/1400 MHz 1.2750V
> Done.

rootzoll, 2019-04-09:
I updated as far as possible all power supplies and made the change to the geekworm heatsink-case.

Found for the US a special power supply that especially addresses the undervoltage problem:
https://www.amazon.com/dp/B07MC7B9X3

closed 2019-04-09 by rootzoll

fluidvoice, 2019-04-09:
@rootzoll  There is a potential the LCD in the Germany shopping list has a problem. See: https://github.com/rootzoll/raspiblitz/issues/460#issuecomment-481399691
Have other people confirmed it works?

rootzoll, 2019-04-09:
I build several RaspiBlitzes with that Quimat LCD from the german shopping list - they all worked.